reuterbal/photobooth #4: Nikon D7000 capture resolution

Tyranus333, 2017-11-02:
Hi,
Python incompetant here (speak slowly with small words).
Using a Nikon D7000 as the camera in a photobooth on a raspberry pi with the photobooth code supplied very kindly here. 
The D7000 is capapble of taking 300dpi images, which it does when triggered manually gphoto2 --capture etc. however when the camera is triggered by the photobooth with no mention or settings that I can see restricting the dpi it returns with a 19MB file and 96dpi image.
Is there a file somewhere that can be changed to fix this to be a standard 5MB JPEG and 300dpi image?
(Also if it's easy to capture RAW files without too much issue this would be of great use)
Thanks,

reuterbal, 2017-11-07:
Hi,
sorry for some delay in replying.

Capturing RAW files is not a problem - in fact, I sometimes forget to switch the mode on my camera and the photobooth crashes because it downloads RAW files. The problem is that RAW files can't be processed immediately in the same way the JPEG files are: it would require to develop them first before being able to assemble them in the 2x2 grid. Thus, it's unfortunately not feasible to operate the photobooth with RAW files.
If you're just interested in the unassembled RAW files, you might want to try to set your camera to RAW+JPEG capture mode. I don't know what gPhoto will do then, but it might download just the JPEG file and you are left with the RAW file on the memory card. 

Regarding the dpi issue: 
The photobooth software downloads the pictures from the camera and stores them as temporary files in /tmp/photobooth_01.jpg, _02.jpg, etc. Afterwards, it assembles them to the 2x2 grid and stores the result in a new file.
Can you check what size and dpi values the files in /tmp have? 

Best,
Balthasar

Tyranus333, 2017-11-08:
Hi,
Thanks for the reply anyway. We've used the app because 96dpi was better than nothing and on processing it seems that it has saved the RAW image as a jpeg rather than just a jpeg.
The camera is set to shoot both, however because the capture is being triggered by libgphoto2 I presume that it's configuration in the capture portion will stipulate to only capture the 1 part (in this case RAW is produced first) and the JPG is dropped. 
The original images are being captured as 96dpi as we have added: 
`        #Copy the photos to HDD
        now = datetime.now()      
        shutil.copy2('/tmp/photobooth_00.jpg',"/tmp/photobooth_photos/photobooth_" + str(now) + "_00.jpg")
        shutil.copy2('/tmp/photobooth_01.jpg',"/tmp/photobooth_photos/photobooth_" + str(now) + "_01.jpg")
        shutil.copy2('/tmp/photobooth_02.jpg',"/tmp/photobooth_photos/photobooth_" + str(now) + "_02.jpg")
        shutil.copy2('/tmp/photobooth_03.jpg',"/tmp/photobooth_photos/photobooth_" + str(now) + "_03.jpg")`
from line 315  just prior to saving the assembeled image.

reuterbal, 2017-11-08:
In that case it seems to be a gphoto-related issue. I don't know how to set the dpi when triggering capture, and it's probably a camera-specific setting as well. If you want to fix it, you probably have to find out how to set the dpi using gphoto, and then modify camera.py for the wrapper that you are using (piggyphoto, gphoto2cffi, system calls). 

A remark regarding your approach to save the original pictures: you can save some copy operations, if you tell the script to save your photos in the right location right away. For that, replace l.372 by the following:

```
now = str(datetime.now())
filenames[x] = self.camera.take_picture("/tmp/photobooth_photos/photobooth_%s_%02d.jpg" % (now, x))
```

reuterbal, 2018-03-17:
I'm closing this due to inactivity.

closed 2018-03-17 by reuterbal