reuterbal/photobooth #5: Unable to capture pictures

danikuci1, 2017-12-20:
Hello and thanks for this project, it looks really cool.

I'm having some trouble getting it to work and hope you can help:
I built it per your instructions, start up the program, and when it goes it take a picture it just reads "ERROR: SERIOUS ERROR" with no other output. 

I thought it might be related to the fact that I was using Raspian Stretch when this was obviously developed pre-stretch. I reflashed my pi with Jessie but am getting the same results.

I have connected and tried this with two different cameras, and both are able to take a picture from command line with `gphoto2` just fine. I also found hackerb9's PR/spork and attempted to use that version but got the same results. 

At this point I am unsure of what to try, or if there is any log file I can interrogate for more info and would appreciate some assistance. Thanks in advance.

reuterbal, 2017-12-20:
Please tell me some more about your setup: What camera are you using? Have you installed any of the Python-Bindings for gPhoto?
What is the output in the command line when the photobooth displays "ERROR: SERIOUS ERROR"?

danikuci1, 2017-12-20:
Cameras are the Nikon D3100 and Olympus E-P3.
gphoto2cffi bindings are installed.

here is the output from it just now:

```bash
pi@photobooth:~/photobooth $ ./photobooth.py 
Gphoto2cffi available
Info: Number of last existing file: 0
Info: Saving assembled pictures as: 2017-12-20/picXXXXX.jpg
No handlers could be found for logger "libgphoto2.gp_port_usb_close [libusb.c:325]"
Your Raspberry Pi is board revision 3
RPi.GPIO version is 0.6.3
SERIOUS ERROR: GPhoto2Error('Unsupported operation',)
Error: SERIOUS ERROR!
pi@photobooth:~/photobooth $ 
```

reuterbal, 2017-12-21:
Thanks for the output. Can you try two things?

1. Have you removed the files mentioned in point 5 of the installation (https://github.com/reuterbal/photobooth#installation)? These cause the system to intersect the access to the camera (to allow for you to copy pictures from the camera to your computer).

2. Can you try using the command line interface? For that, simply comment out ll. 17 to 23 in camera.py:
```
#try:
#    import gphoto2cffi as gp
#    gpExcept = gp.errors.GPhoto2Error
#    gphoto2cffi_enabled = True
#    print('Gphoto2cffi available')
#except ImportError:
#    pass
```

With that it should fall back to using the (slower) cli wrapper, which seems to work. 
Please post the output of this version as well.

reuterbal, 2017-12-21:
On second thoughts, the problem is probably the live preview: According to http://gphoto.sourceforge.net/doc/remote/, the Nikon D3100 doesn't support Viewfinder with gPhoto. There is no mentioning of the Olympus, so I don't know if this one does.

Thus, if 2. above works, try out using gphoto2cffi again and replace l. 104 of camera.py
``` return gphoto2cffi_enabled or piggyphoto_enabled```
by
``` return false```
This disables the preview functionality.

sqkid89, 2018-03-10:
Hi,
I recently came across this project and I had this exact same problem.  I followed your #2 suggestion and that did work.  However, changing to the "return false" resulted in the original problem again.  I'm wondering if there's another line possibly that may need tweaking as well?

Thanks!

reuterbal, 2018-03-10:
Could you post the output from the command line?
Do you have gphoto2cffi installed?

sqkid89, 2018-03-11:
I do have gphoto2cffi installed, rechecked that just a moment ago.  When I get the "Error: Serious Error!" screen, the quit command will not work, nor will Ctrl-C or Ctrl-Z (or anything else I seem to find for suggestions), so I'm unable to see the command line output.  I'm semi new to this, so my apologies if I'm missing something simple.

reuterbal, 2018-03-15:
You can pipe the command line output to files, e.g., using

```sh
sudo python photobooth.py >>photobooth.log 2>>photobooth.err
```

sqkid89, 2018-03-17:
Thank you so much!  I appreciate the help.  the err file contained this:
No handlers could be found for logger "libgphoto2.gp_port_usb_close [libusb.c:325]"

reuterbal, 2018-03-17:
Thanks for the follow-up. What is your setup (camera, pc hardware, ...)?
Could you post the entire log- and err-file?

It's a shot into the blue but maybe your camera model doesn't support the idle setting. I added this because my Canon would keep the sensor active the entire time but it's quite possible this works differently for other makes and models.
For that replace ll. 128 in camera.py by the following
```python
def set_idle(self):
    pass
```

sqkid89, 2018-03-17:
So I got slightly further, slightly back.  I think your recommendation is in the right direction and maybe I just implemented it incorrectly.  Here is how I tweaked camera.py based on your above statement:

    def set_idle(self):
         pass
#        if gphoto2cffi_enabled
#            self.cap._get_config()['actions']['viewfinder'].set(False)
#        elif piggyphoto_enabled:
            # This doesn't work...
#           self.cap.config.main.actions.viewfinder.value = 0

(Like I said, I'm slightly new to editing these kinds of scripts, but I know '#' stops the line from running, and its my understanding that we don't want any of these lines running since we're not trying to give the camera any kind of idle-related commands.  Or am I completely off here...?)

after making that change, I ran the photobooth.py, managed to load, but when I pushed a button to start the photo taking process, I got a serious error.
photobooth.log shows:

Gphoto2cffi available
Info: Number of last existing file: 0
Info: Saving assembled pictures as: 2018-03-17/picXXXXX.jpg
Your Raspberry Pi is board revision 3
RPi.GPIO version is 0.6.3
SERIOUS ERROR: NameError("global name 'false' is not defined",)
Error: SERIOUS ERROR!

reuterbal, 2018-03-17:
This looks very promising now.
Simply replace ```false``` by ```False``` in l.104 (uppercase, python is case-sensitive there). I just realized I made that mistake in my earlier comment...

sqkid89, 2018-03-18:
Thank you! More progress!
Changed that line, now it loads, allows me to begin to take photos, but freezes after the first one has been taken. Camera takes image, but display still shows the "Smile!!! 1 if 4" image.  No new error messages in the .log file, and the only thing I get in the .err file is 

"No handlers could be found for logger "libgphoto2.gp_port_usb_close [libusb.c:325]"

reuterbal, 2018-03-19:
In that case, it seems like an issue somewhere upstream. Once again: what camera make and model are you using? Have you checked that it is supported by gPhoto2? And: can you please attach log-files as well?

sqkid89, 2018-03-20:
Model is Nikon Coolpix S3300, which was purchased specifically because it was supported by gphoto (image capture).  I will try to grab the log files, thanks!

sqkid89, 2018-03-24:
So I'm trying to get a log.  However, when I run photobooth, and it hangs after the first photo is taken, I see no log information if I remote in when its still running, or after I've remotely restarted the pi.  The only log data available to me is from when we had the error due to the case-sensitive false early.

Is it not saving a log due to being forced to exit before it finishes running?  Or am I really showing my novice skills (sorry)?

reuterbal, 2018-03-24:
How are you launching the python program? 

When you start it like this
```bash
sudo python photobooth.py >>photobooth.log 2>>photobooth.err
```
then `photobooth.log` should contain the usual startup information plus maybe something useful, and `photobooth.err` hopefully some error message.

When it's hanging, you can press Ctrl+Alt+F1 to get a different console than the default GUI. After logging in there, you can for example kill the program using
```bash
sudo killall photobooth
```
and return to the GUI by pressing Ctrl+Alt+F7

Just to double-check:
1. Which version of Python and gPhoto2 are you using?
2. Can you capture pictures using the gphoto2-executable on the command line?
3. If 2. works, does the photobooth work if you disable gphoto2-cffi as described somewhere above?
4. Have you executed all the installation steps described in the README? Especially the part where you have to delete some existing system-files?

sqkid89, 2018-03-24:
I've been launching with the line you noted about, which creates the log and err files.  Doing all of this from command line to keep the speed up/install size down.

I was able to get things working when I followed your step of editting 17 to 23 in camera.py, although it was incredibly slow between taking the image and actually moving on to the next countdown.  

Since I was new to all of this, I started with Raspian Jesse, since I read some issues regarding python and other packages.  I then tried to find versions close to the ones you noted: Python is 2.7.9-1 and used the gonzalo updater step.

This is going to sound a bit ridiculous now, because I'm trying to find all of the packages I've got installed - does the instructions in the readme assume that we already have gphoto installed prior to running the gonzalo updater?  I'm thinking maybe I stumbled across the issue

reuterbal, 2018-03-27:
This sounds reasonable so far. gPhoto does not have to be install prior to running the gonzalo script.
Unfortunately I can't spot any issues. One thing that you might try is to run
```bash
sudo apt-get build-dep libgphoto2-dev
```
before compiling gphoto - maybe there is some dependency missing that was left out in the build process.

However, the installation instructions are from 2015 and Debian upgraded their package versions for stretch in the meantime. Another avenue to tackle the problem might be to use Raspbian Stretch and just install gphoto by 
```bash
sudo apt-get install gphoto2
```
instead of the updater-script.

sqkid89, 2018-04-02:
Thank you for your help!  The input is much appreciated.  I am still attempting to resolve, but have to travel for work for the next 2 weeks.  will continue when I return

reuterbal, 2018-05-04:
Glad to hear that it is working for you now! Another source of error I didn't think of. Your issues are most certainly due to the lite version of Raspbian.

This version is not bundled with a desktop environment and as such, lacks many packages that are required for GUI applications. I'm a bit disappointed that pygame-dependencies would not pull in the requirements. 

I'm closing this issue - feel free to open another when problems arise with the bindings.

closed 2018-05-04 by reuterbal