reuterbal/photobooth #6: development branch install?

l0c0luke, 2018-04-01:
I know you are working on this pretty much as I type this, but I am curious if you can give me a hint on how to install the dev branch please?

I am new and fresh to this whole Pi world but I want to start helping and figured I would start with the latest stuff. Am I too early?

reuterbal, 2018-04-03:
Hi,
great to hear that you want to try out the development version. However, this is still very much work in progress, breaking changes are happing all the time, and installation is not documented yet. But worst, it does not yet have the features of the old version. 
The README.md is not yet updated, however dependencies are listed in INSTALL.md.

I recommend to wait some weeks until it enters a more major state. I'd appreciate every beta tester then.

closed 2018-04-03 by reuterbal

reuterbal, 2018-06-15:
The development version is live now. Feel free to try it out!