reuterbal/photobooth #7: ERROR: SERIOUS ERROR in Every Mode exept Manual

Bockwurst2, 2018-04-21:
Hi,
first of all thank you for this nice Projekt.

I use the Canon 500 D like you, but in every other Camara mode then "Manuel" it won't take a Photo.
Is this normal or do i have to change somthing?

I try step 5. from the installation Manual but there are no files.

reuterbal, 2018-04-24:
Hi Bockwurst2,

in principal, the camera mode should be irrelevant. I have successfully used manual mode, aperture automatic and time automatic, but others should work as well.

Please tell me more about your setup:
- Which gPhoto2 wrapper are you using? Command line, piggyphoto, gphoto2-cffi?
- Which gPhoto2 version are you using?
- Please include output and error log of the photobooth.
- What do you mean by "won't take a photo"? Does the camera do anything (e.g., try to focus, do you hear a 'shutter' sound from the mirror, etc.)?

reuterbal, 2018-05-07:
Hi bockwurst2,

thank you for your infos.

>  If i run the camera.py sktipt it returns "Gphoto2cffi available".

This means it is using gphoto2-cffi - this is the default if the corresponding wrapper is installed. If gphoto2-cffi is missing, it tries to use piggyphoto and -- if that one isn't available either -- it uses the command line wrapper.

Could you check, if your camera is set to RAW capturing format? The following line in the log suggests that:
> Can't open image '/home/pi/photobooth/2018-04-21/pix00016.jpg': Unsupported image format",)

Bockwurst2, 2018-05-10:
I've checked the camara. If i try to user RAW+L it the progamm dont work. After i delete the "pix00016.jpg" it has the same Log, except this line.

reuterbal, 2018-05-10:
What happens if you set the camera to 'L' (without RAW)?

Bockwurst2, 2018-05-10:
All previous descriptions refer to the "L" mode. From her profile recognize that they are German. Is it OK for you to continue the conversation in German?

reuterbal, 2018-05-11:
Lieber Bockwurst2,

wir koennen die Konversation auch auf Deutsch fortsetzen, wenn Ihnen das leichter faellt.

Um das Problem nochmal zusammen zu fassen:

* Im manuellen Modus 'M' funktioniert alles wie gewuenscht.
* In jedem anderen Modus (z.B. Blendenvorwahl 'Av', Zeitvorwahl 'Tv', Automatik, ...) werden keine Fotos aufgenommen
* Funktioniert die Anbindung von gPhoto in diesen Modi an sich? Dazu einfach mal die Kamera auf den entsprechenden Modus einstellen und in der Kommandozeile `gphoto2 --capture-image-and-download` eingeben.
* Wichtig: Die Kamera muss auf ein JPEG-Format eingestellt sein (S, M, L), nicht RAW oder RAW+L. Sonst wird die RAW-Datei herunter geladen, die von der Software nicht weiter verarbeitet werden kann.

Bockwurst2, 2018-05-12:
Um den Fehler einzugrenzen und auszuschließen, dass es an meinem Setup lieg habe ich den PI noch einmal neu aufgesetzt.  

- Bei der Installation bin ich nach Ihrer Anleitung vorgegangen. Nach Schrtitt  7 hab  ich im Modus AV getestet. Autofokus funktioniert. LIveview nicht. 

- Nach Schritt 8.1 Funktioniert Liveview aber kein Autofokus.

- In Modus 'M' auch klein Autofocus

- Kamera steht immer auf JPEG-Format 'L' RAW habe ich nur einmal zu Testen versucht, Daher der Fehler mit der defekten JPG.

reuterbal, 2018-05-15:
Lieber Bockwurst2,

danke fuer die weiteren Untersuchungen. Das sieht ja schon ganz gut aus.

> Bei der Installation bin ich nach Ihrer Anleitung vorgegangen. Nach Schrtitt 7 hab ich im Modus AV getestet. Autofokus funktioniert. LIveview nicht.

Das ist so zu erwarten. Ohne die Installation der Bindings wird einfach die Kommandozeilenapplikation von gphoto2 aufgerufen, mit der keine Vorschau moeglich ist. Photos werden vermutlich auch aufgenommen und das endgueltige Bild generiert?

> Nach Schritt 8.1 Funktioniert Liveview aber kein Autofokus.
> In Modus 'M' auch klein Autofocus

Ich vermute, Liveview und auch das Aufnehmen der Fotos funktioniert - nur eben nicht mit korrektem Fokus?
Die Autofokus-Probleme habe ich auch beobachten koennen. Diese sind mir bei den letzten Einsaetzen nicht aufgefallen, da ich ein Objektiv mit langsamen Fokusmotor im Einsatz hatte und so manuell auf den geeigneten Bildbereich fokussiert hatte. Ich werde mir das in den naechsten Tagen genauer ansehen.

> Kamera steht immer auf JPEG-Format 'L' RAW habe ich nur einmal zu Testen versucht, Daher der Fehler mit der defekten JPG.

Das ist gut.

> Vollautomatik funktioniert immer noch nicht. Fehlermeldung wie gehabt.

Das ist seltsam. Das werde ich mir auch ansehen.

Bockwurst2, 2018-05-16:
Vielen dank für die Mühe und das eröffnen des Ticktets gegen gPhoto. Das werde ich im Auge behalten. Sollte es da zu einem Fix kommen werde ich auch noch mal das Photobooth Skript testen und an dieser Stelle Bescheid geben.

reuterbal, 2018-05-18:
Ein kurzes Update: Ich habe anscheinend den notwendigen config-key gefunden, um den Autofokus zurueck zu bringen (siehe der bugreport bei gphoto2).
Um das im Skript einzubauen einfach in camera.py vor Zeile 116 das Folgende einfuegen:
```
self.cap._get_config()['actions']['viewfinder'].set(False)
self.cap._get_config()['settings']['output'].set('Off')
```

reuterbal, 2018-06-14:
Freut mich, dass alles klappt. Ich habe eine neue, komplett ueberarbeite Version der App in den `master`-branch uebertragen. Darin ist die Kamera-Ansteuerung in einen eigenen Prozess ausgelagert und dadurch wesentlich performanter. Die Wartezeiten konnte ich da kaum noch beobachten.

Ich freue mich ueber Testberichte!

Bockwurst2, 2018-06-19:
Hi,  ich habe die neue Version am Wochenende kurz getestet. 
Ein Paar abstürze hatte ich, beim ändern von Einstellungen bzw. beim Starten. Die konnte ich aber nicht sauber Reproduzieren. Da werde ich aber noch mal was mit rumspielen um ordentliche Testberichte einzustellen.   

An der Stelle noch einmal vielen Dank für die viel Arbeit.

Noch eine Frage zur neuen Version. 
Sehe ich das richtig das es keine Slideshow mehr gibt?

reuterbal, 2018-06-19:
Hi,
Slideshow im inaktiven Zustand gibt es (noch) nicht. Fuer eine externe Slideshow kann einfach das alte Skript weiter verwendet werden.

Beim Speichern der Einstellungen wird die Anwendung neu gestartet - dabei wird das Fenster geschlossen und erscheint wieder, aber je nach Hardware erst nach einigen Sekunden. Das ist kein Absturz sondern die einfachste Moeglichkeit, um sauber die aktuellsten Einstellungen zu laden.