reuterbal/photobooth #8: [FEATURE REQUEST] Take only one picture

wikijm, 2018-05-07:
Hi @reuterbal,

Is it possible to take only one picture?
Sadly, my python skills are near to zero.

I assume that it's needed to comment lines 242 to 284, a bunch of other things then 'range' value at line 355 and 356.

Thanks in advance 😃

reuterbal, 2018-05-07:
Hi wikijm,

it's not implemented but you can probably add it yourself quite easily. You will probably have to replace lines 355 to 395 by something like this (untested!):

```python
show_counter(self.pose_time)

# Try each picture up to 3 times
remaining_attempts = 3
while remaining_attempts > 0:
    remaining_attempts = remaining_attempts - 1

    self.display.clear()
    self.display.show_message("S M I L E !!!\n\n" + str(x+1) + " of 4")
    self.display.apply()

    try:
        outfile = self.camera.take_picture(self.pictures.get_next())
        remaining_attempts = 0
    except CameraException as e:
        # On recoverable errors: display message and retry
        if e.recoverable:
            if remaining_attempts > 0:
                self.display.clear()
                self.display.show_message(e.message)  
                self.display.apply()
                sleep(5)
            else:
                raise CameraException("Giving up! Please start over!", False)
        else:
            raise e
```

HTH!

wikijm, 2018-05-08:
Hi @reuterbal,

I'll make a try, and certainly provide to you the final solution if working 😉 

Is it possible to let this issue open during one month?
I'll go to a weeding soon, that's why I'm asking.

reuterbal, 2018-05-11:
Sure, I'll leave this open for a while.

reuterbal, 2018-06-14:
I have completely rewritten the app, allowing to choose any number of pictures to take. See the README for more details.

wikijm, 2018-06-14:
Awesome @reuterbal, thanks a lot!

I'll make a try tomorrow. I see that you leave the possibility to choose how many picture you can have in horizontal and vertical direction, so cool 😃

Thanks a lot for your wonderful work and your time 😉

closed 2018-06-14 by wikijm

FizzyMUC, 2019-06-17:
Just found this closed issue. Just for clarification: In order to only have 1 picture per page, I set it to 1 (horizontal) x 1 (vertical), correct?
Cannot test currently because I can't get it running anyways. Need to resolve during this week for a wedding :D

reuterbal, 2019-06-18:
Yes.