reuterbal/photobooth #10: [FEATURE REQUEST] Take JPEG and RAW Photo and leave RAW in Camera

Bockwurst2, 2018-05-16:
Hi @reuterbal,

Is it possible to take a RAW and JPEG picture but only download and process the JPEG?

I use Gphoto2cffi so i think it use line 123 to write the capture Image without using the SD Card on the Camera to buffer?

Thanks in advance
Bockwurst2

reuterbal, 2018-06-15:
I usually capture photos using RAW format, however, I have to switch to 'L' for photobooth use. Regularly, I forget to switch it back afterwards, which is very annoying. 
I'm going to add a feature that will change the capture format on startup and restores the original setting afterwards.

reuterbal, 2018-07-03:
There is another request (#17) to store the captured pictures on camera. I might add that at some point, however mixing RAW and JPG does seem to much of unnecessary complexity. Feel free to add it, but I'm closing it due to inactivity.

closed 2018-07-03 by reuterbal

FizzyMUC, 2019-06-17:
Just found this closed issue. Just for clarification: In order to only have 1 picture per page, I set it to 1 (horizontal) x 1 (vertical), correct?
Cannot test currently because I can't get it running anyways. Need to resolve during this week for a wedding :D

commit c906ff5e (2018-07-03): Added functionality to switch automatically to JPEG Fine and restore the original setting on teardown. This closes #10