reuterbal/photobooth #11: [Success Story] French wedding in June 2018

wikijm, 2018-06-14:
Hi @reuterbal,

Thanks to your wonderful project, I had the opportunity to offer my sister a great home photo booth for her wedding.

It also allowed me to do a joint activity with my (future) brother-in-law, who was in charge of assembling the wooden form-work and everything that allowed me to integrate the electronics, the Raspberry and the camera.

### Equipment used:

| Picture | Name | Price
| :--------------: | :--------------: | :--------------:
| [image] | [Aukru Raspberry Pi 3 Power supply 5V 3000mA with switch](https://www.amazon.fr/Aukru-Chargeur-Adaptateur-Alimentation-Raspberry/dp/B01M58O9M9/ref=sr_1_19?ie=UTF8&qid=1529013208&sr=8-19&keywords=raspberry%2Bpi%2B2&th=1) | 9,99 €
| [image] | [Waveshare 7inch HDMI LCD (C) Capacitive Touch Screen Display Supports Various Systems for All Ver. Raspberry pi 3 Model B/ 2B/B+/B/A BeagleBone Black Banana Pi/Pro Video Photo Kit with Bicolor Case](https://www.amazon.fr/Waveshare-LCD-HDMI-r%C3%A9solution-Raspberry/dp/B01HPV7M4I/ref=sr_1_58?ie=UTF8&qid=1529013265&sr=8-58&keywords=raspberry%2Bpi%2B2&th=1)  | 67,99 €
| [image] |  [MakerStack 150Pcs 10CM Multicolore Dupont Wire Jumper](https://www.amazon.fr/MakerStack-Multicolore-Solderless-BreadBoard-Raspberry/dp/B07D5XF44S/ref=sr_1_65_sspa?ie=UTF8&qid=1529013292&sr=8-65-spons&keywords=raspberry%2Bpi%2B2&th=1)  | 9,99 €
| [image] | [Long Length Arcade Game HAPP Style Push Button for Mame and Jamma 6 Colors](https://www.ebay.fr/itm/Long-Length-Arcade-Game-HAPP-Style-Push-Button-for-Mame-and-Jamma-6-Colors/132625572130?hash=item1ee119a122:g:yW4AAOSw~Bha~Ovi) | 1,29 €
| [image] | [DR-E8 LP-E8 LPE8 Dummy Batterie + 5V-8.4V Câble USB Pilote ACK-E8 + 5V3A Adaptateur Chargeur pour Canon EOS 550D 600D 650D 700D T4i X6i](https://www.amazon.fr/gp/product/B0748G9GPG/ref=oh_aui_detailpage_o04_s00?ie=UTF8&psc=1) | 22,99 €
| [image] | [StarTech.com USB2HABM1 Câble Mini USB 2.0 30 cm A vers Mini B M/M](https://www.amazon.fr/StarTech-com-USB2HABM1-C%C3%A2ble-Mini-vers/dp/B002L5U7N2/ref=sr_1_3?ie=UTF8&qid=1529015019&sr=8-3&keywords=mini-usb+cable) | 2,39 €
| [image] | [Canon EOS 550D 18 Mpix SLR Digital Camera Lens Kit 18-55mm IS Black](https://www.amazon.fr/dp/B01MCUQJGC/ref=sspa_dk_detail_1?psc=1&pd_rd_i=B01MCUQJGC&pf_rd_m=A1X6FK5RDHNB96&pf_rd_p=1370598536209589035&pf_rd_r=QDTSJQSS77P85KVAK9Z1&pd_rd_wg=nrfGc&pf_rd_s=desktop-dp-sims&pf_rd_t=40701&pd_rd_w=PtHKu&pf_rd_i=desktop-dp-sims&pd_rd_r=f75fea77-7021-11e8-80b3-9131ac438cd8) | 443,99 €

### Result
[image]
[image]
[image]
[image]

### Maintenance
To take care of remote maintenance, I added a Wifi key which allowed me to create an access point and connect me in SSH in order to :

- [Control the Raspberry remotely with the JuiceSSH application (Android)](https://play.google.com/store/apps/details?id=com.sonelli.juicessh)
- [Get the pictures on my smartphone with the AndFTP application (Android)](https://play.google.com/store/apps/details?id=lysesoft.andftp)

If you have any questions or need feedback, please do not hesitate to contact me 😉

reuterbal, 2018-06-15:
Hi @wikijm

thanks a ton for that detailed story! Your photobooth looks incredible.
Have you experienced any software glitches/problems during the event which could have been prevented? I'm always trying to make the software more stable.

wikijm, 2018-06-15:
I had a rather sad problem, but with a revisited version of your script in order to make only one image.

Indeed, the marriage took place over 2 days (From 17h the 1st day to 3h the 2nd day). When the script was launched, the folder corresponding to the 1st date was created, but this was not the case for the 2nd. Result: All the photos of the 2nd day were lost (loss estimated at about twenty images according to the counter of trigger of the reflex).

I will take some time today and probably this weekend to adapt my success story to your template 😉

reuterbal, 2018-06-15:
Don't worry about the template - your feedback is very much appreciated in the current form!
The directory problem is very unfortunate. In fact, I should check whether this could happen with the current version as well!

wikijm, 2018-06-15:
Updated to match [success-story](https://github.com/reuterbal/photobooth/blob/master/.github/ISSUE_TEMPLATE/success-story.md) template :sunglasses:

thyristor59, 2018-08-11:
hello, i'm so sorry for my english, i'm french... and i would like know change english text in french text...

it's for a weeding to my best friend !

I all ready install camera, print and photobooth application!
I start on raspberry pi, i'm a NOOB =)

thanks

thyristor59, 2018-08-11:
bonsoir, je me permet de te transmettre mon mail: pierre-v@live.fr pour éviter de flood le topic!

j'ai en matériel:
- raspberry pi 3B, 
- nikon D3200, 
- et pour le moment en imprimante une canon pixma ip 2850 pour tester!

grâce au tuto, j'ai tout installé et le logiciel est fonctionnel, cependant j'aimerais avoir le texte en français,  je commence a peine avec le raspberry, comme dit précédemment.
Je ne comprends pas comment modifier les fichiers =(

je mis prends a l'avance... Juin 2019...
Je veux surtout comprendre ce que je fais, j'ai soif d'apprendre, toutes aides/conseils me réjouirais!

cordialement =)

reuterbal, 2018-08-21:
As this is included in the Wiki now, I'll close this.

closed 2018-08-21 by reuterbal

richard69800, 2020-05-11:
hello, i live in france, but i would like to knox if you could send to me the plan to do this photobooth please? is it possible to connect a printer whith your system? 
thank you for the answer, best regards from france.

i wish, you will answer me ^^

richard
mail: richardvartanian69@gmail.com

wikijm, 2020-05-11:
> 
> 
> hello, i live in france, but i would like to knox if you could send to me the plan to do this photobooth please? is it possible to connect a printer whith your system?
> thank you for the answer, best regards from france.
> 
> i wish, you will answer me ^^
> 
> richard
> mail: [richardvartanian69@gmail.com](mailto:richardvartanian69@gmail.com)

I'll send you an email to discuss 😉