reuterbal/photobooth #13: Multiple '/usr/lib/python3.4/distutils/dist.py:260: UserWarning: Unknown distribution option' during fresh install

wikijm, 2018-06-16:
Hi @reuterbal,

**Describe the bug**
During a fresh install, I get multiple `/usr/lib/python3.4/distutils/dist.py:260: UserWarning: Unknown distribution option` error during a fresh install

**To Reproduce**
Steps to reproduce the behavior:
Follow [install instructions](https://github.com/reuterbal/photobooth/blob/master/INSTALL.md), from [Configure and update Raspbian](https://github.com/reuterbal/photobooth/blob/master/INSTALL.md#configure-and-update-raspbian) to [Install photobooth with dependencies](https://github.com/reuterbal/photobooth/blob/master/INSTALL.md#install-photobooth-with-dependencies)

**Expected behavior**
A functional installation

**Screenshots**
Here is a 'shell video record' of the install process I played:
https://asciinema.org/a/wci1XLgG52wxv37y1W10kFMm4

**Hardware (please complete the following information):**
 - Device Raspberry Pi 2
 - Camera: N/A
 - GPIO: N/A

**Software (please complete the following information):**
  - OS: Raspbian GNU/Linux 9 (stretch) (partial result of cat /etc/*-release command)
  - Python: 3.4.2 (result of python3 --version command)

**Photobooth log:**
N/A, please find /home/pi/.pip/pip.log instead:
```bash
------------------------------------------------------------
/home/pi/photobooth/.venv/bin/pip run on Sat Jun 16 17:46:57 2018
Obtaining file:///home/pi/photobooth
  Running setup.py (path:/home/pi/photobooth/setup.py) egg_info for package from file:///home/pi/photobooth
    /usr/lib/python3.4/distutils/dist.py:260: UserWarning: Unknown distribution option: 'long_description_content_type'
      warnings.warn(msg)
    /usr/lib/python3.4/distutils/dist.py:260: UserWarning: Unknown distribution option: 'python_requires'
      warnings.warn(msg)
    /usr/lib/python3.4/distutils/dist.py:260: UserWarning: Unknown distribution option: 'project_urls'
      warnings.warn(msg)
    Traceback (most recent call last):
      File "<string>", line 17, in <module>
      File "/home/pi/photobooth/setup.py", line 205, in <module>
        'Source': 'https://github.com/reuterbal/photobooth/',
      File "/usr/lib/python3.4/distutils/core.py", line 108, in setup
        _setup_distribution = dist = klass(attrs)
      File "/home/pi/photobooth/.venv/lib/python3.4/site-packages/setuptools/dist.py", line 266, in __init__
        _Distribution.__init__(self,attrs)
      File "/usr/lib/python3.4/distutils/dist.py", line 280, in __init__
        self.finalize_options()
      File "/home/pi/photobooth/.venv/lib/python3.4/site-packages/setuptools/dist.py", line 301, in finalize_options
        ep.load()(self, ep.name, value)
      File "/home/pi/photobooth/setuptools_scm-2.1.0-py3.4.egg/setuptools_scm/integration.py", line 10, in version_keyword
      File "/home/pi/photobooth/setuptools_scm-2.1.0-py3.4.egg/setuptools_scm/version.py", line 44, in _warn_if_setuptools_outdated
    setuptools_scm.version.SetuptoolsOutdatedWarning: your setuptools is too old (<12)
    Complete output from command python setup.py egg_info:
    /usr/lib/python3.4/distutils/dist.py:260: UserWarning: Unknown distribution option: 'long_description_content_type'

  warnings.warn(msg)

/usr/lib/python3.4/distutils/dist.py:260: UserWarning: Unknown distribution option: 'python_requires'

  warnings.warn(msg)

/usr/lib/python3.4/distutils/dist.py:260: UserWarning: Unknown distribution option: 'project_urls'

  warnings.warn(msg)

Traceback (most recent call last):

  File "<string>", line 17, in <module>

  File "/home/pi/photobooth/setup.py", line 205, in <module>

    'Source': 'https://github.com/reuterbal/photobooth/',

  File "/usr/lib/python3.4/distutils/core.py", line 108, in setup

    _setup_distribution = dist = klass(attrs)

  File "/home/pi/photobooth/.venv/lib/python3.4/site-packages/setuptools/dist.py", line 266, in __init__

    _Distribution.__init__(self,attrs)

  File "/usr/lib/python3.4/distutils/dist.py", line 280, in __init__

    self.finalize_options()

  File "/home/pi/photobooth/.venv/lib/python3.4/site-packages/setuptools/dist.py", line 301, in finalize_options

    ep.load()(self, ep.name, value)

  File "/home/pi/photobooth/setuptools_scm-2.1.0-py3.4.egg/setuptools_scm/integration.py", line 10, in version_keyword

  File "/home/pi/photobooth/setuptools_scm-2.1.0-py3.4.egg/setuptools_scm/version.py", line 44, in _warn_if_setuptools_outdated

setuptools_scm.version.SetuptoolsOutdatedWarning: your setuptools is too old (<12)

----------------------------------------
Cleaning up...
Command python setup.py egg_info failed with error code 1 in /home/pi/photobooth
Exception information:
Traceback (most recent call last):
  File "/home/pi/photobooth/.venv/lib/python3.4/site-packages/pip/basecommand.py", line 122, in main
    status = self.run(options, args)
  File "/home/pi/photobooth/.venv/lib/python3.4/site-packages/pip/commands/install.py", line 290, in run
    requirement_set.prepare_files(finder, force_root_egg_info=self.bundle, bundle=self.bundle)
  File "/home/pi/photobooth/.venv/lib/python3.4/site-packages/pip/req.py", line 1146, in prepare_files
    req_to_install.run_egg_info()
  File "/home/pi/photobooth/.venv/lib/python3.4/site-packages/pip/req.py", line 326, in run_egg_info
    command_desc='python setup.py egg_info')
  File "/home/pi/photobooth/.venv/lib/python3.4/site-packages/pip/util.py", line 716, in call_subprocess
    % (command_desc, proc.returncode, cwd))
pip.exceptions.InstallationError: Command python setup.py egg_info failed with error code 1 in /home/pi/photobooth

```

**Additional context**
N/A

reuterbal, 2018-06-16:
Hi,

I'm sorry that it is not working for you. Thanks for the detailed bug report, though!

However, I can't reproduce the issue on my Raspi 3 with Stretch.
You seem to be using older package versions or maybe are still on Jessie and not Stretch: 
- Raspbian Stretch ships with Python 3.5.3, you seem to be using 3.4.2.
- In the beginning of your video, when calling `apt update`, the distribution specified for your sources is `jessie`. Thus, if you have stretch installed, it seems you replaced your `/etc/apt/sources.list` by an older version.

Could you try again with up-to-date package versions?

wikijm, 2018-06-17:
Hi @reuterbal,

My bad, you say in [Hardware requirements section](https://github.com/reuterbal/photobooth#hardware-requirements) that Python 3.5+ is necessary.

**The solution for this issue is:** _The execution context does not meet the software requirements of the project._

As always, thanks for your time 😃

closed 2018-06-17 by wikijm