reuterbal/photobooth #14: Successfully installed & working

xtremecreative, 2018-06-20:
I am just hobby programmer and trying to build photobooth while surfing accross github I found this and installed. Before this I have tried many but none them worked properly, So very happy and the customisation level is too good I liked that very much. 

I Just want to say that if a developer can include the function for social media sharing or sharing via e-mail that can be the great feature for everyone.

 - Device Raspberry Pi 3B+
 - Camera Picamera
 - GPIO: No
   - (Touch)Screen

**Software (please complete the following information):**
 - OS Raspbian Stretch
 - Python version [e.g. 3

reuterbal, 2018-06-20:
Thank you for your feedback, it is appreciated! Some kind of sharing functionality is on my radar, however I haven't looked into different services and how to add them. 
A restriction is the need for a reliable network connection (e.g., free wi-fi), which is often (at least here in Germany) not the case on-site.

botyas, 2018-09-16:
Apart from adding the option to share on social networks, you could add a gallery so that guests could download the images from their mobile phones via bluetooth or directly by entering a website and downloading.

in my photobooth I have added an 841 tplink router with LEDE, in order to have a wifi network to control the raspberry from there.

I've also seen that you can perform a hotspot with the same raspberry but the router option seemed faster and more comfortable.

reuterbal, 2018-09-21:
Hi @botyas ,

I had the same idea as well :-) I want to integrate it somehow with the photobooth application but not interweave it too much and. At the same time, I want to avoid the hassle of pulling in a big stack of dependencies for a simple webserver and gallery. If someone knows of a simple webserver+gallery option for Python 3, please share it here!

botyas, 2018-09-21:
Yo he estado mirando esta galería pero mi nivel no da para integrarlo en tu photobooth.

https://github.com/dimsemenov/PhotoSwipe

botyas, 2018-10-18:
I'm testing with piwigo, I've already managed to see the galleries with the photos I have in a directory, but I can not get it to update automatically I've added the options to share on social networks or WhatsApp from piwigo.

reuterbal, 2019-06-21:
There is now a simple WebDAV upload functionality in the development branch.

alopezbu, 2019-07-14:
@reuterbal i have found this repository that may be interesting for the share functionalities:
https://github.com/bpatrik/pigallery2

alopezbu, 2019-07-16:
Hi @reuterbal ,  I have been working these two days with the pigallery2 repository that I posted on the previous comment and finally I have achieved to create a web that you can access without internet to share the pictures taken by the photobooth almost at the moment (and my knowledge of programming are very low, almost none) so I really recommend you to use asi I have read that the sharing funcionalities that you were looking were more or same the same as mines.

Thank you very much for your work, and I hope the pigallery repository works for you too!!

reuterbal, 2019-07-17:
Thanks for that, I'll look into it.