reuterbal/photobooth #15: Success!

edconeau, 2018-06-21:
Sorry this isn't the correct format but I am not done with the whole photobooth but wanted to share my experience while it is fresh in my head.

I had success for rasbian 4.14 and python 3.5 on the pi zero w! 

"
virtualenv -p python3 --system-site-packages .venv
"
- gave an error because it looks like this is the default now. 

I had to set the venv to python 3.5, otherwise it kept running 2.7 in the venv.
"
virtualenv -p /home/username/opt/python-3.5.2/bin/python my_project
"

The two times you referenced 
"
source .venv/bin/activate"
There is a period before .venv
I had to use 
"
source venv/bin/activate"

I had a lot of trouble with qt5 so I ended up using 
http://www.tal.org/tutorials/building-qt-58-raspberry-pi-debian-stretch
To install it

I am new, maybe these are obvious to some people or maybe straight up wrong, but it totally worked for me!

Thank you so much for all this work that went into this.
-ed

wikijm, 2018-06-22:
Hi @edconeau,

The Pi Zero W is compatible? Awesome!
And thanks for sharing your solutions about python and qt5.

Could you please describe the equipment you used?

reuterbal, 2018-07-03:
Hi @edconeau 

have you been able to sort out your issues?
I'm closing this due to inactivity for now, feel free to comment again if new issues arise - I will reopen it then.

closed 2018-07-03 by reuterbal