reuterbal/photobooth #16: software freezing issue

axel8417, 2018-06-24:
Hello,

First of all I want to say thank you, the new version of the photobooth is awesome! V1 was great, but this is even better!

But unforunatelly I have got a problem with the buzzer/trigger of my photobooth. It freezes the software at the "get ready ..." screen.

I use a raspberry pi 3b+ with the latest version of raspbian
python 3.5.3
Canon EOS 500D

My buzzer has a 5m long cable which connects it to the raspberry pi 3b+ I use. The error occured every time I wanted to take a picture, and the software stuck at the "get ready..." screen. But I could quit the software with esc, so it didn't freez up totally.
In Version 0.1 I had some problems with this buzzer triggering the photobooth randomly and the photobooth taking two pictures, so I put in a resistor which improved it a little. 
After I removed the resistor I could take at 0-3 pictures before the error occured. Removing the button and just connecting the pins with a short cable workd fine.

**Photobooth log:**
```

2018-06-24 08:43:04,409 - root - INFO - Initializing photobooth
2018-06-24 08:43:04,410 - root - INFO - Photobooth version: 0.3.dev16+g289928f
2018-06-24 08:43:04,413 - root - INFO - Reading config file "/home/pi/photobooth/photobooth/defaults.cfg"
2018-06-24 08:43:04,418 - root - INFO - Reading config file "photobooth.cfg"
2018-06-24 08:43:08,420 - root - INFO - Number of last existing file: 10
2018-06-24 08:43:08,424 - root - INFO - Saving assembled pictures as "2018-06-24/photoboothXXXXX.jpg"
2018-06-24 08:43:08,424 - root - INFO - Using python-gphoto2 bindings
2018-06-24 08:43:08,590 - gphoto2 - WARNING - (b'ptp_list_folder_eos [library.c:7342]') b'storage 0xffffffff, but handle 0x00000000?'
2018-06-24 08:43:08,595 - gphoto2 - WARNING - (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x9109 receiving resp failed: PTP General Error (0x2002)'
2018-06-24 08:43:08,596 - gphoto2 - WARNING - (b'ptp_list_folder_eos [library.c:7358]') b"'ptp_canon_eos_getobjectinfoex ( params, storageids.Storage[k], handle ? handle : 0xffffffff, 0x100000, &tmp, &nroftmp)' failed: PTP General Error (0x2002)"
2018-06-24 08:43:08,662 - root - INFO - Camera summary: Manufacturer: Canon Inc.
Model: Canon EOS 600D
  Version: 3-1.0.2
  Serial Number: 80b7725b79ad4d78bc9e1f7865dc46f9
Vendor Extension ID: 0xb (2.0)

Capture Formats: JPEG
Display Formats: Association/Directory, Script, DPOF, MS AVI, MS Wave, JPEG, CRW, Unknown(b103), Unknown(bf02), Defined Type, Unknown(b104), Unknown(b105)

Device Capabilities:
	File Download, File Deletion, File Upload
	No Image Capture, No Open Capture, Canon EOS Capture, Canon EOS Shutter Button

Storage Devices Summary:

Device Property Summary:
Property 0xd402:(read only) (type=0xffff) 'Canon EOS 600D'
Property 0xd407:(read only) (type=0x6) 1
Property 0xd406:(readwrite) (type=0xffff) 'Unknown Initiator'
Property 0xd303:(read only) (type=0x2) 1
Battery Level(0x5001):(read only) (type=0x2) Enumeration [100,0,75,0,50] value: 100% (100)

2018-06-24 08:43:08,797 - gphoto2 - WARNING - (b'gp_widget_count_choices [gphoto2-widget.c:784]') b"Invalid parameters: '(widget->type == GP_WIDGET_RADIO) || (widget->type == GP_WIDGET_MENU)' is NULL/FALSE."
2018-06-24 08:43:08,798 - gphoto2 - WARNING - (b'gp_widget_count_choices [gphoto2-widget.c:784]') b"Invalid parameters: '(widget->type == GP_WIDGET_RADIO) || (widget->type == GP_WIDGET_MENU)' is NULL/FALSE."
2018-06-24 08:43:08,799 - gphoto2 - WARNING - (b'gp_widget_count_choices [gphoto2-widget.c:784]') b"Invalid parameters: '(widget->type == GP_WIDGET_RADIO) || (widget->type == GP_WIDGET_MENU)' is NULL/FALSE."
2018-06-24 08:43:08,799 - gphoto2 - WARNING - (b'gp_widget_count_choices [gphoto2-widget.c:784]') b"Invalid parameters: '(widget->type == GP_WIDGET_RADIO) || (widget->type == GP_WIDGET_MENU)' is NULL/FALSE."
2018-06-24 08:43:08,800 - gphoto2 - WARNING - (b'gp_widget_count_choices [gphoto2-widget.c:784]') b"Invalid parameters: '(widget->type == GP_WIDGET_RADIO) || (widget->type == GP_WIDGET_MENU)' is NULL/FALSE."
2018-06-24 08:43:08,800 - gphoto2 - WARNING - (b'gp_widget_count_choices [gphoto2-widget.c:784]') b"Invalid parameters: '(widget->type == GP_WIDGET_RADIO) || (widget->type == GP_WIDGET_MENU)' is NULL/FALSE."
2018-06-24 08:43:08,801 - gphoto2 - WARNING - (b'gp_widget_count_choices [gphoto2-widget.c:784]') b"Invalid parameters: '(widget->type == GP_WIDGET_RADIO) || (widget->type == GP_WIDGET_MENU)' is NULL/FALSE."
2018-06-24 08:43:08,801 - gphoto2 - WARNING - (b'gp_widget_count_choices [gphoto2-widget.c:784]') b"Invalid parameters: '(widget->type == GP_WIDGET_RADIO) || (widget->type == GP_WIDGET_MENU)' is NULL/FALSE."
2018-06-24 08:43:08,802 - gphoto2 - WARNING - (b'gp_widget_count_choices [gphoto2-widget.c:784]') b"Invalid parameters: '(widget->type == GP_WIDGET_RADIO) || (widget->type == GP_WIDGET_MENU)' is NULL/FALSE."
2018-06-24 08:43:08,803 - gphoto2 - WARNING - (b'gp_widget_count_choices [gphoto2-widget.c:784]') b"Invalid parameters: '(widget->type == GP_WIDGET_RADIO) || (widget->type == GP_WIDGET_MENU)' is NULL/FALSE."
2018-06-24 08:43:08,803 - gphoto2 - WARNING - (b'gp_widget_count_choices [gphoto2-widget.c:784]') b"Invalid parameters: '(widget->type == GP_WIDGET_RADIO) || (widget->type == GP_WIDGET_MENU)' is NULL/FALSE."
2018-06-24 08:43:08,804 - gphoto2 - WARNING - (b'gp_widget_count_choices [gphoto2-widget.c:784]') b"Invalid parameters: '(widget->type == GP_WIDGET_RADIO) || (widget->type == GP_WIDGET_MENU)' is NULL/FALSE."
2018-06-24 08:43:08,804 - gphoto2 - WARNING - (b'gp_widget_count_choices [gphoto2-widget.c:784]') b"Invalid parameters: '(widget->type == GP_WIDGET_RADIO) || (widget->type == GP_WIDGET_MENU)' is NULL/FALSE."
2018-06-24 08:43:08,805 - gphoto2 - WARNING - (b'gp_widget_count_choices [gphoto2-widget.c:784]') b"Invalid parameters: '(widget->type == GP_WIDGET_RADIO) || (widget->type == GP_WIDGET_MENU)' is NULL/FALSE."
2018-06-24 08:43:08,805 - gphoto2 - WARNING - (b'gp_widget_count_choices [gphoto2-widget.c:784]') b"Invalid parameters: '(widget->type == GP_WIDGET_RADIO) || (widget->type == GP_WIDGET_MENU)' is NULL/FALSE."
2018-06-24 08:43:08,806 - gphoto2 - WARNING - (b'gp_widget_count_choices [gphoto2-widget.c:784]') b"Invalid parameters: '(widget->type == GP_WIDGET_RADIO) || (widget->type == GP_WIDGET_MENU)' is NULL/FALSE."
2018-06-24 08:43:08,806 - gphoto2 - WARNING - (b'gp_widget_count_choices [gphoto2-widget.c:784]') b"Invalid parameters: '(widget->type == GP_WIDGET_RADIO) || (widget->type == GP_WIDGET_MENU)' is NULL/FALSE."
2018-06-24 08:43:08,807 - gphoto2 - WARNING - (b'gp_widget_count_choices [gphoto2-widget.c:784]') b"Invalid parameters: '(widget->type == GP_WIDGET_RADIO) || (widget->type == GP_WIDGET_MENU)' is NULL/FALSE."
2018-06-24 08:43:08,807 - gphoto2 - WARNING - (b'gp_widget_count_choices [gphoto2-widget.c:784]') b"Invalid parameters: '(widget->type == GP_WIDGET_RADIO) || (widget->type == GP_WIDGET_MENU)' is NULL/FALSE."
2018-06-24 08:43:08,808 - gphoto2 - WARNING - (b'gp_widget_count_choices [gphoto2-widget.c:784]') b"Invalid parameters: '(widget->type == GP_WIDGET_RADIO) || (widget->type == GP_WIDGET_MENU)' is NULL/FALSE."
2018-06-24 08:43:08,809 - gphoto2 - WARNING - (b'gp_widget_count_choices [gphoto2-widget.c:784]') b"Invalid parameters: '(widget->type == GP_WIDGET_RADIO) || (widget->type == GP_WIDGET_MENU)' is NULL/FALSE."
[... 103 lines cut ...]
  File "/usr/lib/python3.5/multiprocessing/process.py", line 249, in _bootstrap
    self.run()
  File "/home/pi/photobooth/photobooth/main.py", line 79, in run
    status_code = self.run_camera()
  File "/home/pi/photobooth/photobooth/main.py", line 57, in run_camera
    return photobooth.run()
  File "/home/pi/photobooth/photobooth/Photobooth.py", line 147, in run
    self.trigger()
  File "/home/pi/photobooth/photobooth/Photobooth.py", line 222, in trigger
    self.recvAck()
  File "/home/pi/photobooth/photobooth/Photobooth.py", line 112, in recvAck
    if self.recvEvent(events) != 0:
  File "/home/pi/photobooth/photobooth/Photobooth.py", line 104, in recvEvent
    raise ValueError('Unknown event received', str(event))
ValueError: ('Unknown event received', 'triggered')

```

Maybe the current which arrives at the pin while triggering is to low/not long enough? If I press the button till it captures the image, the chance that it works is higher.

I also got some other warnings/errors which would be nice to fix, but it seems like they don't stop the photobooth working.

Thank you in advance!

-Axel

wikijm, 2018-06-24:
Hi @axel8417,

2 things to take care first:
- You leave your camera serial number in logs, which can be considerate as a sensitive data. You should think of delete it to your issue
- You are talking about 500D model when I see in your logs a reference of 600D model. I assume that logs are more accurate 👼 

For your main issue, could you please try to trigger `gphoto2` only, with your buzzer ?
This way, we will be able to see which errors are coming from gphoto2 and which are coming from `reuterbal/photobooth`

reuterbal, 2018-06-25:
Hi @axel8417,

first of all: thanks for trying out the software and taking the time to report back issues!

Secondly, your Logfile output and the console output don't seem to match each other. There is a difference of 7 seconds in the time stamps. However, I think I know the reason of the issues:

As everything works fine when you use short cables, this looks like an electrical engineering issue and a number of electrical effects interplaying here:
1. Make sure that your button closes the GPIO pin 23 and GND (**not** +3.3V).
2. If you use such long wires, keep in mind that they have a non-neglectable electrical resistance, especially for such low voltages (3.3 V for GPIO) - thus, use wires with a large cross-sectional area to reduce the resistance. If the resistance of the wire is high, this leads to a considerable voltage drop and might affect the PIs ability to detect a button press (because the actual voltage drop might be of the same order than the potential difference). Read about this e.g. [here](https://en.wikipedia.org/wiki/Voltage_drop) and [here](https://en.wikipedia.org/wiki/Electrical_resistance_and_conductance).
3. Depending on your button type, it can generate some bouncing in the electrical signal (meaning, there is not a clean voltage drop pulling the GPIO pin to GND but instead some oscillations in the signal). For most cases this is sufficiently counteracted by the internal pullup resistor of the Pi, however, for such long wiring this might not suffice and you might need an external resistor in series.

The above leads to detection of a second button press, while the software did not yet start handling the first. In the trigger()-function the detection of a button press is disabled to avoid such problems, however the short time frame between the first press and the disabling seems to be too long to effectively avoid these problems.

I changed the order in the trigger()-function to disable the trigger button first and initiate the GUI output afterwards. This might help reduce the problems, however there is a better way to fix this which I might do in the next couple of days (I want to change some things with respect to GPIO anyways).

You should do the following:
1. Use wires with a larger cross-sectional area to reduce the resistance. This should help detecting the button press in the first place.
2. Try out the latest version to see if it helped to avoid the freezing.

axel8417, 2018-06-26:
Hi,
I think too that it is an electrical issue. I have tested it a little today:

1. Sure
2. I am using a normal 230V cable with ~ 0,5mm² cross-sectual area, which should be enough?
3. I removed the button and just connected the two wires

I also tested the latest version.
Unfortunatelly it still does mit work.
I will try a 1,5mm² cable on friday.
I know that there is a setting which changes the sensitivity of the gpio pin. Maybe playing with it would also work?
Should I try powering the button with an external power supply to avoid the voltage drop?
I think it is strange that in V2 the photobooth does not trigger correct (voltage drop, because of high resistance), but in V1 it was triggering randomly (maybe electrical induction -> low resistance after putting in an additional resistance it improved?). Pleased correct me If I am wrong. I am Just wondering why both Versions do not work correctly.

reuterbal, 2018-07-30:
Closing this for now due to inactivity. Open a new issue if you run into problems again.

closed 2018-07-30 by reuterbal