reuterbal/photobooth #17: [ENHANCEMENTS] and success story

michi1g, 2018-06-27:
Hi  @reuterbal,
just tested your software and I'm very impressed! Thanks for all the effort you put in this! All basic features are working as expected.
While using the software I had some ideas for enhancements:

- there are no pictures saved in the camera. This would be a great backup if the pi sd-card fails
- startup parameter to instantly start the photobox with the given config (plug and play)
- possibility to keep the last recorded picture on screen until a button is pressed (GPIO or touchscreen)

My System:
- Device: Raspberry Pi 3B+
- Camera: Canone EOS 600D/70D
- GPIO: no
- Touchscreen: yes 10.1" (however unused)

 **Additional equipment**
- Flash (Godox)

**Software (please complete the following information):**
 - OS: Raspbian Strech
 - Python version: 3.7

**Problems:**
- App crashes when Autofocus doesn't succeed
- If exiting the "front-end" I can't start the "front-end" again. I have to restart the whole software

**Pictures:**
I will add some pictures, when the enclosure redesign is finished.

If you would like to get in touch, just drop by FAU FabLab Tuesday evening 18-20Uhr

reuterbal, 2018-06-28:
Hi @michi1g 

thanks for the feedback!

**Enhancements**:
1. I'll see if it is easily possible to store the pictures in the camera. Generally, it should even be possible to transfer back the assembled picture to the camera storage. This would be a job for a postprocessing module, however, it's low on my agenda.
2. The startup parameter is a good idea. However, on site I usually have to modify date and time first and try it out myself, thus the single button press to get it running doesn't bother me to much... But it's probably a good idea for setups without a touch screen.
3. I want to make the "review-phase" more pretty anyways. I'll probably add a button there and simply change the review time to a very large value (e.g., 5 mins).

**Problems:**
1. Can you provide some logs for the crashes? Usually it should display an error and let you try again, however, I haven't tested that recently.
2. What do you mean by "exiting the fornt-end"? Closing the window? This should quit the entire app, thus you must restart it, of course.

I'll be showing up at FabLab soon anyways to add a back to my box :)

michi1g, 2018-06-29:
Hi @reuterbal!
**Enhancements:**

1. Ok, I thought it would be deleted on purpose. I would just keep all original files taken with the camera in camera and not transfer processed pictures back to the camera.
2. For me it would be really great as everything would start without having to interfere with human error or anything else. Just plug in the power cord and everythin else works out of the box.
3. That would be great. You could add the print button there too.

**Problems:**
Just retested the focus issue and can successfully reproduce this error. Log:
> 2018-06-29 09:59:02,145 - root - INFO - Countdown with preview activated
> 2018-06-29 09:59:02,146 - root - INFO - GPIO disabled
> 2018-06-29 09:59:03,081 - root - INFO - Photobooth triggered
> 2018-06-29 09:59:04,123 - gphoto2 - WARNING - (b'ptp_usb_getresp [usb.c:465]') $
> 2018-06-29 09:59:04,181 - gphoto2 - WARNING - (b'ptp_usb_getresp [usb.c:465]') $
> 2018-06-29 09:59:04,288 - gphoto2 - WARNING - (b'ptp_usb_getresp [usb.c:465]') $
> 2018-06-29 09:59:04,447 - gphoto2 - WARNING - (b'ptp_usb_getresp [usb.c:465]') $
> 2018-06-29 09:59:16,933 - root - INFO - Saving picture as 2018-06-29/photobooth$
> 2018-06-29 09:59:30,616 - root - INFO - Photobooth triggered
> 2018-06-29 09:59:31,660 - gphoto2 - WARNING - (b'ptp_usb_getresp [usb.c:465]') $
> 2018-06-29 09:59:31,716 - gphoto2 - WARNING - (b'ptp_usb_getresp [usb.c:465]') $
> 2018-06-29 09:59:31,821 - gphoto2 - WARNING - (b'ptp_usb_getresp [usb.c:465]') $
> 2018-06-29 09:59:31,981 - gphoto2 - WARNING - (b'ptp_usb_getresp [usb.c:465]') $
> 2018-06-29 09:59:39,187 - gphoto2 - WARNING - (b'gp_context_error') b'Canon EOS$
> 2018-06-29 09:59:39,190 - gphoto2 - WARNING - (b'gp_camera_capture [gphoto2-cam$
> Process CameraProcess-1:
> Traceback (most recent call last):
>   File "/usr/lib/python3.5/multiprocessing/process.py", line 249, in _bootstrap
>     self.run()
>   File "/home/pi/photobooth/photobooth/main.py", line 79, in run
>     status_code = self.run_camera()
>   File "/home/pi/photobooth/photobooth/main.py", line 57, in run_camera
>     return photobooth.run()
>   File "/home/pi/photobooth/photobooth/Photobooth.py", line 147, in run
>     self.trigger()
>   File "/home/pi/photobooth/photobooth/Photobooth.py", line 224, in trigger
>     pics = self.capturePictures()
>   File "/home/pi/photobooth/photobooth/Photobooth.py", line 197, in capturePict$
>     for i in range(self._pic_dims.totalNumPictures)]
> File "/home/pi/photobooth/photobooth/Photobooth.py", line 190, in captureSing$
>     picture = self._cap.getPicture()
>   File "/home/pi/photobooth/photobooth/camera/CameraGphoto2.py", line 129, in g$
>     file_path = self._cap.capture(gp.GP_CAPTURE_IMAGE)
> gphoto2.GPhoto2Error: [-1] Unspecified error

When clicking "Start photobooth" all is starting up as expected. However, when going back to startup menu (pressing esc) and clicking on "Start photobooth" again I get the following errors:

> 2018-06-29 10:05:54,252 - root - INFO - Countdown with preview activated
> 2018-06-29 10:05:54,254 - root - INFO - GPIO disabled
> 2018-06-29 10:05:55,601 - root - INFO - Teardown of camera requested
> 2018-06-29 10:05:55,602 - root - INFO - Teardown of camera
> 2018-06-29 10:05:55,618 - root - INFO - Camera exited with status code 123
> 2018-06-29 10:05:57,433 - root - INFO - Number of last existing file: 1
> 2018-06-29 10:05:57,435 - root - INFO - Saving assembled pictures as "2018-06-29/photoboothXXXXX.jpg"
> 2018-06-29 10:05:57,436 - root - INFO - Using python-gphoto2 bindings
> 2018-06-29 10:05:57,673 - gphoto2 - WARNING - (b'gp_libusb1_open [libusb1.c:421]') b"'libusb_claim_interface (port->pl->dh, port->settings.usb.interface)' failed: Resource busy (-6)"
> 2018-06-29 10:05:57,674 - gphoto2 - WARNING - (b'gp_libusb1_open [libusb1.c:421]') b"'libusb_claim_interface (port->pl->dh, port->settings.usb.interface)' failed: Resource busy (-6)"
> 2018-06-29 10:05:57,675 - gphoto2 - WARNING - (b'gp_port_set_error [gphoto2-port.c:1190]') b'Could not claim interface 0 (Success). Make sure no other program (gvfs-gphoto2-volume-monitor) or kernel module $
> 2018-06-29 10:05:57,676 - gphoto2 - WARNING - (b'gp_port_set_error [gphoto2-port.c:1190]') b'Could not claim interface 0 (Success). Make sure no other program (gvfs-gphoto2-volume-monitor) or kernel module $
> Process CameraProcess-1:
> Traceback (most recent call last):
>   File "/usr/lib/python3.5/multiprocessing/process.py", line 249, in _bootstrap
>     self.run()
>   File "/home/pi/photobooth/photobooth/main.py", line 79, in run
>     status_code = self.run_camera()
>   File "/home/pi/photobooth/photobooth/main.py", line 56, in run_camera
>     self.cfg, cap, self.conn, self.worker_queue)
>   File "/home/pi/photobooth/photobooth/Photobooth.py", line 47, in __init__
>     self.initCamera(config, camera())
>   File "/home/pi/photobooth/photobooth/camera/CameraGphoto2.py", line 42, in __init__
>     self._setupCamera()
>   File "/home/pi/photobooth/photobooth/camera/CameraGphoto2.py", line 58, in _setupCamera
>     self._cap.init(self._ctxt)
> gphoto2.GPhoto2Error: [-53] Could not claim the USB device
> 2018-06-29 10:05:57,688 - gphoto2 - WARNING - (b'gp_port_set_error [gphoto2-port.c:1190]') b'Could not release interface 0 (Success).'
> 2018-06-29 10:05:57,690 - gphoto2 - WARNING - (b'gp_port_set_error [gphoto2-port.c:1190]') b'Could not release interface 0 (Success).'

Hope to meet you on Tuesday evening at our FabLab ;-)

wikijm, 2018-07-06:
Hello,

How was your last FabLab?

Any pictures to share?

reuterbal, 2018-07-08:
Sadly, I haven't had time yet to go.

wikijm, 2018-07-09:
OK @reuterbal 
How about you @michi1g ?

michi1g, 2018-07-09:
I'm there every tuesday. Will post some pictures in the next days. However, I'm going to rebuild it as soon as the new Laser cutter will be available (mid August) as my case is a little bit too small.

reuterbal, 2018-07-09:
Hopefully I find some time next week or the week after - then the summer term is over and I'm finally less occupied with teaching duties... 
@michi1g Are you there on Tuesdays after the semester as well?

michi1g, 2018-07-09:
Yes, every tuesday from 18-20 o'clock

reuterbal, 2018-07-17:
I incorporated the changes (startup parameter, postprocessing screen before returning to idle) into the development branch. After testing them on my Raspi they will be merged into master.

At the moment I don't have time to look into storing the pictures on camera. It would require changes to `getPicture` in [CameraGphoto2.py](https://github.com/reuterbal/photobooth/blob/development/photobooth/camera/CameraGphoto2.py) or [CameraGphoto2Cffi.py](https://github.com/reuterbal/photobooth/blob/development/photobooth/camera/CameraGphoto2Cffi.py) where one must trigger the capture, wait for completion, obtain the file name and then download the file. This could probably be found in the gphoto2 sources. Feel free to create a PR ;-)

However, I added a functionality to keep the single shots additionally to the assembled pictures. This doesn't fulfil the backup idea but at least gives some resilience against failures in the assembly procedure.

commit d9935a78 (2018-07-05): Add argument '--run' to omit the welcome screen, fixes #17
closed 2018-07-17 by reuterbal