reuterbal/photobooth #18: QXcbConnection: Could not connect to display

doc2005, 2018-07-03:
**Describe the bug**
I did the installtion as mentioned an when I start "python photobooth" I got the followig output:

2018-07-03 22:07:38,365 - root - INFO - Initializing photobooth
2018-07-03 22:07:38,367 - root - INFO - Photobooth version: 0.2
2018-07-03 22:07:38,371 - root - INFO - Reading config file "/home/pi/photobooth/photobooth/defaults.cfg"
2018-07-03 22:07:38,377 - root - INFO - Reading config file "photobooth.cfg"
QXcbConnection: Could not connect to display
2018-07-03 22:07:41,121 - root - INFO - Exiting photobooth with status code -6

**Hardware (please complete the following information):**
 - Device Raspberry Pi 2
 - Camera NIKON D50
 - GPIO: same behavior ether

**Software (please complete the following information):**
 - OS Raspbian Stretch fully updated
 - Python version 3.5.3

**Photobooth log:**
```
2018-07-03 22:07:38,365 - root - INFO - Initializing photobooth
2018-07-03 22:07:38,367 - root - INFO - Photobooth version: 0.2
2018-07-03 22:07:38,371 - root - INFO - Reading config file "/home/pi/photobooth/photobooth/defaults.cfg"
2018-07-03 22:07:38,377 - root - INFO - Reading config file "photobooth.cfg"
2018-07-03 22:07:41,121 - root - INFO - Exiting photobooth with status code -6
```

Same behavior with version 0.2

reuterbal, 2018-07-03:
Hi @doc2005 

the error usually points to the lack of an X server. The following possibilities come to mind:
1. You don't have any screen attached.
2. You installed Raspbian lite.
3. You started the photobooth app from the terminal without starting an X session before that.
4. You tried to run the photobooth from a root shell (i.e., called `su` or `sudo -i` before running the photobooth)
5. You tried starting the photobooth app remotely (via SSH) without X forwarding (`ssh -X`)

Could any of the above be the case?

doc2005, 2018-07-04:
ohh, this was embarassing!
I hit case #3... I had a SSH connection to easier edit the stuff.
now it's working.
many thanks for these hints.

best regards
Daniel

closed 2018-07-04 by doc2005