reuterbal/photobooth #19: Issue with screen size

jul1792, 2018-07-04:
Hi reuterbal,
I really like your programme, especially because of the GUI. But this is also my issue. 
This is my setup:

- Raspberry Pi 3
- PiCamera V2
- Canon Selphy CP800
- [This 7" TFT LC Display ](https://www.pollin.de/p/7-17-78-cm-display-set-ls-7-hdmi-dvi-vga-cvbs-120942)
- [This Red Button](https://www.amazon.de/gp/product/B01LZMANZ7/ref=oh_aui_detailpage_o02_s00?ie=UTF8&psc=1) 
- [These Jumper Wires and Breadboard ](https://www.amazon.de/gp/product/B071RG9MFT/ref=oh_aui_detailpage_o01_s01?ie=UTF8&psc=1)

The installation went fine but I have one big issue: The programme isn't fully displayed on my display.
[image]

So, when I want to change the settings, I can't save them. I tried to not change the settings and just start the photobooth but then it loaded forever and didn't show any preview or similar. I think it is because I couldn't select the PiCamera in the settings.

Is there any way to see the full programme? 

Also, I have some more questions:

- is there a way to start the programme on boot and full screen? So I just have to plug in everything and after the boot the fun can start right away.
- are the photos saved anywhere? Could the programme save them on a USB stick or HDD?

Thank you very much in advance! I really would like it to work :)

reuterbal, 2018-07-04:
Dear @jul1792 

thanks for trying out my photobooth!

* To overcome the resolution problems, copy the file `photobooth/defaults.cfg` into the directory above and call it `photobooth.cfg`. This is the file with the user settings and gets created automatically once you click 'Save' in the settings dialogue for the first time.
In that file, change the line with the key `fullscreen` to `fullscreen = True` in the `[Gui]` section.
* Photos are always saved, regardless whether the user chooses to print them or not. By default, they are saved in a subfolder `YYYY-mm-dd`, however, you can change that in the settings dialogue to any location you like. If you plug in a USB drive, you can select that as a destination as well.

Best,
reuterbal

Cleming, 2018-07-04:
Dear @reuterbal 
thank you for your job !
is there a way to start the program on boot like previous versions ?

Thanks from France !

jul1792, 2018-07-04:
Thanks a lot @reuterbal! I can see the full screen and got it working except for the printer. 
I installed my Canon Selphy CP800 via Cups and made it the default printer via the GUI. But when your Programme asked me if I wanted to print the pictures, I clicked Yes. Then it said the picture got sent to the printer. But then nothing happened.
In the settings I can only select the module „PyQt5“. I connected the printer via USB. I already could print a picture via command line, so the printer is working fine.
Do you have any suggestions how I can get it printing via your programme?
And is it normal that you have to select yes or no for printing? Could there be the option that it just prints every picture that is taken?
Again thanks a lot! Really great work you have done there!

jul1792, 2018-07-05:
By saying "try the code version now" do you mean to reinstall the photobooth by scratch or to just do all updates and upgrades? (I am sorry, I am not very familiar with Linux distributions and how they work). 

Unfortunately I don't have a touch screen and when I am building my photobooth into a nice case, there is no possibility to select "Yes" or "No" at the printing question with a keyboard. Can you think of any solution here, e.g. adding some more buttons?

reuterbal, 2018-07-05:
Simply do `git pull` in the photobooth directory. This will get you the updated version.

Regarding printing every picture:
Since this part of the software changed a little bit in the meantime, I'll talk about the development version for this (it will become part of the master probably next week):
The easiest way to achieve this is probably to add 
```python
self._printer.print(picture)
```
in l.103 of this file: https://github.com/reuterbal/photobooth/blob/development/photobooth/gui/GuiPostprocessor.py
and change the postprocess time to 0 (Note that this setting does NOT yet exist in the master branch).

jul1792, 2018-07-05:
Great news: the printing worked!
But it didn‘t work to add the line so the Programme prints the picture automatically. 
Isn‘t there the possibility to add a section in the settings in the GUI, where you can choose between „ask for printing“ and „print automatically“? (I am not familiar with any programming so I am sorry if this is too much to ask). 
Thanks a lot @reuterbal! Without you I wouldn‘t even be able to do this project!

reuterbal, 2018-07-06:
Where did you add the line? As I said, this is only possible in the development branch, which will become part of the master some time next week.
Adding this as a setting requires some changes which I don't have time for at the moment - sorry.

jul1792, 2018-07-08:
I added the line in GuiPostprocess.py (i didn’t find GuiPostprocessor.py anywhere) in line 103. the Script only had 101 lines. 

I will Research in adding a few buttons via GPIO so that the People can select „Print“ or „no print“ with these. Ideally these buttons should immitate „left“ and „Right“ as well as „enter“ and „esc“ on the keyboard. Do you have any expierence with something like this?

jul1792, 2018-07-10:
Ok, I missunderstood this. Can you tell me how I can install the development branch? Is it also as stable as the master branch?

reuterbal, 2018-07-10:
To get the development branch, simply run
```
git checkout -b development origin/development
```
The development branch is not as stable as the master -- this is simply the place where I try out things until they work sufficiently well. However, feel free to try it out and if it doesn't work as expected, return to the master branch by
```
git checkout master
```

jul1792, 2018-07-10:
thanks for the quick response! I will try out the automated printing and will let you know how it worked out.

jul1792, 2018-07-15:
Unfortunately adding the line in the development branch didn‘t do it. The countdown started and then it just stuck at 1. I could exit via esc.

reuterbal, 2018-07-15:
What is the output in the log file?

jul1792, 2018-07-17:
Wow, that sounds great!
I am sorry I couldn't provide the log files, I didn't get to it yet.
I hope I can try out the automated printing tonight, or else I will try it later that week. 
Thanks so much again!

closed 2018-07-17 by reuterbal

reuterbal, 2018-07-17:
Version 0.4 (new master) includes the changes.

reuterbal, 2018-07-18:
Please open a new issue when you encounter new problems. This one was closed already and I might have missed the comment.
This bug was reported a few hours earlier (see #27) and is fixed now. `master` and `development` are at the same code version at the moment - you can switch back to master to avoid problems due to unstable versions.

commit eed71bc2 (2018-07-16): Postprocessing of GUI tasks (e.g. printing) without confirmation optionally possible. Closes #19