reuterbal/photobooth #21: Cannot run the project

FrancoisPerez, 2018-07-06:
Hi,

Thanks for the great work, that really looks impressive !

When I try to execute the project, I always got the following error : 

 File "C:/Users/François Pérez/Desktop/tools/pbooth/photobooth/main.py", line 32, in <module>
    from . import camera, gui
ImportError: cannot import name 'camera'

There must be a misconfiguration somewhere, but I cannot find where.
As you can see, I am runing the project on Windows (but I don't think that really is an issue as I didn't see any OS-specific command
I have tried a few things to fix the import but none of it worked. (simple 'import camera'/putting all camera files to the root ...)
Could you help ?

reuterbal, 2018-07-06:
Hi,
I haven't tried it on Windows yet, thus I can't say for sure that it works. However, I don't see why it shouldn't.

I'd be interested to learn: did the installation go through smoothly? Did you install gphoto2 using MSYS or are you not planning to use gphoto?
In the latter case I would have guessed that the lack of gphoto on Windows platforms would make it impossible to run the `pip install` command successfully (actually, I was thinking to change the gphoto-dependency to be optional in order to make the installation on Windows possible. If it works like this, this might be unnecessary).

One thing that you can try is the following:
1. Copy `photobooth/defaults.cfg` to `photobooth.cfg` in the top folder
2. In the `[Camera]` section change the line
  ```
  module = python-gphoto2
  ``` 
  to
  ```
  module = dummy
  ```

If the import issue is related to gphoto, this should work as it is not using any external dependencies.

FrancoisPerez, 2018-07-09:
Hi, 
Yes, the installation went perfectly fine !
I didn't take too much care about gphoto since I am using OpenCV for this project. Simply editing the CFG file as mentionned should fix any issue regarding OpenCV indeed.

I finally found the fix for my issue. You simply have to load the app as follows : 
[image]

reuterbal, 2018-07-09:
Oh yes, that's right. I should probably update the documentation as this is actually the correct way of loading the app after installing it as a module.
Thanks for pointing that out!

Keep me updated once Windows-specific issues arise. I don't have any Windows machine near me, so I can't really test it, but I guess it would be useful to keep it compatible.