reuterbal/photobooth #22: Screen button don't work

hollesse, 2018-07-06:
**Describe the bug**
If I press the button on the screen nothing happens and it doesn't take any photo.

**To Reproduce**
Steps to reproduce the behavior:
I installed a new raspbian and then followed your instructions. (The description for Disabling the screensaver/screen blanking wasn't clear for me so i skipped that). I connected my Canon eos 600d via usb with the raspberry pi and started the application. 

**Expected behavior**
I expect that it starts to take a photo 

**Hardware (please complete the following information):**
 - Device: Raspberry Pi 3
 - Camera: Canon EOS 600D
 - GPIO: Enabled but not in use

**Software (please complete the following information):**
 - OS: Raspbian Stretch
 - Python version 3.5.3

**Photobooth log:**
```
2018-07-06 18:32:17,050 - root - INFO - Initializing photobooth
2018-07-06 18:32:17,051 - root - INFO - Photobooth version: 0.3.dev23+g2ba7682
2018-07-06 18:32:17,055 - root - INFO - Reading config file "/home/pi/photobooth/photobooth/defaults.cfg"
2018-07-06 18:32:17,060 - root - INFO - Reading config file "photobooth.cfg"
2018-07-06 18:32:18,048 - root - INFO - Using printer ""
2018-07-06 18:32:20,754 - root - INFO - Number of last existing file: 0
2018-07-06 18:32:20,757 - root - INFO - Saving assembled pictures as "2018-07-06/photoboothXXXXX.jpg"
2018-07-06 18:32:20,758 - root - INFO - Using python-gphoto2 bindings
2018-07-06 18:32:20,912 - gphoto2 - WARNING - (b'ptp_list_folder_eos [library.c:7342]') b'storage 0xffffffff, but handle 0x00000000?'
2018-07-06 18:32:20,951 - root - INFO - Camera summary: Manufacturer: Canon Inc.
Model: Canon EOS 600D
  Version: 3-1.0.2
  Serial Number: XXXXX
Vendor Extension ID: 0xb (2.0)

Capture Formats: JPEG
Display Formats: Association/Directory, Script, DPOF, MS AVI, MS Wave, JPEG, CRW, Unknown(b103), Unknown(bf02), Defined Type, Unknown(b104), Unknown(b105)

Device Capabilities:
	File Download, File Deletion, File Upload
	No Image Capture, No Open Capture, Canon EOS Capture, Canon EOS Shutter Button

Storage Devices Summary:
store_00020001:
	StorageDescription: SD
	VolumeLabel: None
	Storage Type: Removable RAM (memory card)
	Filesystemtype: Digital Camera Layout (DCIM)
	Access Capability: Read-Write
	Maximum Capability: 7939817472 (7572 MB)
	Free Space (Bytes): 7789084672 (7428 MB)
	Free Space (Images): -1

Device Property Summary:
Property 0xd402:(read only) (type=0xffff) 'Canon EOS 600D'
Property 0xd407:(read only) (type=0x6) 1
Property 0xd406:(readwrite) (type=0xffff) 'Unknown Initiator'
Property 0xd303:(read only) (type=0x2) 1
Battery Level(0x5001):(read only) (type=0x2) Enumeration [100,0,75,0,50] value: 100% (100)

2018-07-06 18:32:21,050 - gphoto2 - WARNING - (b'gp_widget_count_choices [gphoto2-widget.c:784]') b"Invalid parameters: '(widget->type == GP_WIDGET_RADIO) || (widget->type == GP_WIDGET_MENU)' is NULL/FALSE."
2018-07-06 18:32:21,052 - gphoto2 - WARNING - (b'gp_widget_count_choices [gphoto2-widget.c:784]') b"Invalid parameters: '(widget->type == GP_WIDGET_RADIO) || (widget->type == GP_WIDGET_MENU)' is NULL/FALSE."
2018-07-06 18:32:21,052 - gphoto2 - WARNING - (b'gp_widget_count_choices [gphoto2-widget.c:784]') b"Invalid parameters: '(widget->type == GP_WIDGET_RADIO) || (widget->type == GP_WIDGET_MENU)' is NULL/FALSE."
2018-07-06 18:32:21,053 - gphoto2 - WARNING - (b'gp_widget_count_choices [gphoto2-widget.c:784]') b"Invalid parameters: '(widget->type == GP_WIDGET_RADIO) || (widget->type == GP_WIDGET_MENU)' is NULL/FALSE."
2018-07-06 18:32:21,054 - gphoto2 - WARNING - (b'gp_widget_count_choices [gphoto2-widget.c:784]') b"Invalid parameters: '(widget->type == GP_WIDGET_RADIO) || (widget->type == GP_WIDGET_MENU)' is NULL/FALSE."
2018-07-06 18:32:21,055 - gphoto2 - WARNING - (b'gp_widget_count_choices [gphoto2-widget.c:784]') b"Invalid parameters: '(widget->type == GP_WIDGET_RADIO) || (widget->type == GP_WIDGET_MENU)' is NULL/FALSE."
2018-07-06 18:32:21,056 - gphoto2 - WARNING - (b'gp_widget_count_choices [gphoto2-widget.c:784]') b"Invalid parameters: '(widget->type == GP_WIDGET_RADIO) || (widget->type == GP_WIDGET_MENU)' is NULL/FALSE."
2018-07-06 18:32:21,057 - gphoto2 - WARNING - (b'gp_widget_count_choices [gphoto2-widget.c:784]') b"Invalid parameters: '(widget->type == GP_WIDGET_RADIO) || (widget->type == GP_WIDGET_MENU)' is NULL/FALSE."
2018-07-06 18:32:21,058 - gphoto2 - WARNING - (b'gp_widget_count_choices [gphoto2-widget.c:784]') b"Invalid parameters: '(widget->type == GP_WIDGET_RADIO) || (widget->type == GP_WIDGET_MENU)' is NULL/FALSE."
[... 77 lines cut ...]
    WhiteBalance X B [whitebalancexb]: 0
    Color Space [colorspace]: sRGB (sRGB, AdobeRGB, )
Capture Settings [capturesettings]: 
    Focus Mode [focusmode]: One Shot (One Shot, AI Focus, AI Servo, )
    Canon Auto Exposure Mode [autoexposuremode]: Manual (P, TV, AV, Manual, Bulb, A_DEP, DEP, Custom, Lock, Green, Night Portrait, Sports, Portrait, Landscape, Closeup, Flash Off, )
    Drive Mode [drivemode]: Single (Single, Continuous, Timer 10 sec, Timer 2 sec, Unknown value 0007, )
    Picture Style [picturestyle]: Auto (Standard, Portrait, Landscape, Neutral, Faithful, Monochrome, Auto, User defined 1, User defined 2, User defined 3, )
    Aperture [aperture]: 4 (4, 4.5, 5, 5.6, 6.3, 7.1, 8, 9, 10, 11, 13, 14, 16, 18, 20, 22, 25, )
    Shutter Speed [shutterspeed]: 1/25 (bulb, 30, 25, 20, 15, 13, 10, 8, 6, 5, 4, 3.2, 2.5, 2, 1.6, 1.3, 1, 0.8, 0.6, 0.5, 0.4, 0.3, 1/4, 1/5, 1/6, 1/8, 1/10, 1/13, 1/15, 1/20, 1/25, 1/30, 1/40, 1/50, 1/60, 1/80, 1/100, 1/125, 1/160, 1/200, 1/250, 1/320, 1/400, 1/500, 1/640, 1/800, 1/1000, 1/1250, 1/1600, 1/2000, 1/2500, 1/3200, 1/4000, )
    Metering Mode [meteringmode]: Evaluative (Evaluative, Partial, Spot, Center-weighted average, )
    Bracket Mode [bracketmode]: 0
    Auto Exposure Bracketing [aeb]: off (off, +/- 1/3, +/- 2/3, +/- 1, +/- 1 1/3, +/- 1 2/3, +/- 2, )
Other PTP Device Properties [other]: 
    PTP Property 0xd402 [d402]: Canon EOS 600D
    PTP Property 0xd407 [d407]: 1
    PTP Property 0xd406 [d406]: Unknown Initiator
    PTP Property 0xd303 [d303]: 1
    Battery Level [5001]: 100 (100, 0, 75, 0, 50, )

2018-07-06 18:32:22,699 - root - INFO - Countdown with preview activated
2018-07-06 18:32:22,701 - root - INFO - GPIO enabled (lamp_pin=4, trigger_pin=23, exit_pin=24)
```

**Additional context**
Add any other context about the problem here.

hollesse, 2018-07-09:
@reuterbal sorry but I can't find this hint in the README please check if the hint is really there.

reuterbal, 2018-07-09:
https://github.com/reuterbal/photobooth/blob/master/INSTALL.md#run-photobooth

"You can trigger the countdown via space bar or an external button."

hollesse, 2018-07-09:
ok thanks

reuterbal, 2018-07-17:
Version 0.3 (new master) does now have working buttons.