reuterbal/photobooth #24: No photos taken (PiCamera)

Chefsessel, 2018-07-17:
Hi reuterbal,

your program looks really fine. Thanks for your affords in creating this! Unfortunately I'm struggling to get it to work for me. Maybe you have an advice for me?

 My setup:

- Raspberry Pi 3
- PiCamera V2
- 8" official Pi Touchscreen

I can start photobooth just fine up to the point where it asks "Hit the button" for taking photo. However, upon clicking on the button nothing is happening. A folder with the current date was created in the photobooth folder, but has no content. Not sure if I should get a preview image of the camera because I don't.

The camera itself is working fine, as I tested by taking a still image via bash. In the photobooth settings, I've chosen "picamera"  as camera module.

Including the log as it may contain something helpful

`**Photobooth log:**
018-07-17 12:01:03,715 - root - INFO - Initializing photobooth
2018-07-17 12:01:03,717 - root - INFO - Photobooth version: 0.3.dev23+g2ba7682
2018-07-17 12:01:03,720 - root - INFO - Reading config file "/home/pi/photobooth/photobooth/defaults.cfg"
2018-07-17 12:01:03,725 - root - INFO - Reading config file "photobooth.cfg"
libEGL warning: DRI2: failed to authenticate
qt5ct: using qt5ct plugin
2018-07-17 12:01:07,319 - root - INFO - Using printer "HP_ENVY_Photo_6200_series"
qt5ct: custom style sheet is disabled
2018-07-17 12:01:17,332 - root - INFO - Number of last existing file: 0
2018-07-17 12:01:17,334 - root - INFO - Saving assembled pictures as "2018-07-17/photoboothXXXXX.jpg"
2018-07-17 12:01:17,335 - root - INFO - Using PiCamera
2018-07-17 12:01:19,117 - root - INFO - Countdown with preview activated
2018-07-17 12:01:19,118 - root - INFO - GPIO disabled
2018-07-17 12:03:42,998 - root - INFO - Teardown of camera requested
2018-07-17 12:03:43,000 - root - INFO - Teardown of camera
2018-07-17 12:03:43,002 - root - INFO - Camera exited with status code 123
2018-07-17 12:03:46,269 - root - INFO - Exiting photobooth with status code 0`

Any advice and to what I could try next?

hollesse, 2018-07-17:
have a look to this issue https://github.com/reuterbal/photobooth/issues/22 The button isn't working yet. You have to use spacebar or a gpio button

Chefsessel, 2018-07-17:
aw, thanks a lot! Browsing through the issues helped solving some of the issues already but I obviously missed that one. So sorry! Thanks for answering, hollesse!

reuterbal, 2018-07-17:
The button isn't working yet. The development version includes the buttons.
Thank you @hollesse for pointing to issue #22, another duplicate is #23.

closed 2018-07-17 by reuterbal