reuterbal/photobooth #25: Error on "pip install -e ."

zeph51, 2018-07-17:
Hi

I'm trying to install on a Rpi 3 with raspbian.

Everything work fine until the line "pip install -e ."

Here is a screenshot : 

[image]

My code writing is so rusty :(

What should I do ? 

thanks.

reuterbal, 2018-07-17:
Hi,
It seems to fail while installing gphoto2-cffi. Have you installed all the dependencies before that? 
Check the installation instructions and make sure you did not miss a package.
It requires some additional packages to compile the wheel.

Unfortunately, your screenshots only shows the last lines of the error, thus I can't see at which point it actually fails. Maybe you can provide the entire message?

wikijm, 2018-07-24:
Hi @zeph51,
Could you please follow steps below:

Attach a session recording using https://asciinema.org/
To use it:
```bash
sudo apt-get install -y asciinema && asciinema rec    # Start session recording
commandThatYouWantToExecute                           # Every command that you want to execute
Ctrl+D                                                # Stop recording
y                                                     # YES to upload and get URL to paste here
```

zeph51, 2018-07-27:
Hello :)

Sorry for the delay. I've got sick :(

I've just star all from scratch ! Including raspbian ! It works :)
The app do not find the canon 5d classic but it starts :)

Thanks 👍

closed 2018-07-27 by zeph51

reuterbal, 2018-07-27:
I'm glad that it works now!

Regarding your camera problems: Please try out if gphoto works from the command line. According to their [support list](http://gphoto.org/proj/libgphoto2/support.php) the 5D should be supported, however, only in "normal mode", not in "PTP mode" (whatever that means).

If the camera works on the command line but not in the photobooth app, please open a new issue!