reuterbal/photobooth #26: Make printing selectable via button

ferdinandkeil, 2018-07-17:
First of all I want to thank you for the **awesome work** you have with this software :+1:
A friend and I are in the process of building a photobox and your software was an **huge** time-saver. It already mastered its first wedding with flying colors. Now we want to add a printer (Canon Selphy CP400), so that is where the request is coming from:

It would be awesome if the question whether the picture is printed or not could be answered through a (GPIO) button.

I have tried to understand your code, but I do not see a clear path to implementing this. I guess the most straightforward way would be to trigger PrintState from the Gui process. There you would either trigger a MessageBox or wait for the button till a time-out occurs.

What are your thoughts? I might try to implement this myself, but first I have to get this thing running for the next wedding :grin:

reuterbal, 2018-07-17:
Hi,
thanks for the feedback, I'm glad it proved useful. Please have a look at the updated version (v0.3) which entered master a couple of minutes ago. This version makes this quite easy to implement.

It depends on whether you want to make general 'yes'/'no' buttons, or a dedicated print button.

For the latter and with the new version in place I recommend to add another gpio button in [gpio/__init__.py](https://github.com/reuterbal/photobooth/blob/master/photobooth/gpio/__init__.py) and send a GpioEvent to the GUI process that indicates the desire to print. This should then be handled accordingly in the PostprocessState and ignored otherwise.

For general 'yes'/'no' buttons I would do more or less the same, with respective GpioEvent messages to the GUI process that then trigger GPIO signals themselves. However, this would also require to change the postprocessing screen.

ferdinandkeil, 2018-07-17:
This update should indeed simplify it immensely. And gpiozero should make it much easier to implement blinking for the trigger LED (we had people pressing the lens, so the trigger button can not be obvious enough :man_facepalming:).

I had also added support for an LED flash. Could not get it to work with RPi.GPIO, so I used subprocess and the gpio utility. Will try that again with gpiozero as well. Will keep you updated once it is working :smile_cat:

reuterbal, 2018-07-18:
Haha, yes, the button must be as obvious as possible... have a look at the pastel or dark style - I'm using there a background image with an arrow. People don't get any smarter the later the night...

What kind of LED flash are you using?

Keep me updated!

reuterbal, 2018-07-18:
PS: Thanks for the coffee ;-) It is very much appreciated!

ferdinandkeil, 2018-07-18:
The new themes are awesome :+1: 

Our LED flash is built from a 5 m long LED strip. The strip was mounted on laser-cut aluminium plates. To switch the LEDs we added a small PCB with a power MOSFET and an opto-isolator (to protect the Pi) to the Pi. Using PWM the brightness can be controlled. For now the brightness is lowered during idle and pose state and only raised to full during capture.

[image]

I have just hacked in support for the flash into v0.3, but there are some performance problems. The software PWM of gpiozero does not seem to be up to the task, especially during pose with preview. For now I will keep using the gpio utility.

reuterbal, 2018-07-19:
That's a neat looking photobooth! Funny enough, I was also working on adding a PWM-controled LED strip (however, RGB SMD5050) to serve as a colorful decoy while idle and give some extra lighting when capturing.
Parts arrived just yesterday and I started to put everything together in a breadboard. However, I haven't tested it yet, so I'll see whether performance will be an issue. Maybe I can figure something out.

ferdinandkeil, 2018-07-19:
I am curious to hear how the RGB strip works out. We shied away from using one because we had doubts about getting a correct white balance with it. Also, a pure white LED strip can pump out much more light. Are you doing the PWM control yourself or are you using LEDs with an integrated controller (e.g. WS2811)?

I did some more hacking and have now added the code to my fork. What is working so far:
- LED flash (PWM controlled) is set high during capture and low the rest of the time
- printing has its own state and message, also can be triggered through a button
- start-over can be triggered from a button as well
- of course all buttons have LEDs and they are only lit when there respective button active

The code is not even close to being ready for a PR but it is a start. I will try to clean it up ASAP :bow:

ferdinandkeil, 2018-07-24:
So you were using gpiozero.PWMLED for your tests? I found it especially flickery during processing of the picture.

Achieving a correct white balance is indeed very difficult. Even with an only white LED strip it is almost 
impossible to get completely right.

I added printing as an additional state as I want to display a "Printing" message. Ideally this state would get feedback from the printer and switch to idle once the print is done. If you have any input on how to implement this in a more canonical way I would be happy to hear it.

reuterbal, 2018-07-24:
I used `gpiozero.RGBLED`, which is essentially a class with tuple of three `PWMLED` and some useful helper functions. Flickering was moslty visible in the preview screen itself, which is due to timing discrepancies between the PWM pulses of the LED and frames of the camera.

I understand the intention, however I intended the state machine to mainly cover those states that involve more than one process. My idea was to let each process implement/decide by itself, what it has to do once a certain state is entered.

Of course, adding a printing state is a possible way of implementing the dialog and there are certainly arguments in favor of that. My point of view was the following: 

There are postprocessing tasks that do always happen, and there are postprocessing tasks that require user interaction. An example for the prior is saving the assembled picture to disk. I considered printing as an example for the latter, which could happen among others.

I would use one of the following approaches:

1. Replace/extend the behaviour of `showPostprocess` in `PyQt5Gui` by some 'Wizard'-like series of Gui-Frames, that display a message (e.g., 'Print? Yes/No') that can then be answered one-after-another. Additionally, I would add 'Yes'/'No' buttons to Gpio that simply send `GpioEvent('yes')` or `GpioEvent('no')` to the gui process.
You could even extend that with a `GuiEvent('YesNoQuestion')` that is sent to the gpio process to make the buttons light up at the right time.

2. Add a general state `QuestionState` similar to `ErrorState`, which causes the gui to display the message and the gpio process to light up the buttons. The buttons should behave as in 1 (i.e., send a `GpioEvent` to the gui) and the gui would display a confirmation, possibly wait for completion of the task, before triggering a state change, e.g. to the next `QuestionState` or back to idle.

Regarding the printer feedback: I intentionally left that out. Printing is already a pain in the a** in regular desktop use, I feared too many problems when trying to include it too closely into the photobooth. Especially since different printer models behave differently in similar situations (e.g., paper jam, suppies empty, ...). That is the reason, why I used the printing functionalities of Qt instead of, e.g., talking to cups directly. Another advantage is that the Qt printing features are very generic and platform independent.
Your best option would probably be to use `QPrinterInfo.state()` to find out something about the printer, however, Qt warns
> This state may not always be accurate, depending on the platform, printer driver, or printer itself.
There are also no defined states in the documentation, so I would not recommend to include this in the photobooth.

Another disadvantage when waiting for printer feedback is that minor errors (the most often one is paper empty) cause the photobooth to block, while it could actually continue to take pictures. Jobs will simply queue up and print once paper is refilled. With the relatively low supply levels of Canon Selphy printers, this happens quite often.

reuterbal, 2018-08-21:
As this issue is a bit bloated with other discussions, I'll make #44 the main issue regarding a more general setup of the GPIO module with the ability to add more buttons etc.

closed 2018-08-21 by reuterbal