reuterbal/photobooth #27: Previews: Error_: create_encode() got an unexpected keyword argument 'resolution'

Chefsessel, 2018-07-18:
Hi reuterbal,

Tried out the new master today. Button for taking a photo works now, cool work! Unfortunately ran into another issue. I get an error when trying to take some pics now. (used to work with the old version) I can this error msg when the preview windows should kick in.

Error_: create_encode() got an unexpected keyword argument 'resolution'

It then freezes and I have to kill it to get out. it DOES WORK however if I deactivate preview in the settings.

My setup:

    Raspberry Pi 3
    PiCamera V2 (is set up in the settings)
    8" official Pi Touchscreen

Including the log:

2018-07-18 09:22:14,915 - root - INFO - Photobooth version: 0.3
2018-07-18 09:22:14,919 - root - INFO - Reading config file "/home/pi/photobooth/photobooth/defaults.cfg"
2018-07-18 09:22:14,928 - root - INFO - Reading config file "photobooth.cfg"
2018-07-18 09:22:14,975 - root - INFO - Number of last existing file: 0
2018-07-18 09:22:14,980 - root - INFO - Saving pictures as "2018-07-18/photoboothXXXXX.jpg"
2018-07-18 09:22:14,985 - root - INFO - GPIO disabled
2018-07-18 09:22:14,986 - root - INFO - Number of last existing file: 0
2018-07-18 09:22:14,993 - root - INFO - Saving pictures as "2018-07-18/photobooth_shot_XXXXX.jpg"
libEGL warning: DRI2: failed to authenticate
qt5ct: using qt5ct plugin
2018-07-18 09:22:15,948 - root - INFO - Using printer "HP_ENVY_Photo_6200_series"
qt5ct: custom style sheet is disabled
2018-07-18 09:22:17,284 - root - INFO - Using PiCamera
2018-07-18 09:22:18,309 - root - INFO - Using camera with preview functionality
2018-07-18 09:22:24,357 - root - ERROR - Camera: _create_encoder() got an unexpected keyword argument 'resolution'
Traceback (most recent call last):
  File "/usr/lib/python3.5/runpy.py", line 193, in _run_module_as_main
    "__main__", mod_spec)
  File "/usr/lib/python3.5/runpy.py", line 85, in _run_code
    exec(code, run_globals)
  File "/home/pi/photobooth/photobooth/__main__.py", line 23, in <module>
    sys.exit(photobooth.main(sys.argv))
  File "/home/pi/photobooth/photobooth/main.py", line 188, in main
    status_code = run(argv)
  File "/home/pi/photobooth/photobooth/main.py", line 144, in run
    exit_code = context.handleEvent(event)
  File "/home/pi/photobooth/photobooth/StateMachine.py", line 84, in handleEvent
    self.state.handleEvent(event, self)
  File "/home/pi/photobooth/photobooth/StateMachine.py", line 295, in handleEvent
    raise TypeError('Unknown Event type "{}"'.format(event))
TypeError: Unknown Event type "capture"

Hope this helps for bug hunting! Please keep up the good work!

reuterbal, 2018-07-18:
Hi,
thanks for the feedback.

The problem seems to be caused by the `resolution` setting in l.64 of [`CameraPicamera.py`](https://github.com/reuterbal/photobooth/blob/master/photobooth/camera/CameraPicamera.py). I added this to make the preview update faster, as the full camera resolution is not necessarily required druing preview. However, this keyword was added in PiCamera 1.9 (current is 1.13) and I guess your version is simply too old.

How did you install the photobooth? Did you specify `[picamera]`, i.e., use the following command:
```bash
pip install -e .[picamera]
```

Chefsessel, 2018-07-18:
Followed the readme install instruction. Yes, did install the optional picamera module.

Anything I could try out?

reuterbal, 2018-07-18:
Please do the following:
1. Remove `resolution=self._preview_resolution` from the function call in l.63f. This should fix the problem for you - if not, please post the new logfile.
2. Please post the output of `pip freeze` in your virtual environment. This helps me.

Chefsessel, 2018-07-18:
Yep, deleting the argument solved the issue. Preview works fine again.

As requested, the results of pip freeze

automationhat==0.1.0
blinker==1.3
blinkt==0.1.2
buttonshim==0.0.2
Cap1xxx==0.1.3
chardet==2.3.0
click==6.6
colorama==0.3.7
cryptography==1.7.1
drumhat==0.1.0
envirophat==1.0.0
ExplorerHAT==0.4.2
Flask==0.12.1
fourletterphat==0.1.0
gphoto2==1.8.2
gpiozero==1.4.1
idna==2.2
itsdangerous==0.24
jedi==0.10.2
Jinja2==2.8
keyring==10.1
keyrings.alt==1.3
MarkupSafe==0.23
mcpi==0.1.1
microdotphat==0.2.1
mote==0.0.3
motephat==0.0.2
numpy==1.12.1
oauthlib==2.0.1
opencv-python==3.4.1.15
pantilthat==0.0.6
pexpect==4.2.1
pgzero==1.2
phatbeat==0.1.1
-e git+https://github.com/reuterbal/photobooth.git@3e1341e3d950f3c1009159d361f8276a3520d699#egg=photobooth
pianohat==0.1.0
picamera==1.13
picraft==1.0
piglow==1.2.4
pigpio==1.38
Pillow==4.0.0
pkg-resources==0.0.0
ptyprocess==0.5.1
pyasn1==0.1.9
pycrypto==2.6.1
pygame==1.9.3
pygobject==3.22.0
pyinotify==0.9.6
PyJWT==1.4.2
pyOpenSSL==16.2.0
pyserial==3.2.1
python-apt==1.1.0b5
pyxdg==0.25
rainbowhat==0.1.0
reportlab==3.3.0
requests==2.12.4
requests-oauthlib==0.7.0
RPi.GPIO==0.6.3
RTIMULib==7.2.1
scrollphat==0.0.7
scrollphathd==1.2.1
SecretStorage==2.3.1
sense-emu==1.1
sense-hat==2.2.0
simplejson==3.10.0
six==1.10.0
skywriter==0.0.7
sn3218==1.2.7
spidev==3.3
thonny==2.1.16
touchphat==0.0.1
twython==3.4.0
unicornhathd==0.0.3
urllib3==1.19.1
virtualenv==15.1.0
Werkzeug==0.11.15

reuterbal, 2018-07-18:
Thanks for the output - versions look fine to me, I'll have a look at this and try to reproduce it on my hardware.

reuterbal, 2018-07-18:
I can confirm the bug - it was an oversight on my side. The parameter is called "resize", not "resolution". It is fixed now. Thanks for reporting!

commit e7a4e93c (2018-07-18): Renamed falsely named parameter in Picamera, closes #27
closed 2018-07-18 by reuterbal