reuterbal/photobooth #28: Theme won't scale with window size

hollesse, 2018-07-18:
**Describe the bug**

if I change the window size in the settings to 800 x 480 (Resolution of RasPi Touchdisplay) then the screen doesn't look like it should! Also if I choose full hd resolution it looks weird. 

**To Reproduce**

Steps to reproduce the behavior:
1. Choose theme 'pastel'
2. Go to settings and change the resolution to 800 x 480 or 1920 x 1080
3. Click on start photobooth
4. See weird screen

**Screenshots**

 [Screenshot with 800x480 res](https://imgur.com/1YohyH7)
 [Screenshot with 1920x1080 res](https://imgur.com/m2XNCLM)

**Hardware (please complete the following information)**

 - Device: Raspberry Pi 3
 - Camera: Canon EOS 600D
 - GPIO: No

**Software (please complete the following information)**

 - OS Raspbian Stretch
 - Python version 3.5.3
 
 **Installed packages**
 
 ```
 automationhat==0.1.0
blinker==1.3
blinkt==0.1.2
buttonshim==0.0.2
Cap1xxx==0.1.3
chardet==2.3.0
click==6.6
colorama==0.3.7
cryptography==1.7.1
drumhat==0.1.0
enum34==1.1.6
envirophat==1.0.0
ExplorerHAT==0.4.2
Flask==0.12.1
fourletterphat==0.1.0
gpiozero==1.4.1
idna==2.2
ipaddress==1.0.17
itsdangerous==0.24
Jinja2==2.8
keyring==10.1
keyrings.alt==1.3
MarkupSafe==0.23
mcpi==0.1.1
microdotphat==0.2.1
mote==0.0.3
motephat==0.0.2
numpy==1.12.1
oauthlib==2.0.1
pantilthat==0.0.5
phatbeat==0.1.1
pianohat==0.1.0
picamera==1.13
picraft==1.0
piglow==1.2.4
pigpio==1.38
Pillow==4.0.0
pyasn1==0.1.9
pycrypto==2.6.1
pygame==1.9.3
pygobject==3.22.0
pyinotify==0.9.6
PyJWT==1.4.2
pyOpenSSL==16.2.0
pyserial==3.2.1
pyxdg==0.25
rainbowhat==0.1.0
requests==2.12.4
requests-oauthlib==0.7.0
RPi.GPIO==0.6.3
RTIMULib==7.2.1
scrollphat==0.0.7
scrollphathd==1.2.1
SecretStorage==2.3.1
sense-emu==1.0
sense-hat==2.2.0
simplejson==3.10.0
six==1.10.0
skywriter==0.0.7
sn3218==1.2.7
spidev==3.3
touchphat==0.0.1
twython==3.4.0
unicornhathd==0.0.3
urllib3==1.19.1
Werkzeug==0.11.15
 ```

**Photobooth log**

```
Insert the content of photobooth.log here
```

**Additional context**

Add any other context about the problem here.

reuterbal, 2018-07-18:
Yes, this is not unexpected. The stylesheets are taylored to the resolution of my screen - this especially includes the size of the background image etc.

If you need a different screen size or aspect ratio, I recommend to create a new stylesheet file in [photobooth/gui/Qt5Gui/stylesheets](https://github.com/reuterbal/photobooth/tree/master/photobooth/gui/Qt5Gui/stylesheets) (use one of the existing files as a basis) and adapt it to your needs. 
Afterwards you can include the stylesheet in [`photobooth/gui/Qt5Gui/__init__.py`](https://github.com/reuterbal/photobooth/blob/master/photobooth/gui/Qt5Gui/__init__.py), it will then appear in the settings menu.

You can also try to adapt the existing ones to arbitrary screen sizes - I'd appreciate a PR very much.
I simply didn't have the time to play around with relative units until it was working on different screen sizes.

wikijm, 2018-08-23:
Good evening @hollesse,

Were you able to try the solutions proposed by @reuterbal ?

wikijm, 2018-09-02:
It seems the issue author is not interested in providing more details. Closing this due to inactivity.

closed 2018-09-02 by wikijm