reuterbal/photobooth #29: [Project management] Use of 'Projects' GitHub feature

wikijm, 2018-07-22:
Hi @reuterbal 

Would you agree to enable and use the GitHub 'Projects' feature to track progress of feature additions/requests and incident handling?

I was able to set it up for the following project: https://github.com/alexylem/jarvis/projects/1

This way, one could at a glance quickly see if a feature has a chance to be implemented, and its degree of advancement if under development.

I am not a professional developer (another branch in IT), so it may not correspond to a working method adapted to the subject ;-)

Sincerely,
@wikijm

reuterbal, 2018-07-23:
That is an excellent idea! 

However, I have no experience with that feature of Github. Do you have a suggestion what might be a good layout in this case (i.e., what kind of columns)?

Issues here are mostly either questions on how to install/use things and rarely specific bug reports or explicit feature requests, but they can often be associated with one or multiple (!) feature requests at the end. Is the projects page able to associate single issues to multiple tasks?

I like the layout of your link, do you think it is suitable here? What about ideas that I do not plan on adding? Those might still be good ideas but out of scope for me (too much work at little gain for my own box, unfeasible because I don't have required hardware or knowledge, etc.).

I'd appreciate any input and remarks!

wikijm, 2018-07-23:
Hi  @reuterbal 

To complete my remarks:

- For the OpenJarvis project, I created several 'Project' within the 'Projects' functionality. For photobooh, we could go on the following list:

1. Photobooth Core
2. Photobooth Plugins (optional)
3. Documentation

- For each'project', I propose the following columns:
1. To be discussed
2. Confirmed
3. On-going
4. Blocked (for ideas that you do not plan on adding, maybe?)
5. Done

If you don't mind, I'd like to help you with 'Project' and issues handling, as I do for https://github.com/alexylem/jarvis project.

reuterbal, 2018-07-23:
Hi @wikijm ,
thanks a lot for the valuable insights! I'd really appreciate if you took the time to set this up! What do you need me to do to get you going?
(No hurry, I'll be travelling a lot in August and until then I'm busy fixing up my personal photobooth for a friends wedding.)

Regarding the 'projects' I'm ok with 1 and 3. 
Plugins is probably not applicable (yet) but we have several components that represent the threads in the background, with one for each: master, camera, gui, postprocessing, gpio. Some of them are split into several different modules that provide the same/similar functionality, e.g. different camera modules (gphoto2, picamera, opencv etc.) or possibly different gui backends (at the moment there is only Qt5). However, theese modules are essentially all part of the core.

For the future I was thinking to add more postprocessing features, e.g. upload to dropbox, google photos, instagram, ... or include a small webserver that allows to view and download the pictures via Wifi. These could be deemed something like plugins, but the name is probably too general. Thus I suggest to leave that out for now and add a 'Plugins project' once this becomes closer to the horizon.

With the colums I concur 100%. 'Blocked' could be for ideas I'm not planning on adding or things that are blocked by something upstream (e.g., autofocus issues in gphoto2, see #7 (parts in German)).

wikijm, 2018-07-24:
If it's OK for you, I need you to attribute me few admin rights on the Project feature.
This way, I'll be able to do a PoC and show you some possibilities 😃

reuterbal, 2018-07-25:
I've added you as a collaborator. This should grant you all required rights - fine granular rights management isn't possible for personal repositories.

wikijm, 2018-07-25:
After some thought, I did:
- Easier for the'Project' Documentation
- A bit automated for the'Project' Photobooth Core (more info [here](https://help.github.com/articles/configuring-automation-for-project-boards/))

reuterbal, 2018-07-27:
Looks very good, thank you! I already experienced the automation of being moved to "Done" when closing an issue :)
What do you mean by 'Easier for the Project Documentation'?

wikijm, 2018-07-28:
Hi @reuterbal 

'Easier' means in this case 'with less steps' (only To do => In progress => Done).

reuterbal, 2018-07-30:
Oh sorry, now I get it. I would probably call it 'simpler' instead of 'easier'.

Anyways, thanks a lot for the help!

Another idea: Let's make a third 'project' called 'Success stories' and group all issues labeled as 'success story' there. This way they can be found easily and the issues themselves can be closed. What do you think?

wikijm, 2018-07-30:
Easier to handle, simpler to use 😄 
Pardon my English. I'm just an honest Frenchman 😜 

For success stories, I had thought about it, but I think a Wiki page is more suitable.
Indeed, once the experience feedback is done, it is unlikely to evolve. Or if it evolves, it will probably be linked to another event, and potentially to another version (change of the atmosphere of the wedding, of the accessories, suppression of the errors met before, ...).

The choice is yours, both defend themselves and are possible.

reuterbal, 2018-07-30:
No problem, sorry for being a smart a** 😄

You're right, a Wiki page is a good idea and probably the better place for this.

wikijm, 2018-07-30:
Tip: With a 'Collaborator' profile, I'm not able to activate the Wiki feature.

reuterbal, 2018-07-30:
Sorry, wiki is activated.

wikijm, 2018-07-30:
No need to apologize, it's part of the construction 😃

reuterbal, 2018-08-02:
I created a small overview wiki page: https://github.com/reuterbal/photobooth/wiki/Success-stories
I'll add issues for my old and new photobooth version soon, including pictures...

reuterbal, 2018-08-21:
I'll close this as the projects feature is enabled and working.
Thanks for all the support and taking time to answer all the requests!

closed 2018-08-21 by reuterbal

wikijm, 2018-08-21:
You're welcome, my pleasure  😜