reuterbal/photobooth #32: [Core request] Adding multilingual support

wikijm, 2018-07-23:
Hi @reuterbal,

Is it possible to add a multilingual system?
With each new version, I am compelled to replace the text of the different menus.

For my part, I could propose and maintain a 'translation file' for the French language.

Sincerely,
WikiJM

reuterbal, 2018-07-23:
Hi @wikijm 
this is something that is already on my to-do list ;-) I'd probably opt for Pythons [gettext module](https://docs.python.org/3.5/library/gettext.html) as it is the canonical way to go. I tried to write the code more or less ready for this, so this should be doable in relatively short time.
I can provide English and German versions.

ferdinandkeil, 2018-07-24:
This is on my wish-list as well. I can implement it if you like :smile:

reuterbal, 2018-07-24:
I won't object to that! Let me know if you need help. I'd come around probably not before September.

reuterbal, 2018-07-27:
I introduced the necessary changes into the code in 884df55a2278347c1ae7ce8f89c07d661f02025c. What remains to be done is to add the required infrastructure for extracting the messages and, of course, the translations themselves. This seems to be a bit cumbersome to get it working using setuptools. The best way is probably using [babel](http://babel.pocoo.org/).

reuterbal, 2018-08-03:
I have set-up POedtitor to allow for easy translation: https://poeditor.com/join/project/K7S5aObaJ9
Feel free to join and contribute a new language.

I have to find out how to automatically compile and distribute .mo files when using `pip install`. If someone knows something, help is truly appreciated!

wikijm, 2018-08-03:
100% done for French language 😄 

Have a nice week-end ✌️

thyristor59, 2018-08-12:
Bonjour @wikijm a

Je fais suite à ton mail, 
n'étant pas du tout du milieu informatique, lorsque je modifie un fichier au moment de la sauvegarde j'ai un message d'erreur.?

Pour tout le reste en anglais avec les traducteurs etc.. j'arrive à m'en sortir !

Quand je demandais une traduction, c'est pour les 3 interfaces :

Hit this bouton...
Get ready...
Les décomptes, ça c'est bon 😇
Happy?

Il me semble que si on crée un image avec du texte en français au même format avec le même nom dans le même fichier ça pourrait fonctionner?

Cependant, quel logiciel a était utilisé ?
Quelles sont les dimensions ?

Pour moi le logiciel me correspond amplement.

J'ai pour but de le faire en miroir
Exemple : https://img-3.journaldesfemmes.com/8VEfJCUJNFQVtSN29t5SiG43s0s=/1363x/smart/eb5e882164d34527babd77950c3b1339/ccmcms-jdf/10590981.jpg

wikijm, 2018-08-13:
Bonsoir @thyristor59 

Fichier ./photobooth/photobooth/gui/Qt5Gui/Frames.py
```python
#!/usr/bin/env python3
# -*- coding: utf-8 -*-

# Photobooth - a flexible photo booth software
# Copyright (C) 2018  Balthasar Reuter <photobooth at re - web dot eu>
#
# This program is free software: you can redistribute it and/or modify
# it under the terms of the GNU Affero General Public License as published by
# the Free Software Foundation, either version 3 of the License, or
# (at your option) any later version.
#
# This program is distributed in the hope that it will be useful,
# but WITHOUT ANY WARRANTY; without even the implied warranty of
# MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
# GNU Affero General Public License for more details.
#
# You should have received a copy of the GNU Affero General Public License
# along with this program. If not, see <http://www.gnu.org/licenses/>.

import logging
import os
import subprocess
import sys

from PyQt5 import QtCore
from PyQt5 import QtGui
from PyQt5 import QtWidgets

from .. import modules
from ... import camera
from ... import printer

from . import Widgets
from . import styles

class Welcome(QtWidgets.QFrame):

    def __init__(self, start_action, set_date_action, settings_action,
                 exit_action):

        super().__init__()

        self.initFrame(start_action, set_date_action, settings_action,
                       exit_action)

    def initFrame(self, start_action, set_date_action, settings_action,
                  exit_action):

        btnStart = QtWidgets.QPushButton('Démarrer photobooth')
        btnStart.clicked.connect(start_action)

        btnSetDate = QtWidgets.QPushButton('Définir date/heure')
        btnSetDate.clicked.connect(set_date_action)

        btnSettings = QtWidgets.QPushButton('Paramètres')
        btnSettings.clicked.connect(settings_action)

        btnQuit = QtWidgets.QPushButton('Quitter')
        btnQuit.clicked.connect(exit_action)

        btnLay = QtWidgets.QHBoxLayout()
        btnLay.addWidget(btnStart)
        btnLay.addWidget(btnSetDate)
        btnLay.addWidget(btnSettings)
        btnLay.addWidget(btnQuit)

[... 1050 lines cut ...]
        if self._cfg.getBool('Gui', 'fullscreen'):
            self.showFullScreen()
        else:
            self.setFixedSize(self._cfg.getInt('Gui', 'width'),
                              self._cfg.getInt('Gui', 'height'))
            self.show()

    def closeEvent(self, e):

        reply = QtWidgets.QMessageBox.question(self, 'Confirmation',
                                               "Quit Photobooth?",
                                               QtWidgets.QMessageBox.Yes |
                                               QtWidgets.QMessageBox.No,
                                               QtWidgets.QMessageBox.No)

        if reply == QtWidgets.QMessageBox.Yes:
            e.accept()
        else:
            e.ignore()

    def keyPressEvent(self, event):

        self._handle_key(event)

```

thyristor59, 2018-08-21:
really dsl I had not seen your job ...

maybe a silly question ...

I just need to copy / paste in the respective files?

wikijm, 2018-08-21:
No problem 😸 
Precisely, but don't forget to backup your files before any modification.

thyristor59, 2018-08-21:
when i start the program:

Process GuiProcess-3:
Traceback (most recent call last):
  File "/usr/lib/python3.5/multiprocessing/process.py", line 249, in _bootstrap
    self.run()
  File "/home/pi/photobooth/photobooth/main.py", line 76, in run
    'gui')
  File "/home/pi/photobooth/photobooth/util.py", line 33, in lookup_and_import
    'photobooth.' + package + '.' + result[0])
  File "/home/pi/photobooth/.venv/lib/python3.5/importlib/__init__.py", line 126, in import_module
    return _bootstrap._gcd_import(name[level:], package, level)
  File "<frozen importlib._bootstrap>", line 986, in _gcd_import
  File "<frozen importlib._bootstrap>", line 969, in _find_and_load
  File "<frozen importlib._bootstrap>", line 958, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 673, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 673, in exec_module
  File "<frozen importlib._bootstrap>", line 222, in _call_with_frames_removed
  File "/home/pi/photobooth/photobooth/gui/Qt5Gui/__init__.py", line 25, in <module>
    from .PyQt5Gui import PyQt5Gui  # noqa
  File "/home/pi/photobooth/photobooth/gui/Qt5Gui/PyQt5Gui.py", line 20
    importation argparse
                       ^
SyntaxError: invalid syntax

reuterbal, 2018-08-21:
Somehow, `importation` instead of `import` is written in line 20.

thyristor59, 2018-08-22:
from .PyQt5Gui import PyQt5Gui  # noqa
  File "/home/pi/photobooth/photobooth/gui/Qt5Gui/PyQt5Gui.py", line 71
    self ._gui = None

thyristor59, 2018-08-22:
I'm sorry but I love to understand what I'm doing, that's why I ask so much information!

already thanks to you! I learned a lot about raspberry with you!

here it is very complicated for me, copied / paste do not like me. (although very useful)

I would like to understand the spelling!
what I would also like is to develop an audio part during the countdown.
and I would like to develop it myself! 😉

reuterbal, 2018-08-22:
There went something terribly wrong with your copy-paste. The error-messages are incomplete, with those I can't really help you. I suggest to try copy-paste new or wait for the translation to become ready.

If you want to add features yourself, it is probably best to learn some basics of Python first - at least such that you understand the syntax and basic concepts of the language. There are plenty of Python 3 books and online tutorials in most languages available.

thyristor59, 2018-08-22:
Yes, I think I copy/paste in french...

I work it this afternoon and I read a book 😉
[image]

reuterbal, 2018-08-22:
This book looks pretty complete :)

reuterbal, 2018-08-23:
Translation support is now working in development branch. Only restriction at the moment: You have to run the application from the top directory (which you are doing anyways, probably).

If anyone knows how to tell `pip` / `setuptools` to install the `.mo`-files into the system folder/virtual environment, please let me know...

With that said, the interface language should automatically match the system language (if such a translation is available). 
To select a different language, you have to set the `LANGUAGE` environment variable, e.g.:
```bash
LANGUAGE=fr .venv/bin/python -m photobooth
```
where `fr` stands for French, `de` for German, `en` for English, `es` for Spanish (you see where I'm going... ;-) )
Please test it and report back any issues!

reuterbal, 2018-08-23:
PS: Of course, thank you to all the translators. I'm happy to add more languages (Italian, Russian, Czech, Polish, Swedish, Portuguese, ... anyone?)

thyristor59, 2018-08-26:
Hello, here I am again!
 I thank you everything @reuterbal @wikijm  is ok in french, I'm really happy!

For the audio part, I find on the internet how to do and the music works with what I wrote goal how to integrate it into your program?

explanation:
when there is the count before the picture; at each count I would like a "beep"

and

when the wait for the final fomrat of the 4 photos I would like a music of drying!

ps: if you have to open another topic for this topic no problem!
I just want a few leads to do it myself

wikijm, 2018-08-26:
Hi @thyristor59 

Glad to read that it is working well for you 👍 

Concerning your sound-related request, yes you should create an another issue named something like:
> [Core request] Sounds during countdown

botyas, 2018-09-16:
In the translation into Spanish there are some erores. I attached the files po and mo with the changes I made.

[LC_MESSAGES.zip](https://github.com/reuterbal/photobooth/files/2386444/LC_MESSAGES.zip)

Thank you very much for your work

jsitges, 2018-09-16:
Good day everyone.

I did he translation into Spanish, initially I tried to look for Spanish
from Mexico. This the Spanish errors, or at least some may come from
different word usage.

Anyhow, funny enough, I don't know if the translation only works in the
development version, if it does, I would like to know the name of it for
when I clone it in GitHub.

If I had the master photobooth installed already, then I assume that
renaming the  directory in my computer, then downloading the development
branch would leave me with the development branch working.  It's in a
virtual environment anyways.

Well, I hope someone else finds what I have shared useful.

Julio Sitges Pomares

botyas, 2018-09-20:
To download the version of this article, read and read how it works and how it works with the language you want. I changed the name to the folder and cloned again with the development version because I did not have any modified files. If you have modified something I think there is a command to keep those changes.

https://github.com/reuterbal/photobooth/issues/52#issuecomment-418687154

reuterbal, 2018-09-21:
Hi all,

thanks for testing the translations! Sorry if there are errors/discrepancies between different countries in the Spanish version. I don't know Spanish so I can't judge the translations :-)

@jsitges Would the changed version of @botyas work for you as well? Or should we introduce two version of Spanish ('es_mx' and 'es')?

Regarding the development branch: Please follow the instructions linked to by @botyas . I'm buried deep at work at the moment, so I didn't have time to update documentation properly, which I want to do before merging the development branch into the master and releasing a new version.

commit b6f988bc (2018-08-23): Include .mo files to close #32 (for now)
closed 2018-10-19 by reuterbal