reuterbal/photobooth #33: Autostart on boot

jul1792, 2018-07-24:
Hi everybody,

I already had success in installing the photobooth and running it thanks to @reuterbal. 

Now I would like the programme to start on boot, but I can't get it working. I already tried the method to edit the rc.local-file, but I hadn't success with it. I also use a virtual environment. These are the commands im using to start the programme via command line:
`cd photobooth`
`source .venv/bin/activate`
`python photobooth`

Can anyone help me to get it start up right away after my Raspberry Pi boots? I don't understand much about using the command line and linux, so could anyone provide a detailed instruction?
And is there also a way to not have to choose to start the photobooth, but that I can start right away taking photos?

I use Raspbian Stretch.

Thanks a lot in advance!

reuterbal, 2018-07-24:
Hi @jul1792 

startup options are documented [here](https://github.com/reuterbal/photobooth/blob/master/INSTALL.md#run-photobooth).
To start the photobooth from a script or similar it is best to directly use the python binary of the virtual environment instead of loading the environment first. In your case:
```
/home/pi/photobooth/.venv/bin/python -m photobooth
```
To skip the welcome screen, simply add `--run`.

In an older version of Raspbian/LXDE, applications could be added to autostart by adding them to the file `/home/pi/.config/lxsession/LXDE-pi/autostart`. I have no idea, whether this is still working.
For the photobooth, add the following line:
```
lxterminal -e "/home/pi/photobooth/.venv/bin/python -m photobooth --run"
```
Please let me know whether this works - I'll add it then to the installation manual.

jul1792, 2018-08-14:
Hi @reuterbal,
Unfortunately adding the line in the file you named, didn‘t work. When I started my RPi it showed the desktop with a terminal window opened, in which I couldn’t type in anything. And that‘s it. 

Do you have any other solutions? Thanks a lot!

wikijm, 2018-08-15:
Hi @jul1792 

Here is a quite similar solution (Take care to do a backup and adapt the path of the Photobooth folder before any action):
1. Create autostart.sh file and make it executable
```bash
cd /home/pi/photobooth/
touch ./autostart.sh
chmod +x ./autostart.sh
```

2. Add commands to this file
```bash
cat <<EOT >> autostart.sh
#!/bin/bash
sleep 4      # Wait 4 second before execution
cd /home/pi/photobooth/      # Go to Photobooth directory
source .venv/bin/activate
python -m photobooth
EOT
```
3. Add a new line to '/home/pi/.config/lxsession/LXDE-pi/autostart' file
```bash
echo "@bash /home/pi/photobooth/autostart.sh" >> /home/pi/.config/lxsession/LXDE-pi/autostart
```

4. Reboot your Raspberry to verify if everything's works fine

wikijm, 2018-08-15:
@reuterbal If you think my solution is correct, I will provide to you a pull request for this 😉

jul1792, 2018-08-15:
Hi @wikijm,
thanks for your solution. Unfortunately, it didn't work. The screen simply shows me the Desktop, such as nothing has changed.
The file '/home/pi/.config/lxsession/LXDE-pi/autostart' now looks like this
`lxpanel --profile LXDE-pi
@pcmanfm --desktop --profile LXDE-pi
@xscreensaver -no-splash
@point-rpi
echo "@bash /home/pi/photobooth/autostart.sh" >> /home/pi/.config/lxsession/LXD$
`
Does this look correct to you? Do you have any solution for my problem?

wikijm, 2018-08-15:
My bad, I wasn't very clear on my explanation.

You actually have to play the command, not add this line manually into the file.
The same applies to points 1 and 2.

So your files should look like this:
* /home/pi/photobooth/autostart.sh
```bash
#!/bin/bash
sleep 4      # Wait 4 second before execution
cd /home/pi/photobooth/      # Go to Photobooth directory
source .venv/bin/activate
python -m photobooth
```

* /home/pi/.config/lxsession/LXDE-pi/autostart
```bash
@lxpanel --profile LXDE-pi
@pcmanfm --desktop --profile LXDE-pi
@xscreensaver -no-splash
@point-rpi
@bash /home/pi/photobooth/autostart.sh
```

reuterbal, 2018-08-20:
The file `autostart.sh` can actually be as simple as
```bash
#!/bin/bash
sleep 4 && /home/pi/photobooth/.venv/bin/python -m photobooth
```
or, possibly this line could be added to the autostart configuration directly.

reuterbal, 2018-08-22:
I tested it. Unfortunately, it is not possible to use the command directly as this means that the working directory is wrong. I have added a script [`autostart.sh`](https://github.com/reuterbal/photobooth/blob/development/autostart.sh) to the development branch, which can be used to run the photobooth automatically upon start.

To use this, simply add
```
@/home/pi/photobooth/autostart.sh
```
to `/home/pi/.config/lxsession/LXDE-pi/autostart`

reuterbal, 2018-08-23:
This is now documented in https://github.com/reuterbal/photobooth/wiki/Tips-and-tricks

closed 2018-08-23 by reuterbal

fubl84, 2019-06-13:
In couldn't find the lxsession-folder under /home/pi/.config
Maybe it's because I used a clean raspbian install without any user accounts.
Fortunately I found this solution - maybe it wil be helpful for someone.

The autostart file, regardless which user logs in, can be found under:
/etc/xdg/lxsession/LXDE-pi/autostart

(I'm using the Raspbian Stretch version from April 2019)

wikijm, 2019-06-13:
Thanks @fubl84, I've just added your tip to [this ](https://github.com/reuterbal/photobooth/wiki/Tips-and-tricks)Wiki page.

reuterbal, 2019-06-13:
It is a very bad idea to do this for all users. I changed the Wiki accordingly. Simply create folders and files as required.

wikijm, 2019-06-13:
Good to know that creating manually the required folders is enough.
Thanks @reuterbal 😃