reuterbal/photobooth #34: [Core request] Ability to replace every 'ERROR: ...' by a generic error message

wikijm, 2018-07-24:
**Describe the request**
On the first release, I was have been able to replace the generic error message by something like:
```python
# Catch exceptions and display message
            except CameraException as e:
                self.handle_exception(e.message)
            # Do not catch KeyboardInterrupt and SystemExit
            except (KeyboardInterrupt, SystemExit):
                raise
            except Exception as e:
                print('Try red button/Call WikiJM: ' + repr(e))
                self.handle_exception("Try red button OR Call WikiJM")
```

**To Reproduce**
Steps to reproduce the behavior:
1. Start photobooth with `python -m photobooth`
2. Click on 'START PHOTOBOOTH' without connected camera
3. See error 'ERROR: ...'

**Expected behavior**
Being able to configure and replace 'ERROR: ...' message by something like 'Try to push 'RESET button/Call the tech guy'

**Screenshots**
[image]

**Hardware (please complete the following information)**
N/A

**Software (please complete the following information)**
N/A

 **Installed packages**
N/A

**Photobooth log**
N/A

**Additional context**
N/A

reuterbal, 2018-07-25:
You can easily achieve something similar now. All error messages are routed through `ErrorState` in `StateMachine.py`. Simply replace 
```python
self.message = message
```
by
```python
self.message = 'Try red button OR Call WikiJM'
```
(or whatever you want it to display). This should do the trick.

wikijm, 2018-07-25:
Thanks for this fast answer 👍 

Do you think it is possible to:
1. Add a variable named 'errormessagegeneric' or similar in this part of [./photobooth/defaults.cfg](https://github.com/reuterbal/photobooth/blob/3e1341e3d950f3c1009159d361f8276a3520d699/photobooth/defaults.cfg)
```python
[Photobooth]
# Show preview while posing time (True/False)
show_preview = True
# Greeter time in seconds (shown before countdown)
greeter_time = 3
# Countdown length in seconds (shown before every shot)
countdown_time = 8
# Display time of assembled picture (shown after last shot)
display_time = 5
# Timeout for postprocessing (shown after review)
postprocess_time = 60
# Generic message to display when an error appears on the screen
errormessagegeneric = 'Try the reset button OR Call the tech guy'
```

2. Replace
```python
self.message = message
```
by
```python
self.message = errormessagegeneric
```

If possible, i'll create a pull request for this.

commit d53c1c74 (2018-10-19): Added overwrite error message that is displayed instead of the actual error. Error message itself is still logged. Closes #34