reuterbal/photobooth #35: When saving single images, last image is always corrupt

jalewis, 2018-07-29:
**Describe the bug**

When "Keep single shots" is set, the last image is corrupt.  I'm using an Canon EOS Rebel T6 (1300D), which is supported by gphoto2.  I've tested with multiple numbers of photos and the last photo is always corrupt.  I've swapped out OS and hardware thinking that the RPI could not keep up with writing images.  This had no effect.  I wonder if there are two processes trying to read/write the last file at the same time.  Ex, the file is being written to disk at the same time that the process creates the final picture.

**To Reproduce**.

Steps to reproduce the behavior:
1. Enable "Keep single shots"
2. run photobooth

**Expected behavior**

I expect the last image to not be corrupted.

**Hardware (please complete the following information)**

 - Tested on RaspberryPi 3 B+ and minipc
 - Canon EOS 1300D
 - GPIO: No

wikijm, 2018-07-29:
Hi @jalewis,

Thanks for your bug report.
Due to the lack of knowledge I have in Python, I don't think I can help you, but I tagged your issue as a bug. This way @reuterbal will be able to take it into account.

During this time, do you have a success story [like this one](https://github.com/reuterbal/photobooth/issues/11) to share?

reuterbal, 2018-07-30:
Hi @jalewis,

thanks for reporting the issue!

Which one do you refer to as 'last' picture? To make it clearer:
Say, you selected 2x1 pictures and 'keep single shots'. Then the following files are written in this order:
1. photobooth_shot_00001.jpg
2. photobooth_shot_00002.jpg
3. photobooth00001.jpg

Which one is corrupted? The last 'shot'-file or the assembled picture, which is actually the last one to be written. All files are handled by the same worker process one-after-another.

Unfortunately, I can't reproduce the issue myself. Could you please add the log file?

@wikijm Could you please wait with labeling issues as 'bugs' until they are confirmed? Once they are confirmed, we can attribute them the correct labels and assign them to someone.

wikijm, 2018-07-30:
Hi @reuterbal,

Sorry, I did go a little fast indeed.
That's noted for next time 😇

jalewis, 2018-07-30:
In your example, photobooth_shot_00002.jpg is corrupt.  The last shot before the final image is created.

Log:

2018-07-29 13:34:43,521 - root - INFO - Camera configuration:
Camera Actions [actions]:
    Synchronize camera date and time with PC [syncdatetimeutc]: 0
    Synchronize camera date and time with PC [syncdatetime]: 0
    UI Lock [uilock]: 2
    Drive Canon DSLR Autofocus [autofocusdrive]: 0
    Drive Canon DSLR Manual focus [manualfocusdrive]: None (Near 1, Near 2, Near 3, None, Far 1, Far 2, Far 3, )
    Cancel Canon DSLR Autofocus [cancelautofocus]: 0
    Canon EOS Zoom [eoszoom]: 0
    Canon EOS Zoom Position [eoszoomposition]: 0,0
    Canon EOS Viewfinder [viewfinder]: 2
    Canon EOS Remote Release [eosremoterelease]: None (None, Press Half, Press Full, Release Half, Release Full, Immediate, Press 1, Press 2, Press 3, Release 1, Release 2, Release 3, )
    PTP Opcode [opcode]: 0x1001,0xparam1,0xparam2
Camera Settings [settings]:
    Camera Date and Time [datetimeutc]: 1532889202
    Camera Date and Time [datetime]: 1532885602
    Quick Review Time [reviewtime]: 2 seconds (None, 2 seconds, 4 seconds, 8 seconds, Hold, )
    Camera Output [output]: Off (TFT, PC, TFT + PC, Setting 4, Setting 5, Setting 6, Setting 7, Unknown value 0008, Unknown value 0009, Unknown value 000a, Unknown value 000b, Off, )
    Recording Destination [movierecordtarget]: None (None, )
    EVF Mode [evfmode]: 1 (1, 0, )
    Owner Name [ownername]:
    Artist [artist]:
    Copyright [copyright]:
    Custom Functions Ex [customfuncex]: bc,4,1,2c,3,101,1,0,103,1,0,10f,1,0,2,2c,3,201,1,0,202,1,0,203,1,0,3,14,1,50e,1,0,4,38,4,701,1,0,704,1,0,70e,1,0,811,1,0,
    Focus Info [focusinfo]: eosversion=4,size=5184x3456,size2=5184x3456,points={{0,743,117,181},{-839,393,172,129},{839,393,172,129},{-1394,0,172,129},{0,0,224,222},{1394,0,172,129},{-839,-393,172,129},{839,-393,172,129},{0,-743,117,181}},select={},unknown={ff010000ffff}
    Auto Power Off [autopoweroff]: 0
    Depth of Field [depthoffield]: 0
    Capture Target [capturetarget]: Internal RAM (Internal RAM, Memory card, )
    Capture [capture]: 0
Camera Status Information [status]:
    Serial Number [serialnumber]: a5c341f797fa4c159c3991208b560bad
    Camera Manufacturer [manufacturer]: Canon Inc.
    Camera Model [cameramodel]: Canon EOS Rebel T6
    Device Version [deviceversion]: 3-1.1.0
    Vendor Extension [vendorextension]: None
    Camera Model [model]: 2147484676
    PTP Version [ptpversion]: 256
    Battery Level []: 100%
    Battery Level [batterylevel]: 100%
    Lens Name [lensname]: EF-S18-55mm f/3.5-5.6 IS II
    Serial Number [eosserialnumber]: 272073023559
    Shutter Counter [shuttercounter]: 252
    Available Shots [availableshots]: 80043
Image Settings [imgsettings]:
    Image Format [imageformat]: Medium Fine JPEG (Large Fine JPEG, Large Normal JPEG, Medium Fine JPEG, Medium Normal JPEG, Small Fine JPEG, Small Normal JPEG, Smaller JPEG, Tiny JPEG, RAW + Large Fine JPEG, RAW, )
    Image Format SD [imageformatsd]: Medium Fine JPEG (Large Fine JPEG, Large Normal JPEG, Medium Fine JPEG, Medium Normal JPEG, Small Fine JPEG, Small Normal JPEG, Smaller JPEG, Tiny JPEG, RAW + Large Fine JPEG, RAW, )
    ISO Speed [iso]: Auto (Auto, )
    WhiteBalance X A [whitebalancexa]: 0
    WhiteBalance X B [whitebalancexb]: 0
    Color Space [colorspace]: sRGB (sRGB, )
Capture Settings [capturesettings]:
    Focus Mode [focusmode]: AI Focus (AI Focus, )
    Aspect Ratio [aspectratio]: 3:2 (3:2, )
    Canon Auto Exposure Mode [autoexposuremode]: Green (P, TV, AV, Manual, Bulb, A_DEP, DEP, Custom, Lock, Green, Night Portrait, Sports, Portrait, Landscape, Closeup, Flash Off, )
    Drive Mode [drivemode]: Single (Single, Timer 10 sec, Unknown value 0007, )
    Aperture [aperture]: implicit auto (implicit auto, )
    Shutter Speed [shutterspeed]: auto (auto, )
    Bracket Mode [bracketmode]: 0
    Auto Exposure Bracketing [aeb]: off (off, )
Other PTP Device Properties [other]:
    PTP Property 0xd402 [d402]: Canon EOS Rebel T6
    PTP Property 0xd407 [d407]: 1
    PTP Property 0xd406 [d406]: Unknown Initiator
    PTP Property 0xd303 [d303]: 1
    Battery Level [5001]: 100 (100, 0, 75, 0, 50, )

[... 9 lines cut ...]
2018-07-29 13:35:00,161 - gphoto2 - WARNING - (b'ptp_usb_getresp [usb.c:466]') b'PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)'
2018-07-29 13:35:00,266 - gphoto2 - WARNING - (b'ptp_usb_getresp [usb.c:466]') b'PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)'
2018-07-29 13:35:00,465 - gphoto2 - WARNING - (b'ptp_usb_getresp [usb.c:466]') b'PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)'
2018-07-29 13:35:00,712 - gphoto2 - WARNING - (b'ptp_usb_getresp [usb.c:466]') b'PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)'
2018-07-29 13:35:00,917 - gphoto2 - WARNING - (b'ptp_usb_getresp [usb.c:466]') b'PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)'
2018-07-29 13:35:01,137 - root - INFO - Saving picture as 2018-07-29/photobooth-2018-07-29-13-34-_shot_00002.jpg
2018-07-29 13:35:04,956 - gphoto2 - WARNING - (b'ptp_usb_getresp [usb.c:466]') b'PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)'
2018-07-29 13:35:05,228 - gphoto2 - WARNING - (b'ptp_usb_getresp [usb.c:466]') b'PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)'
2018-07-29 13:35:05,338 - gphoto2 - WARNING - (b'ptp_usb_getresp [usb.c:466]') b'PTP_OC 0x9153 receiving resp failed: Camera Not Ready (0xa102)'
[... cut ...]
2018-07-29 13:44:27,890 - root - INFO - Reading config file "photobooth.cfg"
2018-07-29 13:44:27,979 - root - INFO - Number of last existing file: 0
2018-07-29 13:44:27,986 - root - INFO - Saving pictures as "2018-07-29/photobooth-2018-07-29-13-44-XXXXX.jpg"
2018-07-29 13:44:27,995 - root - INFO - Number of last existing file: 0
2018-07-29 13:44:27,996 - root - INFO - Saving pictures as "2018-07-29/photobooth-2018-07-29-13-44-_shot_XXXXX.jpg"
2018-07-29 13:44:28,003 - root - INFO - GPIO disabled

reuterbal, 2018-07-31:
@jalewis 

thanks for the log. It seems to be missing something in the beginning, however, I might be able to guess the rest.

There is an error reported
```
2018-07-29 13:35:10,592 - root - ERROR - Camera: 'NoneType' object has no attribute 'read'
```
however, the only call to `read` in my code is in the config parser, which isn't used at this point... No idea if this is related.

While digging into possible causes, I found out that I somewhat abused badly the `PIL.Image` objects in combination with `multiprocessing`, which might be the root cause due to concurrency timing issues. I introduced some pickling by hand in the `development` branch, which could counteract this.
Could you test, whether the development branch code version fixes the issue for you?

jalewis, 2018-08-02:
My first tests of the development branch are successful. All images are created correctly.  I will keep testing and report back if I find problems.  Thanks for the quick updates!

reuterbal, 2018-08-02:
Thanks for the response! I'm glad that this seems to fix the problem - I'm quite surprised it didn't happen earlier!

I'll close this, please comment again if the problem returns or open another issue if you encounter new problems.

closed 2018-08-02 by reuterbal

garybreck, 2018-08-20:
I'm n very new to all this code stuff, but my nephew has been helping this old man. With said, 
I just copy every thing in the 2 new .py codes from 
Introduced pickling-by-hand for pictures, possible fix for #35
and  replace the old codes on my Rasberry Pi 
StateMachine.py
and 
util.py
if its more then that "Hello Nephew"

reuterbal, 2018-08-21:
It is actually easier than this: When following the installation instructions, at the point where you [clone the photobooth repository](https://github.com/reuterbal/photobooth/blob/master/INSTALL.md#clone-the-photobooth-repository) replace the command by the following:
```
git clone -b development https://github.com/reuterbal/photobooth.git
```
Hope that helps

garybreck, 2018-08-21:
Sorry, but I'm very green to all this, 
So do I need to delete the file/ folder photobooth? then reinstall or just run this action? 
or start from scratch, reformat the R.P. then reinstall everything?

reuterbal, 2018-08-21:
If you made changes to the Python codes already, it is easiest to delete / rename the folder and just re-install the photobooth application (i.e., start at cloning the repository).

If you haven't made any changes, yet, simply run the following command in the photobooth folder:
```bash
git checkout development
```

This will give you the development version.

reuterbal, 2018-09-02:
With the development version, there is already a french version included. See https://github.com/reuterbal/photobooth/issues/32#issuecomment-415473616 for details