reuterbal/photobooth #36: [Install] Hash error during "pip install -e ." step.

ir0nh3at, 2018-08-03:
**Describe the bug**

When following the installation instructions, all goes well until the "pip install -e ." step.  At that point I receive hash errors that I can't seem to bypass.  I found some suggestions online to try --no-index and --no-cache-dir, neither seems to work.  Output of "pip install -vvv -e ." attached.

Thanks for taking a look.  I'm really trying to get this done this weekend if at all possible.  Thank you!

[pip.txt](https://github.com/reuterbal/photobooth/files/2256171/pip.txt)

ir0nh3at, 2018-08-04:
So I swapped out the SD card - it was pretty slow, so I thought it might have been damaged.  I started from the beginning and had the same issue again with gphoto2.  I then removed it as you suggested and opencv-python errored out again.

I need opencv as I'll be using a Microsoft webcam.  I guess I'll try it on a laptop and see if I have any better luck than the pi.

ir0nh3at, 2018-08-04:
I just followed the instructions on a Mint VM and they worked correctly, no errors.  It looks like it's something with the Pi.  Super frustrating...

reuterbal, 2018-08-21:
If you find out more, please tell us about it. As the installation seems to work for others, I consider this not to be a general problem and will close this.

closed 2018-08-21 by reuterbal