reuterbal/photobooth #37: Error during 'pip install -e .'

jkruszynski, 2018-08-09:
Hey guys,

So I'm looking forward to finally digging into your photobooth. I was able to get a super basic photobooth running a few months ago, but wanted to upgrade the UI, and I came across your project which seems to solve that issue. 

I'm running into an issue with the installation of the app though. I have attached the pip error log here: [pip.log](https://github.com/reuterbal/photobooth/files/2272605/pip.log)

Any help would be greatly appreciated!

reuterbal, 2018-08-09:
Hi @jkr
your Python version is too old. The Photobooth requires at least 3.5, you are using 3.4.
Please upgrade your Raspbian and try again.

jkruszynski, 2018-08-10:
Thanks, have fixed that issue... only to move onto a new one. Some versioning issue with 3.4 and 3.6 pip I think. So I decided to wipe my pi and start clean, and everything worked as expected.

Thanks!

closed 2018-08-10 by jkruszynski