reuterbal/photobooth #38: Printing Issue with Selphy CP1300

MrStonebeard, 2018-08-09:
First, I want to thank you for the wonderful work you've done on this program. It's exactly what I was looking for.

My issue is that after the images have been taken, processed, and sent to the printer, the printer gives an error message saying "Cannot read data!  Cannot print incompatible images or memory card not readable."

I followed your installation instructions exactly. Very thorough and helpful - this is the only issue I've had in the entire process thanks to them.

I am using a Raspberry Pi 3B+ with the latest version of Raspian, Gutenprint 5.2.13. 
Canon SELPHY CP1300, wirelessly connected (CUPS would not recognize via USB)
Raspberry Pi camera V2

I've tested the printer works with the pi by manually printing one of the captured photobooth images using:
lp -d Canon_SELPHY_CP1300 -o raw [image name]

from the command line and it printed just fine.

I am new to Raspberry Pi and Python, and it has been quite a while since I've worked with any flavor of Linux, so please forgive me if this is a newbie issue.

Any help would be appreciated.

Thanks.

[photobooth.log](https://github.com/reuterbal/photobooth/files/2272884/photobooth.log)

reuterbal, 2018-08-09:
Hi @MrStoneboard

I haven't tried printing via WiFi yet, but since USB doesn't work I would try to sort this out first - WiFi is doomed to being harder. 

Once printing via command line works, I would check whether the print job appears on the cups queue correctly. If it does, then the Photobooth works correctly and the issue is somewhere along your printer setup.

wikijm, 2018-08-15:
Hi @MrStonebeard 

Any news about this issue?

botyas, 2018-08-29:
can use any printer, Canon selphy, model 1200 1300 I do not know how to integrate it in my box so that it is hidden and only the printed photo comes out.

reuterbal, 2018-10-19:
Closing this due to inactivity.

closed 2018-10-19 by reuterbal

j-archambault, 2019-06-09:
Hello ! 
I am having the same issue, printing with command line works but that is it.
my configuration is the same as @MrStonebeard

reuterbal, 2019-06-13:
Please try the development branch with the new pycups module. Open a new issue if the problem persists.

j-archambault, 2019-06-14:
Thank you for your answer ! 
As I needed a quickfix I added a simple Printing class that uses the command line.
Side note : I also created a class for the camera using pktriggercord for Pentax cameras.

I can commit the Camera one if you are interested.

BopItFreak, 2022-04-05:
@j-archambault how did you get printing via command line working? I am using the "driverless" drivers on cups. The print doesn't work, it just says "processing" and then errors out and unchecks the "enabled" in the print settings on raspberry pi.

XoVooth, 2022-05-03:
hello @BopItFreak 
did you find the solution to your problem? because I have the same problem with the Cp1300 printer, it works in test page or sending a message via notepad, but not in the printer queue

BopItFreak, 2022-05-03:
Hello. I did solve the issue, but I'm not really sure how. What I can
conclude is that printing images over the network discovered cp1300 simply
gives an invalid data error. It seems only USB connection works with
images. I'd suggest messing with cups settings. The name of my printer also
has _HTTP_ at the end of the name as well. Also, sometimes I have to
restart cups service for it to work. You can do that with sudo service cups
restart

XoVooth, 2022-05-04:
Thanks a lot!!!!

Thanks to you it works!!

I deleted the printer
I opened cup in graphical mode
I added my printer in HTTP network connection
I connected _ to HTTP --> don,c _HTTP_
I restarted cup with the restar command.

and miraculously it works!

I spent a week on the forums and the net without finding anything conclusive

thanks again

anxuae, 2022-05-04:
Good to here! If you have a step-by-step procedure to configure a CP1300, can you write it here: https://github.com/pibooth/pibooth/issues/340 ?