reuterbal/photobooth #39: chang the print out to 4 horizontal on a strip

garybreck, 2018-08-13:
I build a portable Photobooth  Berning Man 

Everything is mounted in a small pelican case
Raspberry Pie 3 B+ 
Mini thermal printer (USB)
Canon Rebel T2i 
Raspberry 7 inch Touchscreen
External USB battery

Raspbian Stretch
Python version 3.5.1

Problem,
when "number of shots per picture" is set out to 4 by 1 , it prints them across (side by side) to fit on the 58mm with of the paper.
Is there any way to change it, so its 4 horizontal Photos one on top of the other like the old school photo booths?

Thanks for your Help
Gary 

 

[image]
[image]
[image]

wikijm, 2018-08-13:
Hi  @garybreck,

Here is an example of configuration you can use:
[image]

And the result:
[image]

Is this fit your needs?

garybreck, 2018-08-13:
Thanks seemed to do the trick! Along with changing the "size of the assembled picture"  now it works.
Thanks

closed 2018-08-13 by garybreck

thyristor59, 2018-08-14:
I tried to make a change for myself when I start the program, it launches a very small page and bueu completely. until reboot the raspberry 😢

I will restart a new memory cars?

wikijm, 2018-08-14:
Hi @thyristor59 

Before you consider a complete reinstallation of your Raspberry, you can try to redo the entire Photobooth installation process.

If you wish to keep a copy of your current installation, remember to rename the folder containing it (I imagine `/home/pi/photobooth` in your case).

thyristor59, 2018-08-15:
[image]

Why i have %y-%m-%d?? And not /Home/user/pi??

wikijm, 2018-08-15:
%Y => Year
%M => Month
%D => Day

With %Y-%M-%D value, a folder with the actual date will be created in /home/pi/photobooth/ directory.
You still have the ability to define a personalized directory if you prefer.