reuterbal/photobooth #40: printing with canon ip2850 pixima

thyristor59, 2018-08-13:
Hello everyone!

I introduce myself, Pierre, 28 years old and completely novice in the field raspberry.
I create two kiosks thanks to the raspberry ... but it is not the subject.

first, a huge thank you for the topic photobooth, very well done / detailed

I still precise, I am novice ...

I did not complete the part Canon selphy CP1200 / CP1300 printer
because I use canon IP 2850 PIXIMA
**_is that bad???_**

I have:
- installed CUPS
- calibrate the printer.
- Print test image

[image]

**Describe the bug**

the print does not start when I click on "print"

**Expected behavior**

I would like to understand where the problem comes from because the image is sent to the printer but remains in the queue 

[image]

[image]
I have unfortunately no image coming out of the printer = (

in advance I wish to thank you for any advice and the time you will take to read me!

I also specify that I am French and that I pass all my texts by translation = /

wikijm, 2018-08-13:
### English
Hi @thyristor59,

Thanks for your presentation and the effort you make to express yourself in English when you know that I also speak French 😜 

This does sound like a driver issue. When the driver's not right, printing sometimes acts like a bit bucket and swallows all output.

To correct this kind  of issue:
1. Go to http://localhost:631. This should be a CUPS administration page.
2. Select _Printers_ and see if your printer is there. If not, add it - see below.
3. Select your printer
4. Select _maintenance_
5. Select _print test page_

Does that work?

To add a printer:
1. Select _Administration_
2. Select _Add Printer_
3. Follow the prompts to find and add your printer and configure it

In addition, you can also follow the following tutorial:
https://www.howtoforge.com/how-to-install-a-canon-printer-on-debian-and-debian-like-systems

### Français
Merci pour ta présentation et l'effort que tu fais pour t'exprimer en anglais, surtout quand tu sais que je parle aussi français 😜 

Cela ressemble à un problème de pilote. Lorsque le pilote n'est pas correct, l'impression agit parfois comme un "aspirateur de données" et avale tout ce qu'on lui envoie sans le rendre.

1. Rends-toi sur http://localhost:631. Tu devrais tomber sur une page d'administration CUPS.
2. Sélectionne Printers et regarde si ton imprimante est là. Si ce n'est pas le cas, ajoute-la - voir ci-dessous.
3. Sélectionne ton imprimante
4. Sélectionne _maintenance_
5. Sélectionne _print test page_

Est-ce que ça marche ?

Pour ajouter une imprimante :
1. Sélectionne _Administration_
2. Sélectionne _Add Printer_
3. Suis les instructions pour trouver et ajouter ton imprimante, puis la configurer

Tu peux également suivre le tutoriel suivant :
https://www.howtoforge.com/how-to-install-a-canon-printer-on-debian-and-debian-like-systems

wikijm, 2018-08-13:
Enfin, tu peux aussi lancer un test d'impression de l'un des fichiers générés par Photobooth.
Pour cela, dans un terminal :
1. Lister les imprimantes :
`lpstat -p -d`

2. Lancer l'impression du fichier 'photobooth0001.jpg' avec l'imprimante 'Canon_IP2850' :
`lp -d  'Canon_IP2850' '/home/pi/photobooth/20180813/photobooth000001.jpg'`

thyristor59, 2018-08-14:
the text image was made without any problem,
however, printing via the terminal does not work with the command lines.

I have managed to print a photobooth print in PDF format after modifying the program parameters.
of course without launching the application

[image]

wikijm, 2018-08-14:
If you are not able to print from command-line, Photobooth will not be able to start printing.

thyristor59, 2018-08-15:
after fully reinstalling the photobooth and the printer, 

now, I know how to print via the terminal however, I have to write it differently than what you indicated to me.

that being said, should I make a modification of the program to show him the way?

If yes, how?

wikijm, 2018-08-15:
You're right, I just changed my original message.
Everything seems OK on your screen captures.

Please follow animation below:
[image]

thyristor59, 2018-08-15:
still not, the tab was already selected, I uncheck, save and quit; revived with the handling of your video, I still have no images coming out of the printer via the photobooth.
however by going into the printer archive, she tells me that the prints are printed
[image]

wikijm, 2018-08-15:
If I understand right:
- Your printer could have been installed with CUPS
- You can print from the command line and from the web interface with the test page
- You can activate the'Print' feature in Photobooth
- When you start a Photobooth print, you see it pass through the CUPS queue, but nothing comes out

Am I correct?

thyristor59, 2018-08-16:
yes, that's absolutely it!

but:
not having the same printer as the tutorial.
I did not realize the part "Canon SELPHY CP1200 / CP1300 printer
Add buster / testing repository "
is that the problem?

I always thank you for your attention to my start on raspberry

thyristor59, 2018-08-16:
for my raspberry experiments, i managed to do the part (add buster / testing repository)
however I now have on the terminal an error and still no impressions
[image]

reuterbal, 2018-08-20:
Hi @thyristor59 

thanks for reporting and sorry for not replying earlier - I'm travelling a lot currently. Thanks to @wikijm for initiating the search for the root cause.

The driver update (buster/testing) is not necessary for your printer but it should not be a problem either.

Could you please check the following:
- Your printer must be selected as the default printer (CUPS setting)
- The paper size in the printer settings must match the paper in the printer (CUPS setting)
- The paper size in the photobooth settings must match the paper in the printer (photobooth setting)

With all that, you should be able to print.
If it still doesn't work, try to enable the PDF printing in the photobooth settings and try to print the PDF from command line using the following command (do not add any additional options, as the photobooth uses default settings as well):
```bash
lp /home/pi/photobooth/print_0.pdf
```

If this doesn't work either, the issue is proably somewhere in the printer configuration in CUPS and not related to the photobooth.

thyristor59, 2018-08-21:
photobooth.cfg:

[Gui]
module = PyQt5
fullscreen = True
width = 1024
height = 600
hide_cursor = False
style = dark

[Camera]
module = gphoto2-commandline

[Gpio]
enable = False
exit_pin = 24
trigger_pin = 23
lamp_pin = 4

[Printer]
enable = True
pdf = False
confirmation = True
module = PyQt5
width = 148
height = 100

[Photobooth]
show_preview = True
greeter_time = 3
countdown_time = 3
display_time = 3
postprocess_time = 60

[Picture]
basedir = %Y-%m-%d
basename = photobooth
num_x = 2
num_y = 2
size_x = 3496
size_y = 2362
min_dist_x = 20
min_dist_y = 20
keep_pictures = False

thyristor59, 2018-08-21:
photobooth.log:

2018-08-19 20:35:09,684 - root - INFO - Initializing photobooth
2018-08-19 20:35:09,685 - root - INFO - Photobooth version: 0.4.dev11+g4ff982c
2018-08-19 20:35:09,688 - root - INFO - Reading config file "/home/pi/photobooth/photobooth/defaults.cfg"
2018-08-19 20:35:09,694 - root - INFO - Reading config file "photobooth.cfg"
2018-08-19 20:35:09,760 - root - INFO - Number of last existing file: 0
2018-08-19 20:35:09,765 - root - INFO - Saving pictures as "2018-08-19/photoboothXXXXX.jpg"
2018-08-19 20:35:09,769 - root - INFO - GPIO disabled
2018-08-19 20:35:09,770 - root - INFO - Number of last existing file: 0
2018-08-19 20:35:09,771 - root - INFO - Saving pictures as "2018-08-19/photobooth_shot_XXXXX.jpg"
2018-08-19 20:35:13,647 - root - INFO - Using printer ""
2018-08-19 20:35:48,965 - root - INFO - Writing config file "photobooth.cfg"
2018-08-19 20:35:49,085 - root - INFO - Restarting photobooth and reloading config
2018-08-19 20:35:49,087 - root - INFO - Photobooth version: 0.4.dev11+g4ff982c
2018-08-19 20:35:49,089 - root - INFO - Reading config file "/home/pi/photobooth/photobooth/defaults.cfg"
2018-08-19 20:35:49,093 - root - INFO - Reading config file "photobooth.cfg"
2018-08-19 20:35:49,127 - root - INFO - Number of last existing file: 0
2018-08-19 20:35:49,133 - root - INFO - Saving pictures as "2018-08-19/photoboothXXXXX.jpg"
2018-08-19 20:35:49,132 - root - INFO - GPIO disabled
2018-08-19 20:35:49,137 - root - INFO - Number of last existing file: 0
2018-08-19 20:35:49,139 - root - INFO - Saving pictures as "2018-08-19/photobooth_shot_XXXXX.jpg"
2018-08-19 20:35:50,074 - root - INFO - Using printer ""
2018-08-19 20:36:00,413 - root - INFO - Exiting photobooth with status code 0
2018-08-20 16:47:12,650 - root - INFO - Initializing photobooth
2018-08-20 16:47:12,653 - root - INFO - Photobooth version: 0.4.dev11+g4ff982c
2018-08-20 16:47:12,657 - root - INFO - Reading config file "/home/pi/photobooth/photobooth/defaults.cfg"
2018-08-20 16:47:12,663 - root - INFO - Reading config file "photobooth.cfg"
2018-08-20 16:47:12,728 - root - INFO - Number of last existing file: 0
2018-08-20 16:47:12,732 - root - INFO - Saving pictures as "2018-08-20/photoboothXXXXX.jpg"
2018-08-20 16:47:12,732 - root - INFO - GPIO disabled
2018-08-20 16:47:12,736 - root - INFO - Number of last existing file: 0
2018-08-20 16:47:12,737 - root - INFO - Saving pictures as "2018-08-20/photobooth_shot_XXXXX.jpg"
2018-08-20 16:47:16,716 - root - INFO - Using printer "Canon_iP2800_series"
2018-08-20 16:47:18,854 - root - INFO - Using python-gphoto2 bindings
2018-08-20 16:47:19,439 - gphoto2 - WARNING - (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90ca receiving resp failed: PTP Operation Not Supported (0x2005)'
2018-08-20 16:47:19,440 - gphoto2 - WARNING - (b'fixup_cached_deviceinfo [library.c:447]') b"'ptp_nikon_get_vendorpropcodes (&camera->pl->params, &xprops, &xsize)' failed: PTP Operation Not Supported (0x2005)"
2018-08-20 16:47:19,614 - gphoto2 - WARNING - (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90ca receiving resp failed: PTP Operation Not Supported (0x2005)'
2018-08-20 16:47:19,615 - gphoto2 - WARNING - (b'fixup_cached_deviceinfo [library.c:447]') b"'ptp_nikon_get_vendorpropcodes (&camera->pl->params, &xprops, &xsize)' failed: PTP Operation Not Supported (0x2005)"
2018-08-20 16:47:19,636 - root - INFO - Camera summary: Manufacturer: Nikon Corporation
Model: D3200
  Version: V1.03
  Serial Number: 72580740000000000000000000000000
Vendor Extension ID: 0xa (1.0)
Vendor Extension Description: microsoft.com: 1.0

Capture Formats: JPEG Undefined Type
Display Formats: JPEG, Undefined Type, Association/Directory, DPOF, Script, Apple Quicktime

Device Capabilities:
	File Download, File Deletion, File Upload
	Generic Image Capture, No Open Capture, Nikon Capture 1, Nikon Capture 2, Nikon Capture 3

Storage Devices Summary:
store_00010001:
	StorageDescription: None
	VolumeLabel:  [Slot 1]
	Storage Type: Removable RAM (memory card)
	Filesystemtype: Digital Camera Layout (DCIM)
	Access Capability: Read Only with Object deletion
	Maximum Capability: 7939817472 (7572 MB)
	Free Space (Bytes): 7816577024 (7454 MB)
	Free Space (Images): 169

Device Property Summary:
Battery Level(0x5001):(read only) (type=0x2) Range [0 - 100, step 1] value: 35% (35)
Image Size(0x5003):(readwrite) (type=0xffff) Enumeration [
	'6016x4000',
	'4512x3000',
	'3008x2000'
[... 465 lines cut ...]
2018-08-21 14:54:08,528 - root - INFO - Number of last existing file: 3
2018-08-21 14:54:08,532 - root - INFO - Saving pictures as "2018-08-21/photoboothXXXXX.jpg"
2018-08-21 14:54:08,536 - root - INFO - Number of last existing file: 0
2018-08-21 14:54:08,537 - root - INFO - Saving pictures as "2018-08-21/photobooth_shot_XXXXX.jpg"
2018-08-21 14:54:09,685 - root - INFO - Using printer "Canon_SELPHY_CP1300"
2018-08-21 14:54:10,948 - root - INFO - Using gphoto2 via command line
2018-08-21 14:54:18,766 - root - INFO - Using camera without preview functionality
2018-08-21 14:55:53,337 - root - INFO - Saving picture as 2018-08-21/photobooth00004.jpg
2018-08-21 14:56:10,546 - root - INFO - Printing picture
2018-08-21 15:06:54,529 - root - INFO - Writing config file "photobooth.cfg"
2018-08-21 15:06:56,654 - root - INFO - Restarting photobooth and reloading config
2018-08-21 15:06:56,655 - root - INFO - Photobooth version: 0.4.dev11+g4ff982c
2018-08-21 15:06:56,658 - root - INFO - Reading config file "/home/pi/photobooth/photobooth/defaults.cfg"
[... cut ...]
2018-08-21 15:07:00,082 - root - INFO - Saving pictures as "2018-08-21/photobooth_shot_XXXXX.jpg"
2018-08-21 15:07:03,761 - root - INFO - Using printer "Canon_SELPHY_CP1300"
2018-08-21 15:07:06,598 - root - INFO - Using gphoto2 via command line
2018-08-21 15:07:13,863 - root - INFO - Using camera without preview functionality
2018-08-21 15:08:27,258 - root - INFO - Saving picture as 2018-08-21/photobooth00005.jpg
2018-08-21 15:09:48,635 - root - INFO - Saving picture as 2018-08-21/photobooth00006.jpg
2018-08-21 15:11:45,073 - root - INFO - Exiting photobooth with status code 0

reuterbal, 2018-08-21:
Is there a reason why you are using the command line wrapper instead of python-gphoto2?

Is the resolution also this bad when you look a the jpg pictures (e.g. `2018-08-21/photobooth00004.jpg`)?

---
### If this is the case, please try the following:

1. Replace the line
```
keep_pictures = False
```
by
```
keep_pictures = True
```

2. Have a look at the produced files (e.g. `2018-08-21/photobooth_shot_00000.jpg`). What resolution do they have?

---
### If the resolution is fine when you look at the jpg pictures:

Please try to print the jpg picture using the regular operating system programs. What is the resolution like then?

thyristor59, 2018-08-21:
so much! :
by changing, false by true I have an exclude quality
[image]

as you can see, the last image bug.
moreover I can make a draw normally then the application bug completely or tells me error messages at the second shoot.

[image]
[image]

I use commandeline because, it's dont fonctionne with other

thyristor59, 2018-08-21:
I put: keep_pictures = false.
I kept a perfect resolution on the printing and I have no problems!
[image]

next step: (I create a new topic?)

how to change the images, "hit this button, get ready ...." to have in french!
which program ? paint? photoshop?
I look to pay you a coffee once functional 😁😁😁

wikijm, 2018-08-21:
> how to change the images, "hit this button, get ready ...." to have in french!
which program ? paint? photoshop?

I provide you a solution for this. Have you tried it?
https://github.com/reuterbal/photobooth/issues/32#issuecomment-412667707

reuterbal, 2018-08-21:
Regarding the problems with the last picture: Please see #35 
This is fixed in the development branch.

reuterbal, 2018-08-21:
Regarding the problems with the last picture: Please see #35 
This is fixed in the development branch.

Regarding the translation question: I'm in the process of adding language selection, however, I have to figure out how to correctly distribute the translation files and didn't have time yet to do this.

wikijm, 2018-08-23:
I assume that your issue is now solved.
I close the issue 😃

closed 2018-08-23 by wikijm