reuterbal/photobooth #42: Camera [-7] I/O problem

DSX42, 2018-08-15:
Hi,
First of all : congrats for this work, it seems very complete. I really want to take my first shot with it ! :-)

**Describe the bug**

I've done every command in root, but it's still the same with the pi user.

In the beginning, my camera was on Av mode (aperture priority) but photobooth returns me an error with "Camera : bad config". I set my camera in auto mode, and now I have : "Camera : I/O problem".

If i take a picture with this command (out of photobooth environnement) : 
gphoto2 --capture-image-and-download
it works and I have the picture on th RPi.

**Hardware (please complete the following information)**

 - Device : Raspberry Pi 3 Model B Rev 1.2
 - Camera Canon EOS 7D Mark I
 - GPIO: Not yet

**Software (please complete the following information)**

 - OS Raspbian Stretch 9.4 up to date
 - Python version 3.5.3
 
 **Installed packages**
 
automationhat==0.1.0
blinker==1.3
blinkt==0.1.2
buttonshim==0.0.2
Cap1xxx==0.1.3
chardet==2.3.0
click==6.6
colorama==0.3.7
cryptography==1.7.1
drumhat==0.1.0
envirophat==1.0.0
ExplorerHAT==0.4.2
Flask==0.12.1
fourletterphat==0.1.0
gphoto2==1.8.3
gpiozero==1.4.1
idna==2.2
itsdangerous==0.24
jedi==0.10.2
Jinja2==2.8
keyring==10.1
keyrings.alt==1.3
MarkupSafe==0.23
microdotphat==0.2.1
mote==0.0.3
motephat==0.0.2
numpy==1.12.1
oauthlib==2.0.1
opencv-python==3.4.2.17
pantilthat==0.0.6
pgzero==1.2
phatbeat==0.1.1
-e git+https://github.com/reuterbal/photobooth.git@88202b9e3037bb3370217232ffbd3826d98e1265#egg=photobooth
pianohat==0.1.0
picamera==1.13
picraft==1.0
piglow==1.2.4
pigpio==1.38
Pillow==4.0.0
pkg-resources==0.0.0
pyasn1==0.1.9
pycrypto==2.6.1
pygame==1.9.3
pygobject==3.22.0
[... 584 lines cut ...]
2018-08-15 22:26:48,455 - gphoto2 - WARNING - (b'camera_init [library.c:7729]') b"'ptp_opensession (params, sessionid)' failed: PTP I/O Error (0x02ff)"
2018-08-15 22:26:49,959 - gphoto2 - WARNING - (b'gp_libusb1_write [libusb1.c:576]') b"'libusb_bulk_transfer (port->pl->dh, port->settings.usb.outep, (unsigned char*)bytes, size, &curwritten, port->timeout)' failed: Operation timed out (-7)"
2018-08-15 22:26:49,960 - gphoto2 - WARNING - (b'gp_libusb1_write [libusb1.c:576]') b"'libusb_bulk_transfer (port->pl->dh, port->settings.usb.outep, (unsigned char*)bytes, size, &curwritten, port->timeout)' failed: Operation timed out (-7)"
2018-08-15 22:26:49,962 - gphoto2 - WARNING - (b'gp_port_write [gphoto2-port.c:406]') b'Writing 16 = 0x10 bytes to port failed: Timeout reading from or writing to the port (-10)'
2018-08-15 22:26:49,963 - gphoto2 - WARNING - (b'gp_port_write [gphoto2-port.c:406]') b'Writing 16 = 0x10 bytes to port failed: Timeout reading from or writing to the port (-10)'
2018-08-15 22:26:49,966 - gphoto2 - WARNING - (b'ptp_usb_sendreq [usb.c:108]') b'PTP_OC 0x1002 sending req failed: Timeout reading from or writing to the port (-10)'
2018-08-15 22:26:49,967 - gphoto2 - WARNING - (b'ptp_usb_sendreq [usb.c:108]') b'PTP_OC 0x1002 sending req failed: Timeout reading from or writing to the port (-10)'
2018-08-15 22:26:49,968 - gphoto2 - WARNING - (b'camera_init [library.c:7729]') b"'ptp_opensession (params, sessionid)' failed: PTP I/O Error (0x02ff)"
2018-08-15 22:26:49,969 - gphoto2 - WARNING - (b'camera_init [library.c:7729]') b"'ptp_opensession (params, sessionid)' failed: PTP I/O Error (0x02ff)"
2018-08-15 22:26:51,472 - gphoto2 - WARNING - (b'gp_libusb1_write [libusb1.c:576]') b"'libusb_bulk_transfer (port->pl->dh, port->settings.usb.outep, (unsigned char*)bytes, size, &curwritten, port->timeout)' failed: Operation timed out (-7)"
2018-08-15 22:26:51,474 - gphoto2 - WARNING - (b'gp_libusb1_write [libusb1.c:576]') b"'libusb_bulk_transfer (port->pl->dh, port->settings.usb.outep, (unsigned char*)bytes, size, &curwritten, port->timeout)' failed: Operation timed out (-7)"
2018-08-15 22:26:51,475 - gphoto2 - WARNING - (b'gp_port_write [gphoto2-port.c:406]') b'Writing 16 = 0x10 bytes to port failed: Timeout reading from or writing to the port (-10)'
2018-08-15 22:26:51,476 - gphoto2 - WARNING - (b'gp_port_write [gphoto2-port.c:406]') b'Writing 16 = 0x10 bytes to port failed: Timeout reading from or writing to the port (-10)'
2018-08-15 22:26:51,477 - gphoto2 - WARNING - (b'ptp_usb_sendreq [usb.c:108]') b'PTP_OC 0x1002 sending req failed: Timeout reading from or writing to the port (-10)'
2018-08-15 22:26:51,478 - gphoto2 - WARNING - (b'ptp_usb_sendreq [usb.c:108]') b'PTP_OC 0x1002 sending req failed: Timeout reading from or writing to the port (-10)'
2018-08-15 22:26:51,479 - gphoto2 - WARNING - (b'camera_init [library.c:7729]') b"'ptp_opensession (params, sessionid)' failed: PTP I/O Error (0x02ff)"
2018-08-15 22:26:51,480 - gphoto2 - WARNING - (b'camera_init [library.c:7729]') b"'ptp_opensession (params, sessionid)' failed: PTP I/O Error (0x02ff)"
2018-08-15 22:26:51,481 - gphoto2 - WARNING - (b'camera_init [library.c:7757]') b"'ret' failed: 'PTP I/O Error' (0x02ff)"
2018-08-15 22:26:51,482 - gphoto2 - WARNING - (b'camera_init [library.c:7757]') b"'ret' failed: 'PTP I/O Error' (0x02ff)"
2018-08-15 22:26:51,483 - gphoto2 - WARNING - (b'gp_context_error') b'PTP I/O Error'
2018-08-15 22:26:51,484 - gphoto2 - WARNING - (b'gp_context_error') b'PTP I/O Error'
2018-08-15 22:26:51,497 - root - ERROR - Camera: [-7] I/O problem
2018-08-15 22:37:06,089 - root - INFO - Exiting photobooth with status code 0

Thanks a lot in advance

reuterbal, 2018-08-20:
Hi @DSX42 
thanks for reporting this issue. Could you please try the following:

Remove ll. 71-78 in file [`photobooth/camera/CameraGphoto2.py`](https://github.com/reuterbal/photobooth/blob/master/photobooth/camera/CameraGphoto2.py) and try if it works without these?

Sometimes, I receive 'Camera: I/O problem' after crashes etc. To fix these, I have to exit the photobooth, turn off the camera, unplug the USB cable, reconnect and turn everything back on.

reuterbal, 2018-10-19:
Closing this due to inactivity.

closed 2018-10-19 by reuterbal