reuterbal/photobooth #44: work without touch screen

botyas, 2018-08-17:
Your project is great, I am currently configuring it in a raspberry pi 3 and everything works perfectly, but the monitor that I have is not tactile and when doing the 4 photos I can not give the start button.

You could add the gpio trigger to move from this screen and add another gpio to print.

I feel my English.

Thank you very much.

reuterbal, 2018-08-20:
It is not necessary to press the "Start button" - it will move forward automatically after the configured time (called "greeter time").

botyas, 2018-08-20:
Ok thank you very much, I would like to add a button to print the images and I can not get it answered from gpio. I added in photobooth / gpio / _init_.py the port and even tried to add to be able to select it in the settings and there is no way it works.

Let's see if you can add it in a next revision.

Thank you for your work.

reuterbal, 2018-08-21:
You will also have to create the corresponding handler in the GUI that actually triggers the printing when the button is pressed. I will think of how to add more GPIO buttons in the future.