reuterbal/photobooth #50: Unable to install PIL/PILLOW

harpreetrekhi, 2018-08-26:
pi@raspberrypi:~/photobooth $ python3 -m photobooth
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/runpy.py", line 183, in _run_module_as_main
    mod_name, mod_spec, code = _get_module_details(mod_name, _Error)
  File "/usr/local/lib/python3.6/runpy.py", line 142, in _get_module_details
    return _get_module_details(pkg_main_name, error)
  File "/usr/local/lib/python3.6/runpy.py", line 109, in _get_module_details
    __import__(pkg_name)
  File "/home/pi/photobooth/photobooth/__init__.py", line 21, in <module>
    from .main import main
  File "/home/pi/photobooth/photobooth/main.py", line 30, in <module>
    from . import camera, gui
  File "/home/pi/photobooth/photobooth/camera/__init__.py", line 22, in <module>
    from PIL import Image, ImageOps
ModuleNotFoundError: No module named 'PIL'
pi@raspberrypi:~/photobooth $
pi@raspberrypi:~/photobooth $ ls -lrth
total 52M
-rw-r--r-- 1 pi pi 4.6K Aug 25 11:42 README.md
-rw-r--r-- 1 pi pi   67 Aug 25 11:42 MANIFEST.in
-rw-r--r-- 1 pi pi  34K Aug 25 11:42 LICENSE.txt
-rw-r--r-- 1 pi pi 8.4K Aug 25 11:42 INSTALL.md
drwxr-xr-x 2 pi pi 4.0K Aug 25 11:42 housing
-rw-r--r-- 1 pi pi 8.8K Aug 25 11:42 setup.py
drwxr-xr-x 2 pi pi 4.0K Aug 25 11:42 screenshots
-rw-r--r-- 1 pi pi  52M Aug 25 12:11 eos5dmk3-gphoto2.5.14.log
drwxr-xr-x 2 pi pi 4.0K Aug 25 12:13 photobooth.egg-info
drwxr-xr-x 2 pi pi 4.0K Aug 25 12:13 2018-08-25
-rw-r--r-- 1 pi pi 6.3K Aug 25 13:04 photobooth.log.2018-08-25
drwxr-xr-x 8 pi pi 4.0K Aug 25 20:41 photobooth
drwxr-xr-x 2 pi pi 4.0K Aug 26 16:29 2018-08-26
-rw-r--r-- 1 pi pi  714 Aug 26 16:29 photobooth.log
pi@raspberrypi:~/photobooth $
pi@raspberrypi:~/photobooth $
pi@raspberrypi:~/photobooth $ cat photobooth.log
2018-08-26 16:29:53,044 - root - INFO - Initializing photobooth
2018-08-26 16:29:53,047 - root - INFO - Photobooth version: 0.4.dev11+g4ff982c
2018-08-26 16:29:53,050 - root - INFO - Reading config file "/home/pi/photobooth/photobooth/defaults.cfg"
2018-08-26 16:29:53,063 - root - INFO - Reading config file "photobooth.cfg"
2018-08-26 16:29:53,146 - root - INFO - Number of last existing file: 0
2018-08-26 16:29:53,151 - root - INFO - Saving pictures as "2018-08-26/photoboothXXXXX.jpg"
2018-08-26 16:29:53,154 - root - INFO - GPIO disabled
2018-08-26 16:29:53,158 - root - INFO - Number of last existing file: 0
2018-08-26 16:29:53,159 - root - INFO - Saving pictures as "2018-08-26/photobooth_shot_XXXXX.jpg"
pi@raspberrypi:~/photobooth $
pi@raspberrypi:~/photobooth $
pi@raspberrypi:~/photobooth $ sudo pip install pillow
Exception:
Traceback (most recent call last):
  File "/usr/local/lib/python3.4/dist-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/usr/local/lib/python3.4/dist-packages/pip/commands/install.py", line 272, in run
    with self._build_session(options) as session:
  File "/usr/local/lib/python3.4/dist-packages/pip/basecommand.py", line 72, in _build_session
    insecure_hosts=options.trusted_hosts,
  File "/usr/local/lib/python3.4/dist-packages/pip/download.py", line 329, in __init__
    self.headers["User-Agent"] = user_agent()
  File "/usr/local/lib/python3.4/dist-packages/pip/download.py", line 93, in user_agent
    from pip._vendor import distro
  File "/usr/local/lib/python3.4/dist-packages/pip/_vendor/distro.py", line 1050, in <module>
    _distro = LinuxDistribution()
  File "/usr/local/lib/python3.4/dist-packages/pip/_vendor/distro.py", line 594, in __init__
    if include_lsb else {}
  File "/usr/local/lib/python3.4/dist-packages/pip/_vendor/distro.py", line 933, in _get_lsb_release_info
    raise subprocess.CalledProcessError(code, cmd, stdout)
subprocess.CalledProcessError: Command 'lsb_release -a' returned non-zero exit status 1
Traceback (most recent call last):
  File "/usr/local/bin/pip", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.4/dist-packages/pip/__init__.py", line 233, in main
    return command.main(cmd_args)
  File "/usr/local/lib/python3.4/dist-packages/pip/basecommand.py", line 251, in main
    timeout=min(5, options.timeout)) as session:
  File "/usr/local/lib/python3.4/dist-packages/pip/basecommand.py", line 72, in _build_session
    insecure_hosts=options.trusted_hosts,
  File "/usr/local/lib/python3.4/dist-packages/pip/download.py", line 329, in __init__
    self.headers["User-Agent"] = user_agent()
  File "/usr/local/lib/python3.4/dist-packages/pip/download.py", line 93, in user_agent
    from pip._vendor import distro
  File "/usr/local/lib/python3.4/dist-packages/pip/_vendor/distro.py", line 1050, in <module>
    _distro = LinuxDistribution()
  File "/usr/local/lib/python3.4/dist-packages/pip/_vendor/distro.py", line 594, in __init__
    if include_lsb else {}
  File "/usr/local/lib/python3.4/dist-packages/pip/_vendor/distro.py", line 933, in _get_lsb_release_info
    raise subprocess.CalledProcessError(code, cmd, stdout)
subprocess.CalledProcessError: Command 'lsb_release -a' returned non-zero exit status 1
pi@raspberrypi:~/photobooth $

harpreetrekhi, 2018-08-26:
pi@raspberrypi:~/photobooth $ sudo pip3 install pillow
Exception:
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/pip/basecommand.py", line 215, in main
    status = self.run(options, args)
  File "/usr/local/lib/python3.6/site-packages/pip/commands/install.py", line 272, in run
    with self._build_session(options) as session:
  File "/usr/local/lib/python3.6/site-packages/pip/basecommand.py", line 72, in _build_session
    insecure_hosts=options.trusted_hosts,
  File "/usr/local/lib/python3.6/site-packages/pip/download.py", line 329, in __init__
    self.headers["User-Agent"] = user_agent()
  File "/usr/local/lib/python3.6/site-packages/pip/download.py", line 93, in user_agent
    from pip._vendor import distro
  File "/usr/local/lib/python3.6/site-packages/pip/_vendor/distro.py", line 1050, in <module>
    _distro = LinuxDistribution()
  File "/usr/local/lib/python3.6/site-packages/pip/_vendor/distro.py", line 594, in __init__
    if include_lsb else {}
  File "/usr/local/lib/python3.6/site-packages/pip/_vendor/distro.py", line 931, in _get_lsb_release_info
    raise subprocess.CalledProcessError(code, cmd, stdout, stderr)
subprocess.CalledProcessError: Command 'lsb_release -a' returned non-zero exit status 1.
Traceback (most recent call last):
  File "/usr/local/bin/pip3", line 11, in <module>
    sys.exit(main())
  File "/usr/local/lib/python3.6/site-packages/pip/__init__.py", line 233, in main
    return command.main(cmd_args)
  File "/usr/local/lib/python3.6/site-packages/pip/basecommand.py", line 251, in main
    timeout=min(5, options.timeout)) as session:
  File "/usr/local/lib/python3.6/site-packages/pip/basecommand.py", line 72, in _build_session
    insecure_hosts=options.trusted_hosts,
  File "/usr/local/lib/python3.6/site-packages/pip/download.py", line 329, in __init__
    self.headers["User-Agent"] = user_agent()
  File "/usr/local/lib/python3.6/site-packages/pip/download.py", line 93, in user_agent
    from pip._vendor import distro
  File "/usr/local/lib/python3.6/site-packages/pip/_vendor/distro.py", line 1050, in <module>
    _distro = LinuxDistribution()
  File "/usr/local/lib/python3.6/site-packages/pip/_vendor/distro.py", line 594, in __init__
    if include_lsb else {}
  File "/usr/local/lib/python3.6/site-packages/pip/_vendor/distro.py", line 931, in _get_lsb_release_info
    raise subprocess.CalledProcessError(code, cmd, stdout, stderr)
subprocess.CalledProcessError: Command 'lsb_release -a' returned non-zero exit status 1.
pi@raspberrypi:~/photobooth $

wikijm, 2018-08-26:
A greeting is always appreciated at the beginning of a conversation. So I'd start by saying, "Hello @harpreetrekhi  😺 "

Could you provide more details please?
Presently, we know nothing about your system configuration (OS version in particular).

reuterbal, 2018-09-02:
Hi @harpreetrekhi 

you are trying to install everything system wide. I'd recommend to use a virtual environment and follow the [installation instructions](https://github.com/reuterbal/photobooth/blob/master/INSTALL.md).

closed 2018-09-02 by reuterbal