reuterbal/photobooth #51: Sounds during countdown

thyristor59, 2018-08-26:
I found thanks to pygame how to introduce music with python.

 I simply wrote this:

```
import pygame.mixer
pygame.init()

pygame.mixer.music.load("myFile.wav")
pygame.mixer.music.play()
```
play this: https://www.youtube.com/watch?v=G4fAO7UrTEs

what I would like is to have this "beep" at each count according to each count chosen by the user (me, 3 for example)
and
another sound while loading the final photo.

I would like to try alone but
Can you tell me which file I need to modify to succeed?

reuterbal, 2018-09-02:
Hi,

pygame is an option, however, I would choose a different approach, as pygame brings a whole new set of dependencies. Qt can play any kind of sounds and provides a number of classes for this. 
See https://doc.qt.io/qt-5/qmediaplayer.html#details as a starting point.

In [`photobooth/gui/Qt5Gui/Frames.py`](https://github.com/reuterbal/photobooth/blob/master/photobooth/gui/Qt5Gui/Frames.py) the appearance of the GUI is implemented. There is the class `CountdownMessage` where you can add the beep sound.

In the same file there is also the class `WaitMessage` where you can add the waiting music.

reuterbal, 2018-09-02:
I'm labelling this as "help wanted" and closing the issue as I'm not able to add this feature in the near future. Feel free to implement it and open a pull request!

For future reference, it is kept as "blocked" in the project page.

closed 2018-09-02 by reuterbal