reuterbal/photobooth #52: Question

carbs10, 2018-08-29:
I love the photobooth software I am new to rasberri pi. My question is how can i add a picture in one of the corners when the file prints. For example "my own design. Jpeg

wikijm, 2018-08-30:
A greeting is always appreciated at the beginning of a conversation. So I'd start by saying, "Hello @carbs10 😺 "

Please take a look in links below and give us a feedback:
https://github.com/reuterbal/photobooth/issues/43
https://github.com/reuterbal/photobooth/wiki/Tips-and-tricks

carbs10, 2018-08-30:
Im not the greatest on modifying files can you send me a sample? Thank you
for the response. I read the post but i dont know how to modify or hide the
picture

wikijm, 2018-08-30:
Without knowing the texts/images you want to add in the background, it is difficult to answer your request.

I may be old-fashioned, but I still think it is important in a dialogue to occasionally place "hello", "hi folks!" or "thank you". That remains my point of view and does not reflect in anything the thought of the creator of the project 😜

closed 2018-09-02 by reuterbal

reuterbal, 2018-09-04:
Hi @botyas 

The background-image is used instead of the white background. You should use a 2x2 grid and select 'omit last picture', then only three shots are taken and the last image is left out. In its position you can add a logo or something similar in the background picture.

Note that this applies to the development version, it is not yet available in any of the releases.

botyas, 2018-09-05:
How can I get the version in development? Well I could help you how to beta tester if you need it.

reuterbal, 2018-09-05:
Please follow the instructions [here](https://github.com/reuterbal/photobooth/issues/35#issuecomment-414638507) or [here](https://github.com/reuterbal/photobooth/issues/35#issuecomment-414665463).