reuterbal/photobooth #53: Texte layout

thyristor59, 2018-09-01:
always looking for progression, can you help me?
I will use the raspberry in portrait mode.
however, the text is no longer in the correct position

[image]
[image]

I would like the "hit this" (appuyez) be at the top and "botton" (ici)  in the center of the screen ...
likewise at the end of the shoot,
the function "print" or "start again"
I would like them to be in the middle of the screen
when I'm in portrait mode
is it possible ?
thank you in advance

thyristor59, 2018-09-01:
it's good I found, file modification / Home / pi / photobooth / photobooth / mistletoe / qt5gui / stylesheets
....
after a good two hours I had lost hope

Sorry

thyristor59, 2018-09-01:
https://community.risevision.com/rise_vision_inc/topics/how_can_i_use_raspberry_pi_on_my_portrait_monitors

a simple little search on google 😉😉

carbs10, 2018-09-01:
Thank you thyristor59

I will try that code. If you ever figure out how to add a code to have a
image/Logo in the background when all 4 photos are updated  at the end of
the photobooth let me know. Ive been trying to add but im lost in doing so

thyristor59, 2018-09-01:
I'm completely new, for the moment I'm trying to include sound for animation ...
it is a project also for the background image, I would send if I get there

carbs10, 2018-09-01:
Hello

Yes i would like to see your finish product when your done

Thanks

reuterbal, 2018-09-02:
Hi @thyristor59 

you found the correct files for modifying the appearance. In [`photobooth/gui/Qt5Gui/stylesheets`](https://github.com/reuterbal/photobooth/tree/master/photobooth/gui/Qt5Gui/stylesheets) are the styleseheets for either of the layouts. Simply edit these files or, even better, add a new one and make it available in [`photobooth/gui/Qt5Gui/__init__.py`](https://github.com/reuterbal/photobooth/blob/master/photobooth/gui/Qt5Gui/__init__.py).
The syntax is more or less identical to CSS (as known from websites).

Regarding the sound questions, see your other issue.

closed 2018-09-02 by reuterbal