reuterbal/photobooth #54: Wedding Photobooth Success

jkruszynski, 2018-09-06:
**Describe your photobooth**
I recently used this software with some modifications for a friends wedding. Everything went great! My hardware was just a DSLR, 7" rpi touch screen attached to my raspberry pi. My goal was to set it up, and not have to worry about it the rest of the night, and it worked as I had hoped. 

**Hardware (please complete the following information):**
 - Device: RPi 3B+
 - Camera Canon EOS 70D
 - GPIO: No
 
 **Additional equipment**
Flash on camera

**Software (please complete the following information):**
 - Raspbian
 - Python 3.6

**Modifications:**
For the wedding, I added 2 main photos: Google Photos integration and adding a footer for a wedding hashtag. 
* Google Photos integration: The purpose for this was to upload the pictures during the wedding to a google photos album so guests could download them almost instantly.
* Footer: I added a footer space at the bottom of each assembled picture and put the wedding hashtag for the couple

Since then, I have added the ability to change the background (from plain white to a custom one created by me) and change to have odd number of photos, such as 3. The purpose of this is to have 2 photos in the top row, 1 photo on the left/right in the bottom row, and the empty space would have a wedding watermark/logo. 

Going forward, it's mostly just tweaking different things, such as photo spacing. Also want to interact with some other API's for a better photo storage option. Some of the 

**Problems:**
My main problems were not with the photobooth software, but working with different API's. I tried a few photo gallery specific websites that provided API, but I could not get them working for some reason. Was kind of a last minute switch to Google Photos, but it worked out well in the end. Also didn't get the lighting I had hoped, nor the angle I wanted. So upgrading my lighting and adding a wide angle lens to my inventory going forward. 

**Pictures:**
Sample assembled pic: 
[image]

carbs10, 2018-09-06:
Hi There
Great Job, Can you help with the code to get the background please, rose82192@hotmail.com

reuterbal, 2018-09-06:
Hi @jkruszynski 

thanks for your feedback and I'm glad the app proved useful for you!

I will have a look at your implementation of the Google Photo upload when starting to work on postprocessing modules. I'm sure it will prove useful!

Support for background pictures and (m x n - 1) pictures is already implemented in the development branch, see for example the question by @carbs10 in #52 
@carbs10: Is there a reason why you don't want to use this solution?

carbs10, 2018-09-06:
@reuterbal

Hello there

I would love to use any solution to add photos, unfortunatly I dont know how to read or write code, I was able yo run your files but thats about it..... im trying to learn but its a bit difficult. 

Thank you

jkruszynski, 2018-09-06:
@reuterbal 

Thanks, it's been a fun project. I'm a semi newbie when it comes to Python, only within the past year or so. So don't have much experience with threading and the worker design you have. And I let this project creep up on me, so I'm sure some of the implementations could be cleaner. The one issue I am having with the google photos implementation is that when I close out the app, the GUI shuts down fine, but continues to run in the command prompt. Something about the google threads is hanging it up, so I just need to learn more about the threading stuff to stop that.

wikijm, 2018-11-28:
Hi @jkruszynski,

You'll be able to see you 'Success story' on the wiki now 😃
Thanks for your feedback 👍

As this is included in the Wiki now, I'll close this.

closed 2018-11-28 by wikijm

MarioJanssen27, 2019-02-06:
Which settings did you use for the collage?

in deutsch

Welche Einstellungen habt ihr für die Collage verwendet?