reuterbal/photobooth #55: Camera features issue

harpreetrekhi, 2018-09-18:
Hello Team Mates,

I was successful in installing and testing this photobooth with gphoto2-commandline interface. However if I am trying to use python-gphoto2 mode then the application is crashing with following errror.

I am attempting this on raspberry pi 2b with Nikon D5200 DSLR.
However, gphoto2-commandline mode works fine.

pi@raspberrypi:~/photobooth $ python3.5 -m photobooth
2018-09-18 16:02:42,878 - root - INFO - Initializing photobooth
2018-09-18 16:02:42,880 - root - INFO - Photobooth version: 0.4.dev11+g4ff982c
2018-09-18 16:02:42,887 - root - INFO - Reading config file "/home/pi/photobooth/photobooth/defaults.cfg"
2018-09-18 16:02:42,897 - root - INFO - Reading config file "photobooth.cfg"
2018-09-18 16:02:43,027 - root - INFO - GPIO disabled
2018-09-18 16:02:43,043 - root - INFO - Number of last existing file: 4
2018-09-18 16:02:43,055 - root - INFO - Saving pictures as "2018-09-18/photoboothXXXXX.jpg"
2018-09-18 16:02:43,067 - root - INFO - Number of last existing file: 0
2018-09-18 16:02:43,073 - root - INFO - Saving pictures as "2018-09-18/photobooth_shot_XXXXX.jpg"
QXcbConnection: Failed to initialize XRandr
qt5ct: using qt5ct plugin
qt5ct: custom style sheet is disabled
2018-09-18 16:02:58,703 - root - INFO - Using python-gphoto2 bindings
2018-09-18 16:02:59,790 - root - INFO - Camera summary: Manufacturer: Nikon Corporation
Model: D5200
  Version: V1.03
  Serial Number: 64534310000000000000000000000000
Vendor Extension ID: 0xa (1.0)
Vendor Extension Description: microsoft.com: 1.0

Capture Formats: JPEG Undefined Type
Display Formats: Undefined Type, Association/Directory, Script, DPOF, Apple Quicktime, JPEG

Device Capabilities:
	File Download, File Deletion, File Upload
	Generic Image Capture, No Open Capture, Nikon Capture 1, Nikon Capture 2, Nikon Capture 3

Storage Devices Summary:
store_00010001:
	StorageDescription: None
	VolumeLabel:  [Slot 1]

	Storage Type: Removable RAM (memory card)
	Filesystemtype: Digital Camera Layout (DCIM)
	Access Capability: Read Only with Object deletion
	Maximum Capability: 15923150848 (15185 MB)
	Free Space (Bytes): 12667781120 (12080 MB)
	Free Space (Images): 810

Device Property Summary:
Battery Level(0x5001):(read only) (type=0x2) Range [0 - 100, step 1] value: 100% (100)
Image Size(0x5003):(readwrite) (type=0xffff) Enumeration [
	'6000x4000',
	'4496x3000',
	'2992x2000'
	] value: '6000x4000'
Compression Setting(0x5004):(readwrite) (type=0x2) Enumeration [0,1,2,4,5,6,7] value: JPEG Fine (2)
White Balance(0x5005):(readwrite) (type=0x4) Enumeration [2,4,5,6,7,32784,32785,32787] value: Automatic (2)
F-Number(0x5007):(read only) (type=0x4) Enumeration [560,630,710,800,900,1000,1100,1300,1400,1600,1800,2000,2200,2500,2900,3200,3600] value: f/5.6 (560)
Focal Length(0x5008):(read only) (type=0x6) Range [1800 - 5500, step 1] value: 55 mm (5500)
Focus Mode(0x500a):(read only) (type=0x4) Enumeration [1,32784,32785,32786,32787] value: AF-A (32786)
Exposure Metering Mode(0x500b):(readwrite) (type=0x4) Enumeration [2,3,4] value: Multi-spot (3)
Flash Mode(0x500c):(readwrite) (type=0x4) Enumeration [2,4,32784] value: Auto (32784)
Exposure Time(0x500d):(read only) (type=0x6) Enumeration [2,3,4,5,6,8,10,12,15,20,25,31,40,50,62,80,100,125,166,200,250,333,400,500,666,769,1000,1250,1666,2000,2500,3333,4000,5000,6250,7692,10000,13000,16000,20000,25000,30000,40000,50000,60000,80000,100000,130000,150000,200000,250000,300000] value: 0.0033 sec (333)
Exposure Program Mode(0x500e):(read only) (type=0x4) Enumeration [1,2,3,4,32784,32785,32786,32787,32788,32790,32791,32792,32793] value: Auto (32784)
Exposure Index (film speed ISO)(0x500f):(readwrite) (type=0x4) Enumeration [100,125,160,200,250,320,400,500,640,800,1000,1250,1600,2000,2500,3200,4000,5000,6400,8000,10000,12800,25600] value: ISO 100 (100)
Exposure Bias Compensation(0x5010):(readwrite) (type=0x3) Enumeration [-5000,-4666,-4333,-4000,-3666,-3333,-3000,-2666,-2333,-2000,-1666,-1333,-1000,-666,-333,0,333,666,1000,1333,1666,2000,2333,2666,3000,3333,3666,4000,4333,4666,5000] value: 0.0 stops (0)
Date & Time(0x5011):(readwrite) (type=0xffff) '20180918T160737'
Still Capture Mode(0x5013):(readwrite) (type=0x4) Enumeration [1,2,32784,32785,32788,32789,32790] value: Single Shot (1)
Burst Number(0x5018):(readwrite) (type=0x4) Range [1 - 100, step 1] value: 1
Focus Metering Mode(0x501c):(readwrite) (type=0x4) Enumeration [2,32784,32785,32786,32787,32788] value: Closest Subject (32785)
Property 0xd303:(read only) (type=0x2) 1
[... 78 lines cut ...]
Live View Status(0xd1a2):(read only) (type=0x2) Range [0 - 1, step 1] value: No (0)
Live View Image Zoom Ratio(0xd1a3):(readwrite) (type=0x2) Range [0 - 5, step 1] value: 0
Live View Prohibit Condition(0xd1a4):(read only) (type=0x6) 2147483648
Property 0xd1ac:(readwrite) (type=0x2) Enumeration [1,2] value: 2
Exposure Display Status(0xd1b0):(read only) (type=0x2) Range [0 - 3, step 1] value: No (0)
Exposure Indicate Status(0xd1b1):(read only) (type=0x1) Range [-60 - 60, step 1] value: 0.0 stops (0)
Info Display Error Status(0xd1b2):(read only) (type=0x2) Range [0 - 1, step 1] value: No (0)
Exposure Indicate Lightup(0xd1b3):(read only) (type=0x2) Range [0 - 1, step 1] value: Yes (1)
Property 0xd1b4:(read only) (type=0x2) Range [0 - 96, step 1] value: 11
Property 0xd1b5:(read only) (type=0x4) Enumeration [0,32784,32785,32786,32787,32800] value: 0
Flash Open(0xd1c0):(read only) (type=0x2) Range [0 - 1, step 1] value: No (0)
Flash Charged(0xd1c1):(read only) (type=0x2) Range [0 - 1, step 1] value: Yes (1)
Property 0xd1f0:(readwrite) (type=0x2) Range [0 - 1, step 1] value: 0
Property 0xd1f1:(read only) (type=0x4) Range [0 - 65535, step 1] value: 810
Property 0xd1f4:(readwrite) (type=0x1) Range [-2 - 2, step 1] value: 0
Active Pic Ctrl Item(0xd200):(readwrite) (type=0x4) Enumeration [1,2,3,4,5,6,201,202,203,204,205,206,207,208,209] value: 1
Change Pic Ctrl Item(0xd201):(read only) (type=0x4) Enumeration [] value: 0

2018-09-18 16:02:59,818 - gphoto2 - WARNING - (b'_get_config [config.c:7471]') b"Type of property 'ISO Auto Hi Limit' expected: 0x0001 got: 0x0002"
2018-09-18 16:02:59,821 - gphoto2 - WARNING - (b'_get_config [config.c:7471]') b"Type of property 'Active D-Lighting' expected: 0x0001 got: 0x0002"
2018-09-18 16:02:59,823 - gphoto2 - WARNING - (b'_get_config [config.c:7471]') b"Type of property 'High ISO Noise Reduction' expected: 0x0001 got: 0x0002"
2018-09-18 16:02:59,830 - gphoto2 - WARNING - (b'_get_config [config.c:7471]') b"Type of property 'Flash Exposure Compensation' expected: 0x0002 got: 0x0001"
2018-09-18 16:02:59,966 - root - ERROR - Camera: [-2] Bad parameters
2018-09-18 16:03:24,356 - root - INFO - Exiting photobooth with status code 0
pi@raspberrypi:~/photobooth $

reuterbal, 2018-09-21:
Hi @harpreetrekhi 

I haven't had the chance to try any Nikon cameras yet, so there might be issues.
At which point do the problems happen? During startup or when you are trying to take a picture?

botyas, 2018-09-23:
try to update gphoto2 with this script

wget https://raw.githubusercontent.com/gonzalo/gphoto2-updater/master/gphoto2-updater.sh && chmod +x gphoto2-updater.sh && sudo ./gphoto2-updater.sh

harpreetrekhi, 2018-09-24:
Hi @reuterbal  the problem occurs during startup.

Okay thanks @botyas , I will try to update this & will revert.

reuterbal, 2018-09-24:
If this doesn't work, you could also try using the gphoto2-cffi bindings - they are a bit slower but use default values for everything. This way we get some feeling whether the problem is buried inside the gphoto2-API or somewhere else.

melangeaddict, 2018-10-12:
@harpreetrekhi I encountered the same problem.  I found that my camera (Canon G7) had issues with all of the 'set_config' functions.  I commented them out of the 'CameraGphoto2.py' file, as well as lines 71-79 and the program proceeded as expected.

reuterbal, 2018-10-16:
That's a good point at @melangeaddict, I should make these optional/resilient.

Woodhomer, 2018-11-11:
@harpreetrekhi 

> Hi @reuterbal the problem occurs during startup.
> 
> Okay thanks @botyas , I will try to update this & will revert.

Hi there,
Did you get it your Nikon D5200 working? I have the same issue with the same camera.

reuterbal, 2018-11-11:
Have you tried the development branch? The issue should be fixed there.

Woodhomer, 2018-11-12:
Thanks for your Reply. Didn't tried this yet because i don't how to proceed.

Can you help me out?

I installed the latest version with gphoto2-cffi bindings.

Woodhomer, 2018-11-12:
> Have you tried the development branch? The issue should be fixed there.

Thanks for your Reply. Didn't tried this yet because i don't how to proceed.

Can you help me out?

I installed the latest version with gphoto2-cffi bindings

reuterbal, 2018-11-12:
Please see https://github.com/reuterbal/photobooth/wiki/Tips-and-tricks#switch-to-development-branch for some advice on how to obtain the development branch.

Woodhomer, 2018-11-12:
Many thanks, i will test this later.

reuterbal, 2018-11-18:
Were you able to get it working?

botyas, 2018-11-18:
they have left me a Nikon D40 camera and although in the list of gphoto2 it appears as compatible, I was surprised that when starting the program it does not respond. with my Canon 60d when starting it makes an opening and it remains in wait but with the Nikon it does not do anything. I have tried with all modes of gphoto2 and all have the same behavior

reuterbal, 2018-11-18:
Are you able to use it with the gphoto2-application on command line?

botyas, 2018-11-18:
tomorrow I will do some more test. now I have only connected it to see if it worked

Woodhomer, 2018-11-19:
@reuterbal: i was not able to get it working. I switched to the development branch but that didn't work. one error had been displayed as "falsches format". Gphoto2-commandline worked but because i need an live-view i now switched to an x86-hardware with another software.

But thanks for your help!

reuterbal, 2018-11-19:
@Woodhomer thank you for taking the time to report back! It would be useful to include the entire `photobooth.log, otherwise it is just guess work what might be going wrong.

reuterbal, 2019-01-10:
Closing this due to inactivity. But I'm interested in experiences with Nikon cameras. Most people seem to use Consumer level Canon cameras.

reuterbal, 2019-01-12:
Hi @TomCawthorn 
this is really helpful feedback. It is particularly noteworthy that it takes a picture before producing the error, thus I assume it is related to the config changes in the setIdle/setActive routines, which are required on Canon cameras to ensure the mirror flips back down and the sensor is turned off.

I applied some changes to the development branch, maybe you could try and see if that fixes this issue?

TomCawthorn, 2019-01-13:
Hey @reuterbal,

Thanks for the speedy reply!

I've just given it a quick demo and everything is now working as expected! 🎊 🎊  Thanks for making the update.

A quick note about preview performance - the preview is fairly glitchy and low quality (but good enough to get by). Have you got any suggestions as to which parameters within the code and on the camera I can play with to improve it? Perhaps using an alternative camera module (like GPHOT2-CFFI)?

Thanks,

Tom

TomCawthorn, 2019-01-13:
And just FYI here's my output which includes:
- Startup
- Initialising the picture taking process with default settings (4 pictures)

```

2019-01-13 13:39:46,164 - root - INFO - Initializing photobooth
2019-01-13 13:39:46,165 - root - INFO - Photobooth version: 0.4.dev76+g1b152c6
2019-01-13 13:39:46,167 - root - INFO - Reading config file "/home/pi/Projects/photobooth-proj/photobooth/photobooth/defaults.cfg"
2019-01-13 13:39:46,171 - root - INFO - Reading config file "photobooth.cfg"
2019-01-13 13:39:46,220 - root - INFO - Number of last existing file: 2
2019-01-13 13:39:46,223 - root - INFO - Saving pictures as "2019-01-13/photoboothXXXXX.jpg"
2019-01-13 13:39:46,226 - root - INFO - Number of last existing file: 0
2019-01-13 13:39:46,227 - root - INFO - Saving pictures as "2019-01-13/photobooth_shot_XXXXX.jpg"
2019-01-13 13:39:46,232 - root - INFO - GPIO disabled
libEGL warning: DRI2: failed to authenticate
qt5ct: using qt5ct plugin
qt5ct: custom style sheet is disabled
inotify_add_watch("/home/pi/.config/qt5ct") failed: "No such file or directory"
2019-01-13 13:39:50,012 - root - INFO - Using python-gphoto2 bindings
2019-01-13 13:39:50,190 - gphoto2 - WARNING - (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x1002 receiving resp failed: PTP Session Already Opened (0x201e)'
2019-01-13 13:39:50,191 - gphoto2 - WARNING - (b'camera_init [library.c:7729]') b"'ptp_opensession (params, sessionid)' failed: PTP Session Already Opened (0x201e)"
2019-01-13 13:39:50,840 - root - INFO - Camera summary: Manufacturer: Nikon Corporation
Model: D5200
  Version: V1.02
  Serial Number: 46719130000000000000000000000000
Vendor Extension ID: 0xa (1.0)
Vendor Extension Description: microsoft.com: 1.0

Capture Formats: JPEG Undefined Type
Display Formats: Undefined Type, Association/Directory, Script, DPOF, Apple Quicktime, JPEG

Device Capabilities:
	File Download, File Deletion, File Upload
	Generic Image Capture, No Open Capture, Nikon Capture 1, Nikon Capture 2, Nikon Capture 3

Storage Devices Summary:
store_00010001:
	StorageDescription: None
	VolumeLabel: NIKON D5200 [Slot 1]

	Storage Type: Removable RAM (memory card)
	Filesystemtype: Digital Camera Layout (DCIM)
	Access Capability: Read Only with Object deletion
	Maximum Capability: 15923150848 (15185 MB)
	Free Space (Bytes): 15923019776 (15185 MB)
	Free Space (Images): 1018

Device Property Summary:
Battery Level(0x5001):(read only) (type=0x2) Range [0 - 100, step 1] value: 20% (20)
Image Size(0x5003):(readwrite) (type=0xffff) Enumeration [
	'6000x4000',
	'4496x3000',
	'2992x2000'
	] value: '6000x4000'
Compression Setting(0x5004):(readwrite) (type=0x2) Enumeration [0,1,2,4,5,6,7] value: JPEG Fine (2)
White Balance(0x5005):(readwrite) (type=0x4) Enumeration [2,4,5,6,7,32784,32785,32787] value: Automatic (2)
F-Number(0x5007):(read only) (type=0x4) Enumeration [350,400,450,500,560,630,710,800,900,1000,1100,1300,1400,1600,1800,2000,2200] value: f/3.5 (350)
Focal Length(0x5008):(read only) (type=0x6) Range [1800 - 5500, step 1] value: 18 mm (1800)
Focus Mode(0x500a):(read only) (type=0x4) Enumeration [1,32784,32785,32786,32787] value: AF-S (32784)
Exposure Metering Mode(0x500b):(readwrite) (type=0x4) Enumeration [2,3,4] value: Multi-spot (3)
Flash Mode(0x500c):(readwrite) (type=0x4) Enumeration [2,4,32784] value: Auto (32784)
Exposure Time(0x500d):(read only) (type=0x6) Enumeration [50,62,80,100,125,166] value: 0.0017 sec (166)
Exposure Program Mode(0x500e):(readwrite) (type=0x4) Enumeration [1,2,3,4,32784,32785,32786,32787,32788,32790,32791,32792,32793] value: Auto (32784)
Exposure Index (film speed ISO)(0x500f):(readwrite) (type=0x4) Enumeration [100,125,160,200,250,320,400,500,640,800,1000,1250,1600,2000,2500,3200,4000,5000,6400,8000,10000,12800,25600] value: ISO 100 (100)
Exposure Bias Compensation(0x5010):(readwrite) (type=0x3) Enumeration [-5000,-4666,-4333,-4000,-3666,-3333,-3000,-2666,-2333,-2000,-1666,-1333,-1000,-666,-333,0,333,666,1000,1333,1666,2000,2333,2666,3000,3333,3666,4000,4333,4666,5000] value: 0.0 stops (0)
Date & Time(0x5011):(readwrite) (type=0xffff) '20190113T145721'
Still Capture Mode(0x5013):(readwrite) (type=0x4) Enumeration [1,2,32784,32785,32788,32789,32790] value: Single Shot (1)
Burst Number(0x5018):(readwrite) (type=0x4) Range [1 - 100, step 1] value: 1
Focus Metering Mode(0x501c):(readwrite) (type=0x4) Enumeration [2,32784,32785,32786,32787,32788] value: Closest Subject (32785)
[... 457 lines cut ...]
2019-01-13 13:42:18,058 - gphoto2 - WARNING - (b'_get_config [config.c:7471]') b"Type of property 'ISO Auto Hi Limit' expected: 0x0001 got: 0x0002"
2019-01-13 13:42:18,063 - gphoto2 - WARNING - (b'_get_config [config.c:7471]') b"Type of property 'Active D-Lighting' expected: 0x0001 got: 0x0002"
2019-01-13 13:42:18,067 - gphoto2 - WARNING - (b'_get_config [config.c:7471]') b"Type of property 'High ISO Noise Reduction' expected: 0x0001 got: 0x0002"
2019-01-13 13:42:18,223 - gphoto2 - WARNING - (b'_get_config [config.c:7471]') b"Type of property 'Flash Exposure Compensation' expected: 0x0002 got: 0x0001"
2019-01-13 13:42:18,518 - root - WARNING - Error while setting camera output to idle: [-2] Bad parameters.
2019-01-13 13:42:18,583 - gphoto2 - WARNING - (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
2019-01-13 13:42:18,686 - gphoto2 - WARNING - (b'ptp_usb_getresp [usb.c:465]') b'PTP_OC 0x90c8 receiving resp failed: PTP Device Busy (0x2019)'
[... cut ...]
2019-01-13 13:42:26,210 - gphoto2 - WARNING - (b'_get_config [config.c:7471]') b"Type of property 'High ISO Noise Reduction' expected: 0x0001 got: 0x0002"
2019-01-13 13:42:26,213 - gphoto2 - WARNING - (b'_get_config [config.c:7471]') b"Type of property 'Flash Exposure Compensation' expected: 0x0002 got: 0x0001"
2019-01-13 13:42:26,220 - root - WARNING - Error while setting camera output to idle: [-2] Bad parameters.
2019-01-13 13:42:30,527 - root - INFO - Saving picture as 2019-01-13/photobooth00003.jpg

```

reuterbal, 2019-01-14:
My impression is that the performance of the preview is largely limited by the poor USB performance of the Raspberry Pi. In my experience, gphoto2-cffi actually performs worse than python-gphoto2.

The error message is due to the configuration setting needed for Canon cameras. I haven't removed it but instead ignore when it fails.
Eventually, there should be some more generic approach of configuration modifications for different camera models, that is more portable.

reuterbal, 2019-01-14:
Closing this as the underlying issue is resolved.

commit 1615f658 (2018-10-19): Enclosed changes to camera settings in try-catch block. Fixes #55
commit 7df2fd66 (2019-01-12): Enclosed config changes in setActive/setIdle in try-catch-blocks to avoid crashes with Nikon cameras (fixes #55)
closed 2019-01-14 by reuterbal