reuterbal/photobooth #56: [Printer Issue] Canon Selphy CP910

eruame, 2018-09-18:
**Describe the bug**
Hi! First a big thank you for this great piece of software! I'm planning to use this on my wedding. But i have the following issue printing images with a Canon Selphy CP910.

I have installed the printer with CUPS. It is the default printer and the settings should be correct.
[image]

I can print from the command line with a simple
`lp photobooth00001.jpg`

The picture is printed correctly.

However, if i want to print from the photobooth app, no file is forwarded to the job list of the printer. Not in the "active" tab in CUPS nor the "all jobs" tab.

Settings in Photobooth:
[image]

**Hardware**

 - Sony Vaio Laptop
 - Camera: Canon EOS 500D
 - GPIO: No
- Printer: Canon Selphy CP910

**Software**

 - linux mint 18.3
 - Python version 3.6.3

**Photobooth log**

[photobooth.log](https://github.com/reuterbal/photobooth/files/2392526/photobooth.log)

**Photobooth config**

```
[Gui]
module = PyQt5
fullscreen = True
width = 1024
height = 600
hide_cursor = False
style = pastel

[Camera]
module = python-gphoto2

[Gpio]
enable = False
exit_pin = 24
trigger_pin = 23
lamp_pin = 4

[Printer]
enable = True
pdf = False
confirmation = True
module = PyQt5
width = 148
height = 100

[Photobooth]
show_preview = True
greeter_time = 3
countdown_time = 4
display_time = 5
postprocess_time = 60

[Picture]
basedir = %Y-%m-%d
basename = photobooth
num_x = 1
num_y = 1
size_x = 3496
size_y = 2362
min_dist_x = 20
min_dist_y = 20
keep_pictures = False
```

Can you help me? What am i missing here?

reuterbal, 2018-09-21:
Hi @eruame 

if you can print from command line then the drivers are fine, i.e., Gutenprint 5.2.11 is sufficient.
However, in your `photobooth.log` the following line
```
2018-09-18 12:25:20,957 - root - INFO - Using printer ""
```
suggests that Qt doesn't find a default printer.

First, I would try making the printer the default printer again (just repeat the steps) and test again.
You should also make sure that the printer is plugged in and turned on before starting the photobooth. This was never an issue for me but well, with printing you never know.

If that doesn't help, you will have to do a little work to dig deeper:
In `photobooth/printer/PrinterPyQt5.py` replace l. 23 by the following:
```python
from PyQt5.QtPrintSupport import QPrinter, QPrinterInfo
```
and insert the following after l. 39:
```
logging.info('Available Printers: {}'.format(QPrinterInfo.availablePrinterNames()))
logging.info('Default Printer: {}'.format(QPrinterInfo.defaultPrinterName())
```

Please post `photobooth.log` again afterwards.

eruame, 2018-09-21:
Hi, thank you for your time and reply.

It seems that Qt5 really cannot find printers. That is, it does not find any printer (I have two printers installed).

However, the problem was of course on my side. For an university project I installed ANACONDA. Which has its own python3 environemt. When issuing the command
```
virtualenv -p python3 --system-site-packages .venv
```
virtualenv decided it was a good idea to use the ANACONDA binaries. Following your tutorial, I installed all packages etc in my normal python installation. Therefore, inside of the ANACONDA environment, I had no full PyQt5 installation. I fixed this the dirty way with this (I do not care to reinstall the Photobooth virtualenv with the correct python3 binarie, to be honest. It's working now -- never touch a running system ;)):
```
source .venv/bin/activate
pip3 install P

closed 2018-09-21 by eruame

reuterbal, 2018-09-24:
I'm glad it is working now. Thanks for taking the time to report back the fix.