reuterbal/photobooth #59: Adding background template/picture

rhysloh, 2018-09-30:
Sorry for being such a noob here, but i really am unable to figure out how to add a background template or photo frame. Please, can anyone give me some instructions?

botyas, 2018-10-01:
In development version you have the option to upload a background image. there is also another check to suppress the last image to be able to place a larger logo in the bottom right position

rhysloh, 2018-10-03:
can you please advise how to make the captured picture fit into the background image? what are the parameters to change?

reuterbal, 2018-10-16:
At the moment, you can only modify the minimum distance between the captured pictures in each direction. The same distance is kept between the border of the final image and the captured pictures.
As many cameras produce pictures with different aspect ratios, this was the simplest way to provide a flexible and well-working solution.

I found it easiest to use the photobooth app to create a sample picture without any background. Then, I use this in GIMP as a separate layer to get the correct image size and align the background correctly, such that everything looks fine. After disabling the sample picture layer I export everything as a JPEG picture and use this as a background image.

rhysloh, 2018-10-17:
thank you very much for the clarification. I used the booth with success last weekend for a friend's gathering and did came up with a slight problem though. Could you consider adding an option to increase the default print copy to 2 or more? currently it's only 1. And photobooth pics are taken in groups, when everyone wants a print, its kinda hard to keep reprinting.