reuterbal/photobooth #60: Does not work with Canon EOS 1000D

jopaing, 2018-10-18:
**Does not work with Canon EOS 1000D**

Tried all camera drivers, but i get different error messages. It seems that there is an issue with camera communication.

First it did not find the camera via "gphoto2 --auto-detect"
Then i reinstalled gphoto2 via Golzalo-gphoto2-updater.

Now it works on shell via "gphoto2 --auto-detect" and "gphoto2 --capture-image-and-download"

For this reason i will only attach the log with gphoto2 command-line setting.

**To Reproduce**

Steps to reproduce the behavior:
1. Try to take photo

**Expected behavior**

take photo

**Screenshots/Screencast**

get popup with message:
Camera: [Errno 2] No such file or directory: '/dev/shm/photobooth.jpg'

**Hardware **

 - Device Raspberry Pi 3
 - Camera Canon EOS 1000D
 - GPIO: Yes

**Software **

 - OS Raspbian Stretch
 - Python version  3.5.3
 
 **Installed packages**
 
 ```
automationhat==0.1.0
blinker==1.3
blinkt==0.1.2
buttonshim==0.0.2
Cap1xxx==0.1.3
cffi==1.11.5
chardet==2.3.0
click==6.6
colorama==0.3.7
coverage==4.2
cryptography==1.7.1
drumhat==0.1.0
empy==3.3.2
envirophat==1.0.0
ExplorerHAT==0.4.2
Flask==0.12.1
fourletterphat==0.1.0
gphoto2==1.8.3
gphoto2-cffi==0.4.2
gpiozero==1.4.1
idna==2.2
itsdangerous==0.24
jedi==0.10.2
Jinja2==2.8
keyring==10.1
keyrings.alt==1.3
MarkupSafe==0.23
mcpi==0.1.1
microdotphat==0.2.1
mock==2.0.0
mote==0.0.3
motephat==0.0.2
nose==1.3.7
numpy==1.12.1
oauthlib==2.0.1
opencv-python==3.4.3.18
pantilthat==0.0.6
pbr==1.10.0
pep8==1.7.0
pgzero==1.2
phatbeat==0.1.1
-e git+https://github.com/reuterbal/photobooth.git@4ff982c95151ef1b7d1ceaab2a6bd72bcfe79b51#egg=photobooth
pianohat==0.1.0
picamera==1.13
picraft==1.0
piglow==1.2.4
pigpio==1.38
Pillow==4.0.0
pkg-resources==0.0.0
pyasn1==0.1.9
pycparser==2.19
pycrypto==2.6.1
pydocstyle==1.0.0
pyflakes==1.3.0
pygame==1.9.3
pygobject==3.22.0
pyinotify==0.9.6
PyJWT==1.4.2
pyOpenSSL==16.2.0
pyserial==3.2.1
python-apt==1.1.0b5
pyxdg==0.25
rainbowhat==0.1.0
requests==2.12.4
requests-oauthlib==0.7.0
RPi.GPIO==0.6.3
RTIMULib==7.2.1
scrollphat==0.0.7
scrollphathd==1.2.1
SecretStorage==2.3.1
sense-emu==1.1
sense-hat==2.2.0
simplejson==3.10.0
six==1.10.0
skywriter==0.0.7
sn3218==1.2.7
spidev==3.3
ssh-import-id==5.6
thonny==2.1.16
touchphat==0.0.1
twython==3.4.0
unicornhathd==0.0.4
urllib3==1.19.1
virtualenv==15.1.0
Werkzeug==0.11.15

 ```

**Photobooth log**

```
2018-10-18 21:53:31,807 - root - INFO - Initializing photobooth
2018-10-18 21:53:31,808 - root - INFO - Photobooth version: 0.4.dev11+g4ff982c
2018-10-18 21:53:31,811 - root - INFO - Reading config file "/home/pi/photobooth/photobooth/defaults.cfg"
2018-10-18 21:53:31,817 - root - INFO - Reading config file "photobooth.cfg"
2018-10-18 21:53:31,906 - root - INFO - Number of last existing file: 0
2018-10-18 21:53:31,911 - root - INFO - Saving pictures as "2018-10-18/photoboothXXXXX.jpg"
2018-10-18 21:53:31,915 - root - INFO - Number of last existing file: 0
2018-10-18 21:53:31,916 - root - INFO - Saving pictures as "2018-10-18/photobooth_shot_XXXXX.jpg"
2018-10-18 21:53:32,304 - root - INFO - GPIO enabled (lamp_pin=4, trigger_pin=24, exit_pin=23)
2018-10-18 21:53:35,223 - root - INFO - Using printer ""
2018-10-18 21:53:35,225 - root - INFO - Using PDF printer
2018-10-18 21:53:46,319 - root - INFO - Using gphoto2 via command line
2018-10-18 21:53:53,083 - root - ERROR - Camera: [Errno 2] No such file or directory: '/dev/shm/photobooth.jpg'
2018-10-18 21:54:07,837 - root - INFO - Exiting photobooth with status code 0

```

JohnsonKeast, 2018-10-18:
I am using the Canon EOS 650 D with the latest Rasbian Update and had problems with Camera Communication due to raw+jpg as botyas has stated above aswell as timeout settings on the camera, Once this was changed on the EOS the booth works fine.

I notice that the 1000D is not "officially" supported by gphoto2 however if you have it working on the command line this would suggest it should work?

jsitges, 2018-10-19:
I have the 1000D and also had communication issues. It would be starting
stopping preview and delaying the capture moment from this by several
seconds.  It did eventually took the image and processed it but it was
extremely laggy.  Cumbersome.  I will get a picam and work with that
instead.

Julio Sitges Pomares

jopaing, 2018-10-19:
Hi,

your hints were great. I needed all of them.
At first camera really had enabled RAW. After disabling, it worked, but every 4th image was not transferred properly. I reinstalled photobooth again and it works now.

I have 2 problems left:

**1. GPIO does not work.**
It is configured and starts, see log:
```
2018-10-19 19:52:37,174 - root - INFO - GPIO enabled (lamp_pin=4, trigger_pin=24, exit_pin=23)
´´´
I get no more gpio info in the log. even when i press the button.

If i test the button with this script it will be recognized:
```
#!/usr/bin/env python
#coding: utf8 
 
import RPi.GPIO as GPIO
 
# Zählweise der Pins festlegen
GPIO.setmode(GPIO.BOARD)
 
# Pin 18 (GPIO 24) als Eingang festlegen
GPIO.setup(18, GPIO.IN, pull_up_down = GPIO.PUD_DOWN)
 
# Schleifenzähler
i = 0
 
# Endlosschleife
while 1:
    # Eingang lesen
    if GPIO.input(18) == GPIO.HIGH:
        # Wenn Eingang HIGH ist, Ausgabe im Terminal erzeugen
        print "Eingang HIGH " + str(i)
        # Schleifenzähler erhöhen
        i = i + 1
´´´

2. preview does not work

```
2018-10-19 19:52:44,705 - root - INFO - Using camera without preview functionality
´´´
I know, that my camera does not have all functions. Which function is needed?
Once i had a very short python script, which had live picture. I do not find the script anymore, but it used pygame and took single preview pictures. One preview picture after the other. It was so fast, that it looked like a real live view on my pi3.  I think this is not the way this photobooth works?

jopaing, 2018-10-19:
Hi,

your hints were great. I needed all of them.
At first camera really had enabled RAW. After disabling, it worked, but every 4th image was not transferred properly. I reinstalled photobooth again and it works now.

I have 2 problems left:

**1. GPIO does not work.**
It is configured and starts, see log:
`
2018-10-19 19:52:37,174 - root - INFO - GPIO enabled (lamp_pin=4, trigger_pin=24, exit_pin=23)
`
I get no more gpio info in the log. even when i press the button.

If i test the button with this script it will be recognized:

```
#!/usr/bin/env python
#coding: utf8 
 
import RPi.GPIO as GPIO
 
# Zählweise der Pins festlegen
GPIO.setmode(GPIO.BOARD)
 
# Pin 18 (GPIO 24) als Eingang festlegen
GPIO.setup(18, GPIO.IN, pull_up_down = GPIO.PUD_DOWN)
 
# Schleifenzähler
i = 0
 
# Endlosschleife
while 1:
    # Eingang lesen
    if GPIO.input(18) == GPIO.HIGH:
        # Wenn Eingang HIGH ist, Ausgabe im Terminal erzeugen
        print "Eingang HIGH " + str(i)
        # Schleifenzähler erhöhen
        i = i + 1

```

**2. preview does not work**

`
2018-10-19 19:52:44,705 - root - INFO - Using camera without preview functionality
`
I know, that my camera does not have all functions. Which function is needed?
Once i had a very short python script, which had live picture. I do not find the script anymore, but it used pygame and took single preview pictures. One preview picture after the other. It was so fast, that it looked like a real movie on my pi3.  I think this is not the way this photobooth works?

jopaing, 2018-10-19:
I think i really have a problem with the camera.
Now i startet it again and had the same issue like before.
The last of the 4 shots makes trouble:

`
2018-10-19 21:01:26,811 - root - INFO - Using camera without preview functionality
2018-10-19 21:01:51,899 - root - INFO - Saving picture as 2018-10-19/photobooth_shot_00001.jpg
2018-10-19 21:02:06,672 - root - INFO - Saving picture as 2018-10-19/photobooth_shot_00002.jpg
2018-10-19 21:02:20,634 - root - INFO - Saving picture as 2018-10-19/photobooth_shot_00003.jpg
2018-10-19 21:02:33,262 - root - ERROR - Camera: image file is truncated (61 bytes not processed)
2018-10-19 21:02:33,922 - root - INFO - Saving picture as 2018-10-19/photobooth_shot_00004.jpg
2018-10-19 21:02:49,016 - root - INFO - Saving picture as 2018-10-19/photobooth00006.jpg
2018-10-19 21:03:06,784 - root - INFO - Printing picture
`

The number of bytes is different, but it is always the last shot.
It has stripes or colour problems. So the lower right picture in collection is not OK.

I fear it is time for a new camera.

reuterbal, 2018-10-19:
The problem with the last picture is known and was fixed some time ago in the development branch. Unfortunately I have hardly found the time to update the documentation to match all the new features, that is why those changes did not end up in the master branch, yet. However, to avoid future confusion, I have merged the development branch now.

Please update the photobooth code and try again.

For a preview functionality simply switch to one of the other gphoto2-bindings: python-gphoto2 or gphoto2-cffi

jopaing, 2018-10-19:
thank you for the fast help. Previe with python-gphoto2 works great. preview with gphoto2-cffi does not wirk and it is really slow.
python-gphoto is the friend.
Last picture is fixed - great.

Now i only have the issue with gpio. Also any ideas?

reuterbal, 2018-10-19:
I'm glad it's working now. 

Regarding the GPIO issue: in your example script you are using the BOARD numbering, however the photobooth app relies on gpiozero, which uses the BCM numbering scheme.
Your trigger button should be connected to BCM pin 24 and any ground pin, with the numbering according to [this layout](https://gpiozero.readthedocs.io/en/stable/recipes.html#pin-numbering).
Is it possible that you are simply using the wrong pins?

jopaing, 2018-10-19:
The PIN is correct. PIN 18 is BCM24. 
But i use different circuit. BCM24 is with 10kOhm to Ground. When i push the button, i connect it to 3,3V.
So it is a different use of the gpio. I removed the resistant and pull gpio down to GND. Now it works.
From circuit it is not the optimal thing, but it works.
Thank you.

Coffee will arrive this weekend.

reuterbal, 2018-10-19:
Great to hear that this is resolved as well! I'll close this, as it is now a collection of different issues. Feel free to open a fresh issue when new problems arise!

closed 2018-10-19 by reuterbal