reuterbal/photobooth #61: Trigger not working

jonherstein, 2018-10-23:
I have been able to get the lamp to work, but not a trigger button. I connect one pin to GPIO23, and the other to a Ground pin. I have tried two different types of buttons; one is a foot pedal and the other is a remotely controlled relay switch. Neither actually triggers the photobooth.

Also, is there a way to get rid of the "Happy?" and just go back to the beginning every time?

Thanks, it's a great project and it's ALMOST working for me!

jonherstein, 2018-10-23:
I figured it out! Thanks again! (I think the GPIO piece could be better documented for n00bs like me!

reuterbal, 2018-10-23:
Hi @jonherstein 

you can get rid of the "Happy"-screen by setting postprocessing time to 0. That way it will immediately return to the beginning after having shown the assembled picture.

I'm happy to improve the documentation! What precisely was missing in your case?

JohnsonKeast, 2018-10-23:
I am just going through the process of setting up the button and am also having troubles, are there resistors or other items required other than just connecting a button to the appropriate pin and a ground?

Also I would like to make the photobooth start automatically when the pi is powered up and have unsuccessfully tried to use bashrc and adding a script to localrc, any information on these things would be really appreciated thankyou :)

jsitges, 2018-10-23:
I am no programmer, but I have a program successfully start, and restart
whenever necessary using systemctl. Google it but you basically write a
strict format script that tells the system where your program is and then
reload the daemon, enable your script and finally start it all using sudo
systemctl.  Very neat, just keep an eye out and in your future searches I
am sure you'll find a tutorial on how to get it done.

Julio Sitges Pomares

reuterbal, 2018-10-24:
Hi @JohnsonKeast 

there are no additional parts needed for the trigger - a simple button is all. MJust make sure you use the correct pins - the photobooth uses the numbering according to the BCM layout.

Regarding the autostart, I documented a mechanism that is working for me [in the wiki](https://github.com/reuterbal/photobooth/wiki/Tips-and-tricks#autostart-application-on-raspberry-pi). It has the advantage of little changes to the system and, more importantly, avoids running the photobooth as root.

jonherstein, 2018-10-24:
I think the only thing that was confusing for me, as a relatively
inexperienced hardware guy, is what two pins to connect the switches to.
Does each switch get connected to a numbered GPIO pin and to Ground pin? Is
this true for both input switches and the lamp switch?

reuterbal, 2018-10-24:
I have put up [my wiring schematics](https://github.com/reuterbal/photobooth/blob/master/supplementals/wiring/wiring_bb.png). I hope that helps.
In principle: each switch must be connected to its own GPIO pin, but all can be connected to the same ground pin.

jonherstein, 2018-10-24:
Ok, I understand your wiring now for the lamp. Mine is a bit different. I
have a switch controlled power outlet that I need to trigger. So I don’t
need a 12V line, just a simple on/off to trigger a normally closed switch
to open.

Last question: what is the RGB lamp used for, and is there any
configuration settings for it?

JohnsonKeast, 2018-10-25:
Hi again, 

I have put this script 

#!/bin/bash

cd "$(dirname "$0")"
.venv/bin/python -m photobooth

into the /home/pi/photobooth directory and put the following line into the /home/pi/.config/lxsession/LXDE-pi/autostart file but nothing happens on start, below is the auto start file in case ive missed something?

@lxpanel --profile LXDE-pi
@pcmanfm --desktop --profile LXDE-pi
@xscreensaver -no-splash
@point-rpi
@bash /home/pi/photobooth/photobooth.sh
@home/pi/photobooth/autostart.sh - this is the line I added

Any help greatly appreciated.

Coffees coming your way.

Kind regards

Johnson

reuterbal, 2018-10-26:
HI @JohnsonKeast 

it looks good but not quite right. A number of things:
1. You should remove the "old" autostart-command from `/home/pi/.config/lxsession/LXDE-pi/autostart`
```
@bash /home/pi/photobooth/photobooth.sh
```
However, I don't think this should be causing any issues.
2. I think you missed a "/" in your `/home/pi/.config/lxsession/LXDE-pi/autostart`. The line should read like this:
```
@/home/pi/photobooth/autostart.sh
```
(Pay attention to the leading "/" after the "@")
3. Make sure the autostart-script is executable, i.e., in a terminal, execute the following command:
```
chmod a+x /home/pi/photobooth/autostart.sh
```
Hope that'll fix it.

JohnsonKeast, 2018-11-11:
Hi Again, 

Sorry to have to come back to you again but I followed your instructions above and still have no autostart unfortunately.

@lxpanel --profile LXDE-pi
@pcmanfm --desktop --profile LXDE-pi
@xscreensaver -no-splash
@point-rpi/
@/home/pi/photobooth/autostart.sh - this is the line I added the / to

I ran the chmod successfully in the terminal however the pi still starts up to the desktop screen.

Any other idea here?

Kind regards

Johnson

reuterbal, 2018-11-11:
Can you run the `autostart.sh` script from a command line to start the photobooth?