reuterbal/photobooth #62: GPIO Problem

serpentes80, 2018-10-24:
every time i boot the pi, start the photobooth everything seems ok.
if I exit the program and then restart it (or making changes in the settings and press save & restart) I get tons of these lines...

2018-10-24 22:38:51,506 - root - INFO - GPIO enabled (lamp_pin=4, trigger_pin=3, exit_pin=24, rgb_pins=(27, 22, 17))
2018-10-24 22:38:51,511 - root - INFO - GPIO enabled (lamp_pin=4, trigger_pin=3, exit_pin=24, rgb_pins=(27, 22, 17))
2018-10-24 22:38:51,517 - root - INFO - GPIO enabled (lamp_pin=4, trigger_pin=3, exit_pin=24, rgb_pins=(27, 22, 17))
2018-10-24 22:38:51,522 - root - INFO - GPIO enabled (lamp_pin=4, trigger_pin=3, exit_pin=24, rgb_pins=(27, 22, 17))
2018-10-24 22:38:51,533 - root - ERROR - Gpio: Failed to add edge detection

regardless which gpios are defined - even with the default.

system is the latest strech image fresh installed and fully patched

reuterbal, 2018-10-25:
Hi @serpentes80 

this sounds as if something does not exit correctly when you quit the program (Save & restart is basically quit and start again).
Could you please check the following: 
1. boot the pi
2. start the photobooth
3. make sure it works properly
4. exit the program
5. in a terminal, run
```
ps aux | grep python
```
and paste the output here.
6. start the photobooth again
7. verify that it is *not* working now...

serpentes80, 2018-10-25:
went through your steps until 5...
``` 
pi@Fotobox:~ $ ps aux | grep python
pi         982  0.0  0.0   4372   556 pts/0    S+   17:12   0:00 grep --color=auto python
pi@Fotobox:~ $
```
so no process running... but the error is still there if I start the program again.

reuterbal, 2018-10-26:
That's really strange, I have never encountered an issue like this.
Let's take a step back and simplify things.

Searching that issue on the web suggests it might be either related to the lack of root-rights, however, sudo-less GPIO should be default since 2015. The other possibility is the chosen pins, as some have some special features and can not be used for certain things.

Maybe try out [the script by @jopaing](https://github.com/reuterbal/photobooth/issues/60#issuecomment-431464055) posted in #60 and see if you can run it repeatedly. Make sure you test it with your pins (mind the numbering of the pins: the photobooth uses the BCM-layout, the script uses the BOARD-layout).

If it works once and fails the next time, try with sudo afterwards.

serpentes80, 2018-10-26:
@reuterbal 
I just tested the script you mentioned... I can run it as often as I want without problems.
As I mentione in my opening post, I've already tried the default GPIOs.

I even just did another clean fresh installation on a fresh downloaded raspbian stretch image, following your installation guide.

and:

same problem.

boot --> photobooth start --> everything works (including the GPIOs)
exit photobooth --> photobooth start --> the error occures...
 :(

serpentes80, 2018-10-27:
@sratxp 
bam there it was...

just put a clean downloaded img on the raspi's sd card, but skipped the patching.
looks, like the gpio problem isn't present there...
@reuterbal 
seems like it's something system related after the patching...

sratxp, 2018-10-27:
Yes, it should be the system update that causes problem. As my old image is working. Still it would be nice to understand the why this is happening. As there might be a problem in the Fotobox software.

But at least it works now.

reuterbal, 2018-10-29:
I was able to reproduce the issue this weekend. It is definitely connected to some upstream changes, probably in `RPi.GPIO` or possibly in `gpiozero`.

The problem is probably due to a lack of proper cleanup, and my guess is that gpiozero does not issue this correctly. One of the nice things about gpiozero is that it does not require you to do this but usually does it automatically in the destructor.

I haven't found out yet what exactly changed and what has to be done to overcome this problem, as I wasn't able to find a changelog for the Raspbian packages.

@sratxp the fact that this simple script produces the same kind of problems, I think the best thing would be to open a bug report in the [gpiozero project](https://github.com/RPi-Distro/python-gpiozero). When you do so, please put the link the to issue here.

reuterbal, 2019-01-12:
The following issue seems to be related: https://github.com/RPi-Distro/python-gpiozero/issues/687

Can anyone confirm that the problem was fixed with a later version of gpiozero or RPi-GPIO?

reuterbal, 2019-05-31:
This seems to have went away with changes upstream. Closing this - please reopen if problem persists.

closed 2019-05-31 by reuterbal

chino-lu, 2019-06-27:
I was trying to use the software on a Asus TinkerBoard
It is not working since gpizero does not exist for the tinker board.

Do you think it would be possible to change from gpiozero back to RPi.GPIO?
I think like this it would be more easy to replace for another GPIO system like the one from ASUS