reuterbal/photobooth #63: Leave original Pictures on Camera for backup purpose

serpentes80, 2018-10-25:
hey there,

would be great if there were a possibility to at least leave the orginal photos on the camera's sd card.

this should be easy to change, shouldn't it?
I just didn't find the clue ;)

reuterbal, 2018-10-26:
Hi @serpentes80 

unfortunately it is not that simple. Trigger-download-delete is a single gphoto-command but leaving pictures on the camera requires to do each step yourself. It goes in the same direction as earlier requests (#10, #17), where people asked for RAW+JPEG-support. In those issues I detailed some of the necessary steps, but I don't have time to do it at the moment.

If you want it simply for backup purposes, plug in a USB-drive and save a copy there.
You can either duplicate the save command in [`PictureSaver`](https://github.com/reuterbal/photobooth/blob/master/photobooth/worker/__init__.py) to write it again to a different location, or add a second `PictureSaver` to the `Worker` class with a different basename.

serpentes80, 2018-10-26:
gnaaa.
think it was too late yeterday ;)
gphoto2 -set-config capturetarget=1 solved that thing...
now the images taken get stored on the camera and stay there ;)

sry to bother you with that :)

closed 2018-10-26 by serpentes80

whatzat, 2020-10-04:
Hi @serpentes80 Can I ask where you made this change to the gphoto2 command to retain the images on the camera?