reuterbal/photobooth #65: Success story 1.0

botyas, 2018-10-30:
At last I was able to put the photobooth in use at a friend's wedding and everything went well. At first I had problems with the screen saver because with the commands that appear in the installation it did not work for me.

Then one of the guests stretched a photo before it finished printing and the printer was blocked for a few minutes and some photos were not printed.

Solved those first problems everything worked perfectly and we could enjoy making photos of all the guests.

After this first start-up I realized some things that still need to be polished and I'm already working to make the necessary improvements so everything is perfectly.

**Describe your photobooth**
I assembled all the devices in a wooden box with openings for the webcam, monitor, and push-button. through the back door you access the printer to change the cartridges and in case of any problem you can quickly access the interior.
  In this box, and due to lack of time, the printer tray is out for quick access to replace the paper. This was a problem since some guest tried to take the photo before it finished printing. Now I am mounting a ramp to place the printer inside and that only the photo will fall when I finish printing.

**Hardware:**
 - DeviceRaspberry Pi 3
 - Camera Logitech C920HD
 - GPIO: Yes
 
 **Additional equipment**
  - Canon Selphy CP1200

**Software:**
 - OS Raspbian
 - Python version 3

**Problems:**
 - countdown slows down a lot (5Seg almost 20Seg)
 - As I use a webcam, some photos went dark due to lack of light. I tried to put a led bulb that lights up when making the image.

**Pictures:**

[image]
[image]
[image]
[image]

reuterbal, 2018-10-30:
Hi @botyas 
thank you for sharing your success story!

I have made the same experience: the printer is sometimes the weakest link. Especially when there is a queue of pictures to be printed, it sometimes gets stuck (simply showing "busy"). Usually the fix is pulling the power cord of the printer and turning it back on however, sometimes single jobs get stuck in the CUPS queue and need to be restarted by hand.

Proper lights are a recurring issue. I solved it with a cheap studio lamp.

What I find strange is your countdown issue. Does this also occur if you use a different camera?

reuterbal, 2018-10-30:
In my case, when the printer runs out of supplies it is sufficient to replace them and it continues to print automatically.

The light is a simple always-on studio lamp. Additionally I'm using an RGB-strip for some illumination that does also light up bright white during capture. However, this adds only little to the total lighting.

What do you mean by putting the trigger LED directly to 12v? Of course if it is not connected to any GPIO pin it is always on. For 12V LEDs you'll have to have some transistor (I'm using a MOSFET) to switch the 12V power of the LED using the 3.3V GPIO pins.

sratxp, 2018-10-31:
> What I find strange is your countdown issue. Does this also occur if you use a different camera?

I do also have the issue with slow countdown. If I increase the resolution to Full HD the countdown is tremendously slow. My solution was to go to 720p resolution. I believe the problem is that raspberry is to slow for high resolutions as on my laptop I don't have any issues with countdown.

reuterbal, 2018-10-31:
That is quite possible. Actually, I have never checked what resolution my DSLR produces when being asked for preview pictures. I should probably include a reduced resolution into the OpenCV-wrapper during preview - I think some visible pixels during preview are bearable if this means a smoother preview with high fps-rates. Would you agree?

sratxp, 2018-10-31:
Sorry I should have been more specific. I've meant the desktop resolution. I'm not sure whether changing the desktop resolution has any effect on the preview resolution. But I do aggree with you that smoother preview is more important than preview resolution.

botyas, 2018-10-31:
It can be, I did all the tests at home with my Canon 60D DSLR and the time worked fine. I also tried other webcams and did not notice much difference. But the one that I put in the end is one that they gave me the same day and I could not do all the tests. The resolution of the webcam is FHD 1080 so it is very possible that it is due to that.

botyas, 2018-11-01:
which transistors you have used to control the trigger LED and the RGB strip. I have put an IRF520N and I can not get it to fire.

botyas, 2018-11-05:
Version 1.1 in progress. He has already made several modifications and adjustments while waiting for some other component to finish the rest.

- The printer has been changed so that it is completely inside. and so avoid getting the photo before it finishes printing.
[image]

[image]

- I made a very simple modification to the printer, adding a relay with a crontab script that executes on boot so that it turns on only without having to open it to turn it on.

[image]

- I replaced the transistors and now if the LED on the button is turned on.

- I have added a transistor for lighting, I still have not decided whether to put a car headlight or a 220v lamp (I have a relay ready in case the consumption goes up a lot.

How can I not add new outputs to the GPIO inside the program? I think I'm going to use a relay in the same output as the LED on the button to connect the illumination lamp.

There is still to finish configuring the router to connect to the raspberry and download the images.

To see if you get a system to share the images online or locally without affecting much to the operation. I will now finish what I have in hand and will make a clean installation to verify that the slowdown of the account back is not due to the fact that the server was in the background for the gallery website.

alopezbu, 2019-05-31:
> I am using an IRLZ34N ([specs](http://www.irf.com/product-info/datasheets/data/irlz34n.pdf)).
> The IRF520N ([specs](https://www.infineon.com/dgdl/irf520n.pdf?fileId=5546d462533600a4015355e339ee1983)) won't work as its maximum gate voltage is 4V but the GPIO pin will only deliver 3V. Thus it does not fully turn on.

And @reuterbal  the resistors you are using are of 12 KΩ +10% ??? Just to verify ;D

reuterbal, 2019-05-31:
They are 10 kOhm - does not really matter, simply use something large.