reuterbal/photobooth #66: Flash via gpio

unmanagedtn, 2018-11-18:
Is there any facilities to trigger a flash via gpio?

thebluehill, 2018-11-18:
I'm using the blue LED. But you have to modify the rgb-script.

reuterbal, 2018-11-18:
It is probably best to add an additional GPIO pin for a flash to [`photobooth/gpio/__init__.py`](https://github.com/reuterbal/photobooth/blob/master/photobooth/gpio/__init__.py). Use `_lamp` as a template for how to add it.

PrezP, 2018-11-30:
The picamera library has a built in way to do this, see https://picamera.readthedocs.io/en/release-1.13/recipes2.html#using-a-flash-with-the-camera

reuterbal, 2018-12-05:
I'd rather use some GPIO solution directly as many use gphoto2 instead of the picamera module.

BlackDolphinX, 2019-08-31:
I changed setActive in CameraPicamera.py:

`def setActive(self): 
 if self._cap is None or self._cap.closed:
            self._cap = PiCamera()
            self._cap.flash_mode = 'on'`

Works like a charm!

In order to use the flash trigger with the PiCamera it is required to setup a Flash-Pin like:
https://picamera.readthedocs.io/en/release-1.13/recipes2.html#using-a-flash-with-the-camera

Frank4365, 2020-05-04:
Hello
i'm not an expert in Python sripts and it is difficult to find out where and what to add in a py file to have an additionial GPIO pin giving HIGH for 0.5 second when taking the picture with the PiCamera.
The additional pin should be use as a shutter/trigger pin for a DSLR or flashing ligths.
I´m also not able to implenet the solution posted by BlackDolphinX  but this sound easiest way.
thank's for your help
F.