reuterbal/photobooth #67: CUPS Forbidden While Adding Printer (Fixed)

unmanagedtn, 2018-11-19:
This is related to CUPS and not the photobooth software but if the could be added to a FAQ or notes it might help others.

after installing cups with 
'_sudo apt-get install cups_' , reboot and then  cd to _/etc/cups_ and  '_sudo nano cups-files.conf_' and then  append '_pi_' to the end of the line "**SystemGroup lpadmin**" so it looks like **SystemGroup lpadmin pi**
and issue the command '_sudo systemctl restart cups_' to allow the pi user to add a printer.

reuterbal, 2018-11-19:
That is actually a good idea. Thank you for that.
However, the preferred way of doing this would be to add user pi to the lpadmin group.
I'll update the documentation.

reuterbal, 2018-11-28:
I just realized, it is already in the documentation: https://github.com/reuterbal/photobooth/blob/master/INSTALL.md#add-user-pi-to-group-lpadmin

closed 2018-11-28 by reuterbal