reuterbal/photobooth #68: Open CV install for webcam

unmanagedtn, 2018-11-21:
https://yoursunny.com/t/2018/install-OpenCV3-PiZero/

used this site to install OpenCV on my pi2 to use a usb webcam... might have to also install 

pip3 install opencv-python
sudo apt-get install libatlas-base-dev
sudo apt-get install libjasper-dev
sudo apt-get install libqtgui4
sudo apt-get install python3-pyqt5

to get it to work...

reuterbal, 2019-01-10:
Thanks for sharing this. We should include this in the documentation.
I don't have a USB webcam, so I can't test OpenCV on the Pi.

Wikinaut, 2019-02-18:
I wanted to use Photobooth with my Logitech Webcam C922 PRO on my Raspberry 2B+ and followed the instructions of @unmanagedtn. However, it was not working, after selecting "opencv" as webcam camera driver, I saw something like

```
File "/usr/local/lib/python3.5/dist-packages/cv2/__init__.py", line 4, in <module>
    from .cv2 import *
ImportError: libQtTest.so.4: cannot open shared object file: No such file or directory
```
After searching for the error message, I immediately found this solution https://stackoverflow.com/questions/49799798/can-not-import-opencv-in-python3-in-raspberry-pi3/50048216 , which **fixed** the problem and made Photobooth then **working** with my Webcam Logitech C922 PRO:

### Corrigendum:
* add ```sudo apt install libqt4-test```

### Cumulative fix:
```
pip3 install opencv-python
sudo apt-get install libatlas-base-dev
sudo apt-get install libjasper-dev
sudo apt-get install libqtgui4
sudo apt-get install python3-pyqt5
sudo apt install libqt4-test
```

By the way, on my Raspberry, it took only a few seconds to install all of them (not 2+ hrs).

reuterbal, 2019-06-27:
I think this is obsolete now, as Raspbian buster includes OpenCV 3.2.0. Can someone confirm?