reuterbal/photobooth #69: How does printing work? White bar on picture..

jopaing, 2018-11-23:
Hi,

with help of you, i have the photobooth working very well.
Only issue is printing.
When i print the picture on my DS-RX1, i get a white bar at right side of the picture and on bottom of print.
On left side, picture is missing some mm.

When i print the generated picture wia gimp, everything is perfect, so it is something with the printing part of the script.

Here are my details:
* Camera produces picture with 3888 x 2592 pixels
* Changing resolution of final picture seems to have no effect, only to the quality. For Testing i tried extreme values like 3888 x 2592 pixels and 150x100 pixels. Normally i use 1800x1200, because my printer does 300x300 dpi and paper is 4"x6" (102mm x 152mm)
* When i use the paper size 102mm x 152mm in settings, i get the white bars and left is picture information missing. Changing the size to higher values changes the behavour. White bar below and right diappear, but also more information on left side is missing. Looks like a kind of zoom for me.

How does printing work? Is there a way to print out the final picture without modification? If i could print the camera picture or created print picture directly it would work. Printer has its own logic to make it fit to printed paper.

I assume gimp works in that way.

reuterbal, 2018-11-28:
Hi,

the photobooth tries to use as many default settings as possible for printing as this is always a delicate and fragile topic.
Regarding the zoom-effect you should check in CUPS what default settings are specified there.

However, I experience the same behaviour with small white borders. The Qt printing system provides two values: pageRect() and paperRect(). In my setup I am again using pageRect in https://github.com/reuterbal/photobooth/blob/master/photobooth/printer/PrinterPyQt5.py to determine the print out size, which produces these small borders. On the other hand, paperRect() gets rid of these borders but cuts off parts of the image on the sides.

Try out either and let me know if you are able to find a stable configuration!

jopaing, 2018-12-03:
I found, that it s a problem of my printer calibration. It sooms not to be a special problem of the photobooth software. I created a picture with 1800x1200 pixel. As i see, i have the same problem with printing by other tools.

Here is the image for other users who want to check their printer.
Just open file in Gimp and print. For me Gutenberg driver is the best, but not perfect.

[image]

reuterbal, 2018-12-05:
Thanks for sharing! What is the workflow to calibrate the printer?

jopaing, 2020-01-22:
Solution found for my own system.
Photo printers seem to ignore the white border of the picture.

My printer DNP DS-RX1) starts left with first (colourful) pixel. So at the right, there is a white bar.
When changeing "Min. Distance Borders to Shots[px]" in settingt to "0" it works perfect.

I assume when you fill the border with any value, it should also work with border.

patkan, 2021-10-30:
For my printer(CP3100) the issue seems to be that the driver has no idea of how the paper looks normally. When printing with CUPS, a white border is on every picture. However, if you print from somewhere else with CUPS on this printer BEFORE you print with this application, the border disappears. In addition setting the image size explicitly in the CUPS call has helped me. You can see this in https://github.com/patkan/photobooth/blob/master/photobooth/printer/PrinterPyCups.py in my fork. (I call CUPS and pass the MediaSize as additional information)